project-chip/connectedhomeip #23920: [Platform] android build error :'NoneType' object has no attribute 'rfind'

chenguojiao, 2022-12-06:
### Reproduction steps

```bash
run TARGET_CPU=arm64 ./scripts/examples/android_app_ide.sh
then

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...[|]======================================================================
It looks like this version of Python does not support SSL. This is common
when using Homebrew. If using Homebrew please run the following commands.
If not using Homebrew check how your version of Python was built.

brew install openssl  # Probably already installed, but good to confirm.
brew uninstall python && brew install python
======================================================================
Traceback (most recent call last):
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 795, in <module>
    sys.exit(main())
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 787, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 457, in setup
    result = step(spin)
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 546, in cipd
    cipd_client = cipd_wrapper.init(
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 357, in init
    bootstrap(client, silent, rosetta=rosetta)
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 311, in bootstrap
    tmp.write(client_bytes(rosetta))
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 218, in client_bytes
    conn = https_connect_with_proxy(CIPD_HOST)
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 192, in https_connect_with_proxy
    conn = httplib.HTTPSConnection(url.hostname, url.port)
  File "/usr/local/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/http/client.py", line 1409, in __init__
    super(HTTPSConnection, self).__init__(host, port, timeout,
  File "/usr/local/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/http/client.py", line 851, in __init__
    (self.host, self.port) = self._get_hostport(host, port)
  File "/usr/local/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/http/client.py", line 885, in _get_hostport
    i = host.rfind(':')
AttributeError: 'NoneType' object has no attribute 'rfind'
```

### Platform

android

### Type

Manually tested with SDK

chenguojiao, 2022-12-06:
MacOs

closed 2022-12-08 by chenguojiao

gen-zi, 2023-02-28:
Has this problem been solved