project-chip/connectedhomeip #23927: Bridged Level Control is not exposed to Apple Home

lecndav, 2022-12-06:
### Reproduction steps

```bash
I extended the bridge example with the level control cluster and I am able to read/write to the endpoints using the chip-tool on the command line. Unfortunately the endpoints do not get exposed to apples Home App. I don't see any devices after I added the bridge.

This is the code I used to add the level control cluster:

#define DEVICE_TYPE_DIMMABLE_LIGHT 0x0101
DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(levelControlAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(ZCL_CURRENT_LEVEL_ATTRIBUTE_ID, INT8U, 1, 0),
DECLARE_DYNAMIC_ATTRIBUTE(ZCL_OPTIONS_ATTRIBUTE_ID, BITMAP8, 1, 0),
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

constexpr CommandId levelControlIncomingCommands[] = {
    app::Clusters::LevelControl::Commands::Move::Id,
    app::Clusters::LevelControl::Commands::MoveToLevel::Id,
    app::Clusters::LevelControl::Commands::MoveToLevelWithOnOff::Id,
    app::Clusters::LevelControl::Commands::MoveWithOnOff::Id,
    app::Clusters::LevelControl::Commands::Step::Id,
    app::Clusters::LevelControl::Commands::StepWithOnOff::Id,
    app::Clusters::LevelControl::Commands::Stop::Id,
    app::Clusters::LevelControl::Commands::StopWithOnOff::Id,
    kInvalidCommandId,
};

DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(bridgedLevelControlClusters)
DECLARE_DYNAMIC_CLUSTER(ZCL_LEVEL_CONTROL_CLUSTER_ID, levelControlAttrs, levelControlIncomingCommands, nullptr),
DECLARE_DYNAMIC_CLUSTER(ZCL_DESCRIPTOR_CLUSTER_ID, descriptorAttrs, nullptr, nullptr),
DECLARE_DYNAMIC_CLUSTER(ZCL_BRIDGED_DEVICE_BASIC_CLUSTER_ID, bridgedDeviceBasicAttrs, nullptr, nullptr)
DECLARE_DYNAMIC_CLUSTER_LIST_END;

DECLARE_DYNAMIC_ENDPOINT(bridgedLevelControlEndpoint, bridgedLevelControlClusters);

const EmberAfDeviceType gBridgedLevelControlDeviceTypes[] = {{DEVICE_TYPE_DIMMABLE_LIGHT, DEVICE_VERSION_DEFAULT}, {DEVICE_TYPE_BRIDGED_NODE, DEVICE_VERSION_DEFAULT}};
```
```

### Bug prevalence

Every time 

### GitHub hash of the SDK that was being used

9c0ca132b88df505909636dc26ba6763e0f57735

### Platform

core

carol-apple, 2022-12-06:
Please use the following to share description and logs of the issue you are running into. Once submitted, please share the feedback link here.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

lecndav, 2022-12-07:
Ok, here is the link: https://feedbackassistant.apple.com/feedback/11857698

tcarmelveilleux, 2022-12-07:
Overall, I would hypothesize that this is because the Device Type ID for the bridged endpoint is not set to a device type where the controller would expect to see LevelControl. Adding a cluster to an endpoint is usually not sufficient to make the endpoint react to it. Device Types are the core data model feature used to determine how to process a given endpoint.

bzbarsky-apple, 2022-12-07:
@lecndav Just to check: what does your `emberAfSetDynamicEndpoint` look like?

lecndav, 2022-12-08:
@bzbarsky-apple I haven't modified it, it is from the sdk in `attribute-storage.cpp`. Or do you mean `addDeviceEndpoint`? In that case, I didn't modify this one either. It's the original one from `main.cpp` in the `bridge-app` example.

bzbarsky-apple, 2022-12-09:
@lecndav I meant what does your call to `emberAfSetDynamicEndpoint` look like?  And if you are using `AddDeviceEndpoint`, wha does the call to _that_ look like?

lecndav, 2022-12-09:
@bzbarsky-apple  Here is the complete main.cpp

`/*
 *
 *    Copyright (c) 2021 Project CHIP Authors
 *    All rights reserved.
 *
 *    Licensed under the Apache License, Version 2.0 (the "License");
 *    you may not use this file except in compliance with the License.
 *    You may obtain a copy of the License at
 *
 *        http://www.apache.org/licenses/LICENSE-2.0
 *
 *    Unless required by applicable law or agreed to in writing, software
 *    distributed under the License is distributed on an "AS IS" BASIS,
 *    WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 *    See the License for the specific language governing permissions and
 *    limitations under the License.
 */

#include <AppMain.h>
#include <platform/CHIPDeviceLayer.h>
#include <platform/PlatformManager.h>

#include <app-common/zap-generated/af-structs.h>

#include <app-common/zap-generated/attribute-id.h>
#include <app-common/zap-generated/ids/Clusters.h>
#include <app/ConcreteAttributePath.h>
#include <app/EventLogging.h>
#include <app/chip-zcl-zpro-codec.h>
#include <app/reporting/reporting.h>
#include <app/util/af-types.h>
#include <app/util/af.h>
#include <app/util/attribute-storage.h>
#include <app/util/util.h>
#include <credentials/DeviceAttestationCredsProvider.h>
#include <credentials/examples/DeviceAttestationCredsExample.h>
#include <lib/core/CHIPError.h>
#include <lib/support/CHIPMem.h>
#include <lib/support/ZclString.h>
#include <platform/CommissionableDataProvider.h>
#include <setup_payload/QRCodeSetupPayloadGenerator.h>
#include <setup_payload/SetupPayload.h>

#include <pthread.h>
#include <sys/ioctl.h>

#include "CommissionableInit.h"
#include "Device.h"
#include "main.h"
#include <app/server/Server.h>

#include <cassert>
#include <iostream>
#include <vector>

using namespace chip;
using namespace chip::Credentials;
using namespace chip::Inet;
using namespace chip::Transport;
using namespace chip::DeviceLayer;
using namespace chip::app::Clusters;

namespace {

const int kNodeLabelSize = 32;
// Current ZCL implementation of Struct uses a max-size array of 254 bytes
const int kDescriptorAttributeArraySize = 254;

[... 1022 lines cut ...]
    gRooms.push_back(&room1);
    gRooms.push_back(&room2);
    gRooms.push_back(&room3);

    gActions.push_back(&action1);
    gActions.push_back(&action2);
    gActions.push_back(&action3);

    {
        pthread_t poll_thread;
        int res = pthread_create(&poll_thread, nullptr, bridge_polling_thread, nullptr);
        if (res)
        {
            printf("Error creating polling thread: %d\n", res);
            exit(1);
        }
    }

    // Run CHIP

    chip::DeviceLayer::PlatformMgr().RunEventLoop();

    return 0;
}
`

bzbarsky-apple, 2022-12-10:
So at first glance: that endpoint has level control but not the On/Off cluster, and does not have the LT feature turned on in the Level Control feature map.  So it's not following the spec for the "Dimmable Light" device, and that might be a factor here...

closed 2022-12-12 by lecndav

bzbarsky-apple, 2022-12-17:
@lecndav Glad to help!

lecndav, 2022-12-19:
@bzbarsky-apple Sorry to bother you again. My next step was to try to add a color control device. I added the required clusters, attributes and commands according to the spec (at least the best of my knowledge), but the device (`Light4`) does not show up. Can you take a look at my sample code?

The spec gives an example for a bridged color controlled device, where the device is in fact a composed device. (at 9.2. Endpoint Composition, Version 1.0-prerelease-ipr). I tried that and it didn't work either.
I think it isn't logical that a bridged color control device should be implemented with a composed device, because a level controlled device isn't also implemented that way. A color control device is just another superset of it. Does this have another point which it don't see?

```
/*
 *
 *    Copyright (c) 2021 Project CHIP Authors
 *    All rights reserved.
 *
 *    Licensed under the Apache License, Version 2.0 (the "License");
 *    you may not use this file except in compliance with the License.
 *    You may obtain a copy of the License at
 *
 *        http://www.apache.org/licenses/LICENSE-2.0
 *
 *    Unless required by applicable law or agreed to in writing, software
 *    distributed under the License is distributed on an "AS IS" BASIS,
 *    WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 *    See the License for the specific language governing permissions and
 *    limitations under the License.
 */

#include <AppMain.h>
#include <platform/CHIPDeviceLayer.h>
#include <platform/PlatformManager.h>

#include <app-common/zap-generated/af-structs.h>

#include <app-common/zap-generated/attribute-id.h>
#include <app-common/zap-generated/ids/Clusters.h>
#include <app/ConcreteAttributePath.h>
#include <app/EventLogging.h>
#include <app/chip-zcl-zpro-codec.h>
#include <app/reporting/reporting.h>
#include <app/util/af-types.h>
#include <app/util/af.h>
#include <app/util/attribute-storage.h>
#include <app/util/util.h>
#include <credentials/DeviceAttestationCredsProvider.h>
#include <credentials/examples/DeviceAttestationCredsExample.h>
#include <lib/core/CHIPError.h>
#include <lib/support/CHIPMem.h>
#include <lib/support/ZclString.h>
#include <platform/CommissionableDataProvider.h>
#include <setup_payload/QRCodeSetupPayloadGenerator.h>
#include <setup_payload/SetupPayload.h>

#include <pthread.h>
#include <sys/ioctl.h>

#include "CommissionableInit.h"
#include "Device.h"
#include "main.h"
#include <app/server/Server.h>

#include <cassert>
#include <iostream>
#include <vector>

using namespace chip;
using namespace chip::Credentials;
using namespace chip::Inet;
using namespace chip::Transport;
using namespace chip::DeviceLayer;
using namespace chip::app::Clusters;

namespace {

[... 1173 lines cut ...]
    // gRooms.push_back(&room1);
    // gRooms.push_back(&room2);
    // gRooms.push_back(&room3);

    // gActions.push_back(&action1);
    // gActions.push_back(&action2);
    // gActions.push_back(&action3);

    {
        pthread_t poll_thread;
        int res = pthread_create(&poll_thread, nullptr, bridge_polling_thread, nullptr);
        if (res)
        {
            printf("Error creating polling thread: %d\n", res);
            exit(1);
        }
    }

    // Run CHIP

    chip::DeviceLayer::PlatformMgr().RunEventLoop();

    return 0;
}
```

bzbarsky-apple, 2022-12-20:
@lecndav I don't know offhand.  Following the instructions from @carol-apple above might be better; might get you in touch with people who know what Home is looking for here.

stefano055415, 2023-01-17:
@lecndav  

I don't see the level light on homekit

I followed your example, but I didn't understand how to create the feature map.

could you give me an example

thank you

stefano055415, 2023-01-18:
@lecndav  @bzbarsky-apple 

Hi
I always get 0 from ZCL_LEVEL_CONTROL_CLUSTER_ID
can you help me?

EmberAfStatus emberAfExternalAttributeWriteCallback(EndpointId endpoint, ClusterId clusterId,
																										const EmberAfAttributeMetadata *attributeMetadata, uint8_t *buffer)
{
	uint16_t endpointIndex = emberAfGetDynamicIndexFromEndpoint(endpoint);

	if (endpointIndex < CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT)
	{
		Accessory *accessory = accessories[endpointIndex];

		if ((accessory->IsReachable()) && (clusterId == ZCL_ON_OFF_CLUSTER_ID))
		{
			printf("write ZCL_ON_OFF_CLUSTER_ID %d\n", *buffer);
			accessory->SetOn(*buffer == 1);
			return EMBER_ZCL_STATUS_SUCCESS;
		}

		if ((accessory->IsReachable()) && (clusterId == ZCL_LEVEL_CONTROL_CLUSTER_ID))
		{
			if ((attributeMetadata->attributeId == ZCL_CURRENT_LEVEL_ATTRIBUTE_ID))
			{
				printf("write ZCL_LEVEL_CONTROL_CLUSTER_ID %d\n", *buffer);
				accessory->SetLevel(buffer);
			}
			else
			{
				return EMBER_ZCL_STATUS_FAILURE;
			}
			return EMBER_ZCL_STATUS_SUCCESS;
		}
	}

	return EMBER_ZCL_STATUS_FAILURE;
}

lecndav, 2023-01-18:
@stefano055415 Have you applied to patch from [here](https://github.com/project-chip/connectedhomeip/issues/22523#issuecomment-1251440724). I don't know if this fixes your issue or if it's related to something else (and got already fixed).

bzbarsky-apple, 2023-01-19:
@stefano055415 I suggest filing a separate issue instead of commenting in a closed one, with details about the exact problem you are running into.

weihao042122, 2023-02-09:
@lecndav did you fix the problem about the color control device dose no show up to the Apple Home,

lecndav, 2023-02-13:
@weihao042122 No!