project-chip/connectedhomeip #23928: [BUG] Android CHIP Tool crashes when subscribing in Light On/Off & Level Cluster

markaj-nordic, 2022-12-06:
### Reproduction steps

```bash
1. Commission the light bulb accessory.
2. Go to LIGHT ON/OFF & LEVEL CLUSTER section.
3. Click Subscribe button.

After pressing the Subscribe in this window, the app crashes with following logs:

java.lang.NumberFormatException: For input string: ""
 at java.lang.Integer.parseInt(Integer.java:627)
 at java.lang.Integer.parseInt(Integer.java:650)
 at com.google.chip.chiptool.clusterclient.OnOffClientFragment$showSubscribeDialog$1$1.invokeSuspend(OnOffClientFragment.kt:107)
 at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
 at kotlinx.coroutines.DispatchedKt.resumeCancellable(Dispatched.kt:457)
 at kotlinx.coroutines.intrinsics.CancellableKt.startCoroutineCancellable(Cancellable.kt:26)
 at kotlinx.coroutines.CoroutineStart.invoke(CoroutineStart.kt:109)
 at kotlinx.coroutines.AbstractCoroutine.start(AbstractCoroutine.kt:154)
 at kotlinx.coroutines.BuildersKt__Builders_commonKt.launch(Builders.common.kt:54)
 at kotlinx.coroutines.BuildersKt.launch(Unknown Source:1)
 at kotlinx.coroutines.BuildersKt__Builders_commonKt.launch$default(Builders.common.kt:47)
 at kotlinx.coroutines.BuildersKt.launch$default(Unknown Source:1)
 at com.google.chip.chiptool.clusterclient.OnOffClientFragment.showSubscribeDialog$lambda-7(OnOffClientFragment.kt:105)
 at com.google.chip.chiptool.clusterclient.OnOffClientFragment.lambda$Q_xHxDRcLoxYxIbsECbiC6kJKoU(Unknown Source:0)
 at com.google.chip.chiptool.clusterclient.-$$Lambda$OnOffClientFragment$Q_xHxDRcLoxYxIbsECbiC6kJKoU.onClick(Unknown Source:8)
 at android.view.View.performClick(View.java:7253)
 at android.view.View.performClickInternal(View.java:7230)
 at android.view.View.access$3500(View.java:822)
 at android.view.View$PerformClick.run(View.java:27766)
 at android.os.Handler.handleCallback(Handler.java:883)
 at android.os.Handler.dispatchMessage(Handler.java:100)
 at android.os.Looper.loop(Looper.java:224)
 at android.app.ActivityThread.main(ActivityThread.java:7584)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)

Also the Subscription window in the LIGHT ON/OFF & LEVEL CLUSTER contains faulty GUI layout (overlapping captions) used when passing minimum and maximum subscription interval values.
```

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

1.0.0

### Platform

android

joonhaengHeo, 2022-12-14:
When I checked the log, it seems to be caused by not filling in the minInterval and maxInterval values in the dialog shown when I clicked the subscribe button. If you put that value in and it works, it doesn't seem to be a problem.

joonhaengHeo, 2022-12-15:
@markaj-nordic 
If you look at the Screen View you shared, I understand that the PR below I requested is the code before it was merged.
https://github.com/project-chip/connectedhomeip/pull/23377

Before the above PR is merged, the Dialog's EditText overlaps and the Value is not entered properly.

In my opinion, the following string is visible in Screen View, so it seems that the min or max value has not been entered.
https://github.com/project-chip/connectedhomeip/blob/b49c3ed920f7b95fc0f95ce5e8bdb82c85e757d8/examples/android/CHIPTool/app/src/main/res/values/strings.xml#L142

yunhanw-google, 2023-03-11:
fixed.

closed 2023-03-11 by yunhanw-google