project-chip/connectedhomeip #23930: [BUG] Fail to build EFR32 lighting-app example

lboue, 2022-12-06:
### Reproduction steps

```bash
1. cd ~/connectedhomeip
2. source scripts/activate.sh
3. ./scripts/examples/gn_efr32_example.sh ./examples/lighting-app/silabs/efr32/ ./out/efr32-lighting-app BRD4161A

+ USE_WIFI=false
+ SILABS_THREAD_TARGET='"../silabs:ot-efr32-cert"'
+ USAGE='./scripts/examples/gn_efr32_example.sh <AppRootFolder> <outputFolder> <silabs_board_name> [<Build options>]'
+ '[' 3 == 0 ']'
+ '[' 3 -lt 2 ']'
+ ROOT=./examples/thermostat/efr32/
+ OUTDIR=./out/thermostat-app
+ '[' 3 -gt 2 ']'
+ SILABS_BOARD=BRD4161A
+ shift
+ shift
+ shift
+ '[' 0 -gt 0 ']'
+ '[' -z BRD4161A ']'
+ BUILD_DIR=./out/thermostat-app/BRD4161A
+ echo BUILD_DIR=./out/thermostat-app/BRD4161A
BUILD_DIR=./out/thermostat-app/BRD4161A
+ '[' false == true ']'
+ '[' -z '' ']'
+ gn gen --check --fail-on-unused-args --export-compile-commands --root=./examples/thermostat/efr32/ '--args=silabs_board="BRD4161A"' ./out/thermostat-app/BRD4161A
ERROR at //third_party/connectedhomeip/third_party/openthread/BUILD.gn:27:23: Unable to load "/home/ubuntu/connectedhomeip/examples/thermostat/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/BUILD.gn".
      public_deps = [ "${openthread_root}:libopenthread-ftd" ]
                      ^-------------------------------------
AUTO-DETECTING input matter IDL file.
ubuntu@RPI4:~/connectedhomeip$ 
```
```

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

b14956185f67ae82074ce7c5aece2be1c0411557

### Platform

efr32

### Platform Version(s)

Python 3.10.6

### Anything else?

Full log:
```
ubuntu@RPI4:~$ cd ~/connectedhomeip
ubuntu@RPI4:~/connectedhomeip$ source scripts/activate.sh

  WELCOME TO...

         â–ˆ
         â–ˆ
     â–„   â–ˆ   â–„                                â–ˆ     â–ˆ
     â–€â–€â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–€â–€      â–„â–€â–€â–€â–„ â–„â–€â–€â–€â–„    â–„â–€â–€â–€â–€â–„â–ˆ  â–€â–€â–ˆâ–€â–€â–€â–€â–€â–ˆâ–€â–€   â–„â–€â–€â–€â–€â–„    â–„â–€â–€
   â–€â–ˆâ–„       â–„â–ˆâ–€   â–ˆ     â–ˆ     â–ˆ  â–ˆ      â–ˆ    â–ˆ     â–ˆ    â–ˆâ–„â–„â–„â–„â–„â–„â–ˆ  â–ˆ   
     â–€â–ˆâ–„   â–„â–ˆâ–€     â–ˆ     â–ˆ     â–ˆ  â–ˆ      â–ˆ    â–ˆ     â–ˆ    â–ˆ         â–ˆ   
  â–„â–ˆâ–ˆâ–€â–€â–ˆ   â–ˆâ–€â–€â–ˆâ–ˆâ–„  â–ˆ     â–ˆ     â–ˆ   â–€â–„â–„â–„â–„â–€â–ˆ    â–€â–„â–„   â–€â–„â–„   â–€â–„â–„â–„â–„â–€   â–ˆ   
 â–€â–€    â–ˆ   â–ˆ    â–€â–€

  ACTIVATOR! This sets your shell environment variables.
[... 83 lines cut ...]
+ SILABS_THREAD_TARGET='"../silabs:ot-efr32-cert"'
+ USAGE='./scripts/examples/gn_efr32_example.sh <AppRootFolder> <outputFolder> <silabs_board_name> [<Build options>]'
+ '[' 3 == 0 ']'
+ '[' 3 -lt 2 ']'
+ ROOT=./examples/lighting-app/silabs/efr32/
+ OUTDIR=./out/efr32-lighting-app
+ '[' 3 -gt 2 ']'
+ SILABS_BOARD=BRD4161A
+ shift
+ shift
+ shift
+ '[' 0 -gt 0 ']'
+ '[' -z BRD4161A ']'
+ BUILD_DIR=./out/efr32-lighting-app/BRD4161A
+ echo BUILD_DIR=./out/efr32-lighting-app/BRD4161A
BUILD_DIR=./out/efr32-lighting-app/BRD4161A
+ '[' false == true ']'
+ '[' -z '' ']'
+ gn gen --check --fail-on-unused-args --export-compile-commands --root=./examples/lighting-app/silabs/efr32/ '--args=silabs_board="BRD4161A"' ./out/efr32-lighting-app/BRD4161A
ERROR at //third_party/connectedhomeip/third_party/openthread/BUILD.gn:27:23: Unable to load "/home/ubuntu/connectedhomeip/examples/lighting-app/silabs/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/BUILD.gn".
      public_deps = [ "${openthread_root}:libopenthread-ftd" ]
                      ^-------------------------------------
AUTO-DETECTING input matter IDL file.

```

chenek, 2022-12-08:
Try to refer to steps in https://sunmaysky.blogspot.com/2022/08/how-to-setup-and-build-matterchip-efr32.html

lboue, 2022-12-08:
The tutorial indicated does not work because the path of the files has changed.

andy31415, 2022-12-09:
Tot works for me:

```shell
./scripts/build/build_examples.py --target efr32-brd4161a-light build
```

In terms of commands if you want to run manually:

```shell
#> ./scripts/build/build_examples.py --target efr32-brd4161a-light --dry-run --log-level warn build
# Commands will be run in CHIP project root.
cd "."

# Generating efr32-brd4161a-light
gn gen --check --fail-on-unused-args --export-compile-commands --root=/workspace/examples/lighting-app/silabs/efr32 '--args=silabs_board="BRD4161A" sl_matter_version_str="v1.0-master-d1c9050543"' /workspace/out/efr32-brd4161a-light

# Building efr32-brd4161a-light
ninja -C /workspace/out/efr32-brd4161a-light
```

lboue, 2022-12-09:
[andy31415](https://github.com/andy31415). 
I tried this with lastest commit d1c905054303b24a32eb6c70431acc02e0319077.

### build_examples.py

```
ubuntu@RPI4:~/connectedhomeip$ ./scripts/build/build_examples.py --target efr32-brd4161a-light build
2022-12-09 15:26:19 INFO    Building targets: efr32-brd4161a-light
2022-12-09 15:26:19 INFO    Preparing builder 'efr32-brd4161a-light'
2022-12-09 15:26:19 INFO    Generating /home/ubuntu/connectedhomeip/out/efr32-brd4161a-light
2022-12-09 15:26:19 INFO    Generating efr32-brd4161a-light
2022-12-09 15:26:20 INFO    ERROR at //third_party/connectedhomeip/third_party/openthread/BUILD.gn:27:23: Unable to load "/home/ubuntu/connectedhomeip/examples/lighting-app/silabs/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/BUILD.gn".
2022-12-09 15:26:20 INFO          public_deps = [ "${openthread_root}:libopenthread-ftd" ]
2022-12-09 15:26:20 INFO                          ^-------------------------------------
Traceback (most recent call last):
  File "/home/ubuntu/connectedhomeip/./scripts/build/build_examples.py", line 208, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/home/ubuntu/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/ubuntu/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/ubuntu/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/home/ubuntu/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/ubuntu/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/ubuntu/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/ubuntu/connectedhomeip/./scripts/build/build_examples.py", line 198, in cmd_build
    context.obj.Build()
  File "/home/ubuntu/connectedhomeip/scripts/build/build/__init__.py", line 67, in Build
    self.Generate()
  File "/home/ubuntu/connectedhomeip/scripts/build/build/__init__.py", line 62, in Generate
    builder.generate()
  File "/home/ubuntu/connectedhomeip/scripts/build/builders/efr32.py", line 264, in generate
    self._Execute(cmd, title=title)
  File "/home/ubuntu/connectedhomeip/scripts/build/builders/builder.py", line 104, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/home/ubuntu/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['gn', 'gen', '--check', '--fail-on-unused-args', '--export-compile-commands', '--root=/home/ubuntu/connectedhomeip/examples/lighting-app/silabs/efr32', '--args=silabs_board="BRD4161A" sl_matter_version_str="v1.0-master-d1c905054-dirty"', '/home/ubuntu/connectedhomeip/out/efr32-brd4161a-light'] failed: 1
```

### run manually

```
ubuntu@RPI4:~/connectedhomeip$ gn gen --check --fail-on-unused-args --export-compile-commands --root=/home/ubuntu/connectedhomeip '--args=silabs_board="BRD4161A" sl_matter_version_str="v1.0-master-d1c9050543"' ./out/efr32-brd4161a-light
ERROR at //BUILD.gn:122:3: Unknown function.
  pw_python_pip_install("pip_install_matter_packages") {
  ^--------------------
```

andy31415, 2022-12-09:
is this a clean environment? Try:

```
which python # curious here what your env is ... 
git submodule sync --recursive
git submodule update -f --init --recursive
```

If that still fails, do `rm -rf .environment` (path may be different depending on what `which python` says.
In a fresh shell do the `source scripts/bootstrap.sh` again

andy31415, 2022-12-09:
For me:

```
❯ ls -l third_party/openthread/ot-efr32/openthread/BUILD.gn
.rw-r--r-- andrei andrei 3.3 KB Wed Oct  5 12:03:26 2022  third_party/openthread/ot-efr32/openthread/BUILD.gn

❯ md5sum third_party/openthread/ot-efr32/openthread/BUILD.gn
923ed8c10b386e83a52b16411523345c  third_party/openthread/ot-efr32/openthread/BUILD.gn
```

lboue, 2022-12-09:
Thanks. I will run these commands then bootstrap and check if that fixes my problem.

lboue, 2022-12-09:
Thanks. This solved my problem. I managed to build efr32-brd4161a-light:
```
ubuntu@RPI4:~/connectedhomeip$ ls -l ./out/efr32-brd4161a-light/
total 93384
-rw-rw-r-- 1 ubuntu ubuntu       74 Dec  9 21:40 args.gn
-rw------- 1 ubuntu ubuntu   741585 Dec  9 21:40 build.ninja
-rw------- 1 ubuntu ubuntu    49930 Dec  9 21:40 build.ninja.d
-rw------- 1 ubuntu ubuntu        0 Dec  9 21:40 build.ninja.stamp
-rwxr-xr-x 1 ubuntu ubuntu      297 Dec  9 19:25 chip-efr32-lighting-example.flash.py
-rw-rw-r-- 1 ubuntu ubuntu  2745981 Dec  9 19:48 chip-efr32-lighting-example.hex
-rwxrwxr-x 1 ubuntu ubuntu 57029628 Dec  9 19:48 chip-efr32-lighting-example.out
-rw-rw-r-- 1 ubuntu ubuntu  8159594 Dec  9 19:48 chip-efr32-lighting-example.out.map
-rwxrwxr-x 1 ubuntu ubuntu  2806814 Dec  9 19:48 chip-efr32-lighting-example.s37
-rw-rw-r-- 1 ubuntu ubuntu 23319552 Dec  9 19:25 compile_commands.json
-rwxrwxr-x 2 ubuntu ubuntu     5634 Nov 16 21:40 efr32_firmware_utils.py
-rw-rw-r-- 2 ubuntu ubuntu    17062 Nov 16 21:40 firmware_utils.py
drwxrwxr-x 7 ubuntu ubuntu     4096 Dec  9 21:39 gen
drwxrwxr-x 2 ubuntu ubuntu     4096 Dec  9 19:48 lib
-rw-rw-r-- 1 ubuntu ubuntu      111 Dec  9 19:25 lighting_app.flashbundle.txt
drwxrwxr-x 6 ubuntu ubuntu     4096 Dec  9 21:39 obj
drwxrwxr-x 3 ubuntu ubuntu     4096 Dec  9 19:25 protocol_buffer
drwxrwxr-x 3 ubuntu ubuntu     4096 Dec  9 19:25 python
-rw-rw-r-- 1 ubuntu ubuntu      247 Dec  9 21:40 relative_path_transformations.json
-rw-rw-r-- 1 ubuntu ubuntu   750661 Dec  9 21:40 toolchain.ninja
```

closed 2022-12-09 by lboue