project-chip/connectedhomeip #23931: [TC-SWITCH-1.1]Switch Cluster--Feature-map value mismatch.

Rajashreekalmane, 2022-12-06:
**Test case:**
3.1.1. [TC-SWTCH-1.1] Global attributes with server as DUT	

**Problem**
As per test plan [TC-SWITCH-1.1] FeatureMap attribute showing value mismatch error when executed, DUT should
responds with feature map value 31 based on the depended attribute implementation  but currently DUT responding feature map value has 1.

**expected behavior:**- Based on the depended attribute implementation DUT should responds the feature map value as 31
**actual behavior:-** DUT is responding the feature map value as 1

**steps to reproduce:-** ./chip-tool switch read feature-map 1 1

**system configuration**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform :  wifi and onnetwork
Commit-ID:9c0ca132b88df505909636dc26ba6763e0f57735	
Dut application: all-clusters-app

**Test-plan Reference**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/switch.adoc#tc-swtch-1-1-global-attributes-with-server-as-dut

**Spec Reference**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/Switch.adoc#features

PFA log below:-
[Untitled 2.txt](https://github.com/project-chip/connectedhomeip/files/10166738/Untitled.2.txt)

Rajashreekalmane, 2022-12-07:
@bzbarsky-apple 

The list of Features in switch cluster is:-
- Latching Switch
- Momentary Switch
- Momentary Switch Release
- Momentary Switch LongPress
- Momentary Switch MultiPress

I believe that  attributes and events in Switch cluster with above feature as conformance are implemented it resembles that all the 
above feature is implemented so while reading feature map value it should be 31 instead 1.

bzbarsky-apple, 2022-12-09:
> The list of Features in switch cluster is:-

Yes, but some of those features are defined in the spec as being mutually exclusive.  So you can't implement all 5 features at the same time on the same endpoint per spec.

Rajashreekalmane, 2022-12-09:
Got it thank you @bzbarsky-apple

closed 2022-12-09 by Rajashreekalmane