project-chip/connectedhomeip #23946: Simulated test command execution error

manjunath-grl, 2022-12-07:
### Reproduction steps / Feature

```bash
Summary:
   TH not moving to the next step when DUT sends command to TH.
Steps to reproduce:
1. Add test TC_I_3_2_Simulated or any tests which sends command in steps.
2. Build app1
3. Run tests TC-I-3.1_simulated tests
4. Pair DUT and TH (Step 1)
5. When DUT sends the command to TH after success in TH side logs it not moves to next step

Added Diff link and execution logs below:
```

### Platform

other

### Type

YAML tested

### Anything else?

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Simualted_CommandNextStep_Error
Execution logs:
[DUT_I_3_2Logs.zip](https://github.com/project-chip/connectedhomeip/files/10174264/DUT_I_3_2Logs.zip)
[TH_I_3_2_Logs.zip](https://github.com/project-chip/connectedhomeip/files/10174265/TH_I_3_2_Logs.zip)

bzbarsky-apple, 2022-12-08:
@vivien-apple

vivien-apple, 2023-05-15:
I just tried that by adding `Test_TC_I_3_2_Simulated` to `app1`, build and launched with: `./out/app1/chip-app1 --command Test_TC_I_3_2_Simulated`

I can see TH moving to the next step when commands are received until I receive a `**** Test Complete: Test_TC_I_3_2_Simulated` log.

@manjunath-grl  Can you check if you still have the issue ?

manjunath-grl, 2023-06-15:
@vivien-apple @raju-apple Issue is fixed closing the issue.

closed 2023-06-15 by manjunath-grl