project-chip/connectedhomeip #23948: Merge app1 and app2 sample linux app

manjunath-grl, 2022-12-07:
### Reproduction steps / Feature

```bash
Summary:
        Currently when we need to add new cluster for sample apps here we need to update both app1 and app2 config.zap file.
```

### Platform

other

### Type

YAML tested

### Anything else?

Screenshot:
[image]

lboue, 2022-12-07:
What sample are you talking about in this specific case? Could you specify the file?

manjunath-grl, 2022-12-07:
@krypton36 opening up this ticket as discussed last week

manjunath-grl, 2022-12-07:
> 

@lboue i am talking about the file which is in examples/placeholder/linux/apps/app1 and app2