project-chip/connectedhomeip #23964: [BUG] setpoint-raise-lower command of thermostat cluster cannot be executed via any command-by-id form

cabin15, 2022-12-08:
### Reproduction steps

```bash
Precondition
Controller: chip-tool (linux)
Controlee: all-clusters-app (linux)
1. Perform commissioning between chip-tool and all-clusters-app
2. Send invoke command to all-clusters-app using below command and this works as expected.
./chip-tool thermostat setpoint-raise-lower 0 10 12 1
3. I converted the command on Step #2 using command-by-id on "any" cluster, and executed. This returns INVALID_COMMAND with TLVError 26
./chip-tool any command-by-id 0x0201 0x00 "{\"0x00\":0,\"0x01\":10}" 12 1
```

### Bug prevalence

Whenever I do this.

### GitHub hash of the SDK that was being used

3af0acd06cfb4a3e1b1a38c4d2091942ed0ef8a5

### Platform

core

### Platform Version(s)

N/A

### Anything else?

[Controller]

./chip-tool any command-by-id 0x0201 0x00 "{\"0x00\":0,\"0x01\":10}" 12 1
[1670461780.230256][38011:38011] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1670461780.233231][38011:38011] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1670461780.233413][38011:38011] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1670461780.233575][38011:38011] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1670461780.234005][38011:38011] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-N5cBVq)
[1670461780.234448][38011:38011] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1670461780.234490][38011:38011] CHIP:DL: NVS set: chip-counters/reboot-count = 10 (0xA)
[1670461780.235344][38011:38011] CHIP:DL: Got Ethernet interface: enp3s0
[1670461780.236046][38011:38011] CHIP:DL: Found the primary Ethernet interface:enp3s0
[1670461780.237137][38011:38011] CHIP:DL: Got WiFi interface: wlp4s0
[1670461780.237172][38011:38011] CHIP:DL: Failed to reset WiFi statistic counts
[1670461780.237219][38011:38011] CHIP:IN: UDP::Init bind&listen port=0
[1670461780.237285][38011:38011] CHIP:IN: UDP::Init bound to port=53653
[1670461780.237307][38011:38011] CHIP:IN: UDP::Init bind&listen port=0
[1670461780.237364][38011:38011] CHIP:IN: UDP::Init bound to port=34146
[1670461780.237378][38011:38011] CHIP:IN: BLEBase::Init - setting/overriding transport
[1670461780.237392][38011:38011] CHIP:IN: TransportMgr initialized
[1670461780.237423][38011:38011] CHIP:FP: Initializing FabricTable from persistent storage
[1670461780.237619][38011:38011] CHIP:TS: Last Known Good Time: 2022-12-08T09:28:40
[1670461780.238234][38011:38011] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xF528914BD5305E02, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1670461780.238620][38011:38011] CHIP:FP: Fabric index 0x2 was retrieved from storage. Compressed FabricId 0x87B6A9114D9FD2DB, FabricId 0x0251428E925103CA, NodeId 0x000000000001B669, VendorId 0xFFF1
[1670461780.239939][38011:38011] CHIP:ZCL: Using ZAP configuration...
[1670461780.242059][38011:38011] CHIP:CTL: System State Initialized...
[1670461780.242118][38011:38011] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1670461780.242153][38011:38011] CHIP:CTL: Setting attestation nonce to random value
[1670461780.242181][38011:38011] CHIP:CTL: Setting CSR nonce to random value
[1670461780.242225][38011:38011] CHIP:IN: UDP::Init bind&listen port=5550
[1670461780.242272][38011:38011] CHIP:IN: UDP::Init bound to port=5550
[1670461780.242288][38011:38011] CHIP:IN: UDP::Init bind&listen port=5550
[1670461780.242330][38011:38011] CHIP:IN: UDP::Init bound to port=5550
[1670461780.242343][38011:38011] CHIP:IN: TransportMgr initialized
[1670461780.242463][38011:38013] CHIP:DL: CHIP task running
[1670461780.242711][38011:38013] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1670461780.242742][38011:38013] CHIP:CTL: Setting attestation nonce to random value
[1670461780.242815][38011:38013] CHIP:CTL: Setting CSR nonce to random value
[1670461780.244757][38011:38013] CHIP:CTL: Generating NOC
[... 180 lines cut ...]
[1670461780.479271][37882:37882] CHIP:DMG: 			},
[1670461780.479291][37882:37882] CHIP:DMG: 		},
[1670461780.479321][37882:37882] CHIP:DMG: 		
[1670461780.479344][37882:37882] CHIP:DMG: 	],
[1670461780.479378][37882:37882] CHIP:DMG: 	
[1670461780.479401][37882:37882] CHIP:DMG: 	InteractionModelRevision = 1
[1670461780.479423][37882:37882] CHIP:DMG: },
[1670461780.479506][37882:37882] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_0201 e=1 p=o
[1670461780.479542][37882:37882] CHIP:DMG: AccessControl: allowed
[1670461780.479568][37882:37882] CHIP:DMG: Received command for Endpoint=1 Cluster=0x0000_0201 Command=0x0000_0000
[1670461780.479617][37882:37882] CHIP:DMG: Command handler moving to [ Preparing]
[1670461780.479649][37882:37882] CHIP:DMG: Command handler moving to [AddingComm]
[1670461780.479683][37882:37882] CHIP:DMG: Command handler moving to [AddedComma]
[1670461780.479715][37882:37882] CHIP:ZCL: Failed to dispatch command, TLVError=../../third_party/connectedhomeip/src/lib/core/CHIPTLVReader.cpp:185: CHIP Error 0x00000026: Wrong TLV type
[1670461780.479772][37882:37882] CHIP:DMG: Decreasing reference count for CommandHandler, remaining 0
[1670461780.479848][37882:37882] CHIP:EM: <<< [E:62102r M:103515895 (Ack:153381104)] (S) Msg TX to 1:000000000001B669 [5E02] --- Type 0001:09 (IM:InvokeCommandResponse)
[1670461780.479890][37882:37882] CHIP:IN: (S) Sending msg 103515895 on secure session with LSID: 18891
[1670461780.480041][37882:37882] CHIP:DMG: Command handler moving to [CommandSen]
[1670461780.480071][37882:37882] CHIP:DMG: Command handler moving to [AwaitingDe]
[1670461780.480181][37882:37882] CHIP:EM: >>> [E:62101r M:67311059 (Ack:260735549)] (U) Msg RX from 0:B6CA54FCC5408530 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1670461780.480220][37882:37882] CHIP:EM: Found matching exchange: 62101r, Delegate: (nil)
[1670461780.480256][37882:37882] CHIP:EM: Rxd Ack; Removing MessageCounter:260735549 from Retrans Table on exchange 62101r
[1670461780.481617][37882:37882] CHIP:EM: >>> [E:62102r M:153381105 (Ack:103515895)] (S) Msg RX from 1:000000000001B669 [5E02] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1670461780.481661][37882:37882] CHIP:EM: Found matching exchange: 62102r, Delegate: (nil)
[1670461780.481707][37882:37882] CHIP:EM: Rxd Ack; Removing MessageCounter:103515895 from Retrans Table on exchange 62102r

cabin15, 2022-12-08:
I wonder if this is a bug or my mistake using the wrong command.
It would be so kind if someone could check this.
Thank you.

closed 2022-12-08 by bzbarsky-apple

cabin15, 2022-12-08:
@bzbarsky-apple 
I really appreciate your super quick response :)
Let me try your command and get back to here.
Awesome!!

bzbarsky-apple, 2022-12-08:
@cabin15 You're welcome!  I created https://github.com/project-chip/connectedhomeip/pull/23965 to make this a little more self-documenting.

cabin15, 2022-12-08:
@bzbarsky-apple 
It would really help for those who are taking benefit from connectedhomeip.
Excellent!!!