project-chip/connectedhomeip #23971: Would CSA accept to certify Matter Bridge for Thread Nodes that do not have BLE feature ?

DanijelHodzicCG, 2022-12-08:
Consider the setup where we have Thread BR and Thread device that are commission with in-bound passphrase based method. So, there is no BLE channel enabled for out-of-band commissioning.

Would it be ok to implement Matter functionality only on Thread Border Router so that it can act as Matter Bridge for devices behind it?
Would CSA certify such implementation?

bzbarsky-apple, 2022-12-08:
This does not seem like an issue with the SDK code.  Please reach out to the CSA to ask them what they would or would not certify....

clapre, 2022-12-08:
Please reach out to help@csa-iot.org for questions on joining and certification issues.
Thanks.

DanijelHodzicCG, 2022-12-09:
Thank you very much