project-chip/connectedhomeip #23993: [TC-IDM-4.2] SDK Support required to give an access to particular Controller when two controller on same fabrics

Rajashreekalmane, 2022-12-09:
Test Case ID :- 
3.4.2. [TC-IDM-4.2] Subscription Response Messages from DUT Test Cases. [{DUT_Server}]

Problem: 
As per Test-plan  test steps are expecting to give an access to a particular TH when both TH1 and TH2 are in same fabrics 

Test steps :-
2, 3, 4 and 5 

Test-plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-4-2-subscription-response-messages-from-dut-test-cases-dut_server

vivien-apple, 2022-12-19:
For the different steps here is how I would proceed:

```
./out/debug/standalone/chip-tool pairing code 0x12344321 MT:-24J029Q00KA0648G00 
./out/debug/standalone/chip-tool accesscontrol write acl '[{"privilege":5, "authMode":2, "subjects":[112233], "targets":null, "fabricIndex": 1},{"privilege":5, "authMode":2, "subjects":[1], "targets":[{"cluster":6, "endpoint":1, "deviceType":null}, {"cluster":0, "endpoint":2, "deviceType":null}], "fabricIndex": 1}]' 0x12344321 0

Test Case 1:

# Try to subscribe to the occupancy attribute onto endpoint 1 for cluster “occupancy” from a node id that have access to the “onoff” cluster onto endpoint 1
./out/debug/standalone/chip-tool occupancysensing subscribe occupancy 1 5 0X12344321 1 --commissioner-nodeid 0x1 -> INVALID_ACTION

Test Case 2:
# Try to subscribe to all attributes onto endpoint 1 for cluster “occupancy” from a node id that have access to the “onoff” cluster onto endpoint 1
./out/debug/standalone/chip-tool occupancysensing subscribe 0xFFFFFFFF 1 5 0X12344321 1 --commissioner-nodeid 0x1 -> INVALID_ACTION

Test Case 3:
# Allow everything onto endpoint 1
./out/debug/standalone/chip-tool accesscontrol write acl '[{"privilege":5, "authMode":2, "subjects":[112233], "targets":null, "fabricIndex": 1},{"privilege":5, "authMode":2, "subjects":[1], "targets":[{"cluster":null, "endpoint":1, "deviceType":null}], "fabricIndex": 1}]' 0x12344321 0

# Try to subscribe to all attributes onto endpoint 2 for cluster 0x6 from a node id that have access to everything onto endpoint 1
./out/debug/standalone/chip-tool occupancysensing subscribe 0xFFFFFFFF 1 5 0X12344321 2 --commissioner-nodeid 0x1 -> INVALID_ACTION

Test Case 4:
# Try to subscribe to all attributes onto endpoint 1 for cluster 0x6 from a node id that does not have any access to the node
./out/debug/standalone/chip-tool onoff subscribe-by-id 0xFFFFFFFF 1 5 0x12344321 1 —commissioner-nodeid 0x2 -> INVALID_ACTION
```

cjandhyala, 2023-01-06:
@vivien-apple , The TC https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#342-tc-idm-42-subscription-response-messages-from-dut-test-cases-dut_server is talking about verifying receiving ""UNSUPPORTED_ACCESS" status response." but not INVALID_ACTION.

bzbarsky-apple, 2023-01-09:
Looking at the spec, in "8.4.3.2. Incoming Read Request and Subscribe Request Action Processing" it says:

> * Else if the path is a concrete path:
...
> ** Else if reading from the attribute in the path requires a privilege that is not granted to access the cluster in the path, an AttributeStatusIB SHALL be generated with the UNSUPPORTED_ACCESS Status Code.
...
> * If no error free existent paths remain, then AttributeRequests are considered empty.
...
> * If this action is in response to a Subscribe Request action,
> ** If both AttributeRequests and EventRequests are empty
> *** a Status Response Action with the INVALID_ACTION Status Code SHALL be sent to the initiator,

so for the case of a subscription when all paths fail, INVALID_ACTION is the right response per spec and the test plan is wrong.

Rajashreekalmane, 2023-04-21:
Issue has been resolved in Test-plan hence closing this issue

closed 2023-04-21 by Rajashreekalmane