project-chip/connectedhomeip #24056: [Platform] Android CHIPTool crash when shutdown subscriptions

Allen-Guof, 2022-12-13:
### Reproduction steps

```bash
1. Connected an esp32 device to CHIPTool.
2. Click "SENSOR CLUSTERS", then double click "WATCH" button
```

### Platform

android

### Type

Manually tested with SDK

### Anything else?

The logs:
```
Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 3296 (e.chip.chiptool), pid 3296 (e.chip.chiptool)
pid: 3296, tid: 3296, name: e.chip.chiptool  >>> com.google.chip.chiptool <<<
      #01 pc 0000000000102668  /data/app/com.google.chip.chiptool-UQ_o_CoWn8HeM0hvdRtPlA==/base.apk (offset 0x5ec000)
      #02 pc 0000000000102408  /data/app/com.google.chip.chiptool-UQ_o_CoWn8HeM0hvdRtPlA==/base.apk (offset 0x5ec000)
      #03 pc 000000000011e3d4  /data/app/com.google.chip.chiptool-UQ_o_CoWn8HeM0hvdRtPlA==/base.apk (offset 0x5ec000) (Java_chip_devicecontroller_ChipDeviceController_shutdownSubscriptions+68)
      #11 pc 00000000003ad5e8  [anon:dalvik-classes.dex extracted in memory from /data/app/com.google.chip.chiptool-UQ_o_CoWn8HeM0hvdRtPlA==/base.apk] (chip.devicecontroller.ChipDeviceController.shutdownSubscriptions+4)
      #14 pc 000000000001247a  [anon:dalvik-classes5.dex extracted in memory from /data/app/com.google.chip.chiptool-UQ_o_CoWn8HeM0hvdRtPlA==/base.apk!classes5.dex] (com.google.chip.chiptool.clusterclient.SensorClientFragment.unsubscribeSensorCluster+54)
      #17 pc 0000000000011fde  [anon:dalvik-classes5.dex extracted in memory from /data/app/com.google.chip.chiptool-UQ_o_CoWn8HeM0hvdRtPlA==/base.apk!classes5.dex] (com.google.chip.chiptool.clusterclient.SensorClientFragment.onViewCreated$lambda-3+66)
      #20 pc 0000000000011e94  [anon:dalvik-classes5.dex extracted in memory from /data/app/com.google.chip.chiptool-UQ_o_CoWn8HeM0hvdRtPlA==/base.apk!classes5.dex] (com.google.chip.chiptool.clusterclient.SensorClientFragment.lambda$UB59qL_8pfM1t7cOC5e8Tf3j5SI)
      #23 pc 0000000000008880  [anon:dalvik-classes5.dex extracted in memory from /data/app/com.google.chip.chiptool-UQ_o_CoWn8HeM0hvdRtPlA==/base.apk!classes5.dex] (com.google.chip.chiptool.clusterclient.-$$Lambda$SensorClientFragment$UB59qL_8pfM1t7cOC5e8Tf3j5SI.onCheckedChanged+4)
```

yunhanw-google, 2023-02-08:
https://github.com/project-chip/connectedhomeip/issues/23928
@joonhaengHeo  same issue?

joonhaengHeo, 2023-02-08:
@yunhanw-google 
No, it looks like a different problem.
For #23928, this is a problem caused by the lack of input value before subscribe, and this does not seem to be the problem.

MrLiuYunPing, 2023-02-09:
We cannot unsubscribe until the bug is fixed.

yunhanw-google, 2023-03-10:
fixed.

closed 2023-03-10 by yunhanw-google