project-chip/connectedhomeip #24076: [BUG] iPhone12Mini 16.2 Bluez disconnected immediately for darwin chip-tool with rasp4 chip-lighting-app

robinmo, 2022-12-14:
### Reproduction steps

```bash
1.run rasp4 chip-lighting-app
2.run darwin chip-tool on iPhone12mini with iOS 16.2
3.scan the qrcode generated with chip-lighting-app link https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A-24J0YXE00KA0648G00
4.the terminal for ssh rasp4 shows:
[1671008257.766338][3359:3362] CHIP:DL: Device 4B:50:01:11:8C:5D (Path: /org/bluez/hci0/dev_4B_50_01_11_8C_5D) Connected
[1671008258.866166][3359:3362] CHIP:DL: Bluez disconnected
[1671008258.866251][3359:3362] CHIP:DL: Bluez notify CHIPoBluez connection disconnected
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

9c0ca13 - Fix Linux standalone job. (#23897)

### Platform

darwin

### Platform Version(s)

iOS 16.2

### Anything else?

rasp4 full log:

```
ubuntu@ubuntu:~$ apps/chip-lighting-app 
[1671008251.647105][3359:3359] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1671008251.652608][3359:3359] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1671008251.652661][3359:3359] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
[1671008251.658982][3359:3359] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1671008251.659342][3359:3359] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1671008251.659516][3359:3359] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1671008251.659910][3359:3359] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-IabAmT)
[1671008251.660617][3359:3359] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1671008251.660690][3359:3359] CHIP:DL: NVS set: chip-counters/reboot-count = 2 (0x2)
[1671008251.661705][3359:3359] CHIP:DL: Got Ethernet interface: eth0
[1671008251.662408][3359:3359] CHIP:DL: Found the primary Ethernet interface:eth0
[1671008251.663146][3359:3359] CHIP:DL: Got WiFi interface: wlan0
[1671008251.663231][3359:3359] CHIP:DL: Failed to reset WiFi statistic counts
[1671008251.663276][3359:3359] CHIP:SPT: *** WARNING: Using temporary passcode 20202021 due to no neither --passcode or --spake2p-verifier-base64 given on command line. This is temporary and will disappear. Please update your scripts to explicitly configure onboarding credentials. ***
[1671008251.663306][3359:3359] CHIP:SPT: *** WARNING: Using temporary test discriminator 3840 due to --discriminator not given on command line. This is temporary and will disappear. Please update your scripts to explicitly configure discriminator. ***
[1671008251.663331][3359:3359] CHIP:SPT: PASE PBKDF iterations set to 1000
[1671008251.663376][3359:3359] CHIP:SPT: LinuxCommissionableDataProvider didn't get a PASE salt, generating one.
[1671008251.671166][3359:3359] CHIP:DL: Device Configuration:
[1671008251.671235][3359:3359] CHIP:DL:   Serial Number: TEST_SN
[1671008251.671272][3359:3359] CHIP:DL:   Vendor Id: 65521 (0xFFF1)
[1671008251.671309][3359:3359] CHIP:DL:   Product Id: 32769 (0x8001)
[1671008251.671345][3359:3359] CHIP:DL:   Hardware Version: 0
[1671008251.671369][3359:3359] CHIP:DL:   Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
[1671008251.671392][3359:3359] CHIP:DL:   Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)
[1671008251.671423][3359:3359] CHIP:DL:   Manufacturing Date: (not set)
[1671008251.671447][3359:3359] CHIP:DL:   Device Type: 257 (0x101)
[1671008251.671469][3359:3359] CHIP:-: ==== Onboarding payload for Standard Commissioning Flow ====
[1671008251.671509][3359:3359] CHIP:SVR: SetupQRCode: [MT:-24J042C00KA0648G00]
[1671008251.671559][3359:3359] CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
[1671008251.671582][3359:3359] CHIP:SVR: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A-24J042C00KA0648G00
[1671008251.671615][3359:3359] CHIP:SVR: Manual pairing code: [34970112332]
[1671008251.671650][3359:3359] CHIP:-: ==== Onboarding payload for Custom Commissioning Flows ====
[1671008251.671705][3359:3359] CHIP:SVR: SetupQRCode: [MT:-24J0YXE00KA0648G00]
[1671008251.671745][3359:3359] CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
[1671008251.671768][3359:3359] CHIP:SVR: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A-24J0YXE00KA0648G00
[1671008251.671803][3359:3359] CHIP:SVR: Manual pairing code: [749701123365521327694]
[... 140 lines cut ...]
[1671008251.779326][3359:3359] CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
[1671008251.779345][3359:3359] CHIP:SVR: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A-24J0YXE00KA0648G00
[1671008251.779391][3359:3359] CHIP:SVR: Manual pairing code: [749701123365521327694]
[1671008251.780764][3359:3364] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1671008251.780902][3359:3362] CHIP:DL: TRACE: Bus acquired for name MATTER-3840
[1671008251.781170][3359:3362] CHIP:DL: CREATE service object at /chipoble/0d1f/service
[1671008251.781941][3359:3362] CHIP:DL: Create characteristic object at /chipoble/0d1f/service/c1
[1671008251.782498][3359:3362] CHIP:DL: Create characteristic object at /chipoble/0d1f/service/c2
[1671008251.782842][3359:3362] CHIP:DL: CHIP BTP C1 /chipoble/0d1f/service
[1671008251.782872][3359:3362] CHIP:DL: CHIP BTP C2 /chipoble/0d1f/service
[1671008251.782897][3359:3362] CHIP:DL: CHIP_ENABLE_ADDITIONAL_DATA_ADVERTISING is FALSE
[1671008251.815398][3359:3359] CHIP:DL: PlatformBlueZInit init success
[1671008251.821402][3359:3362] CHIP:DL: BluezPeripheralRegisterAppDone done
[1671008251.821786][3359:3359] CHIP:DL: HandlePlatformSpecificBLEEvent 16392
[1671008251.821868][3359:3362] CHIP:DL: Create adv object at /chipoble/0d1f/advertising
[1671008251.822311][3359:3362] CHIP:DL: SET service data to {'0xFFF6': <[byte 0x00, 0x00, 0x0f, 0xf1, 0xff, 0x01, 0x80, 0x00]>}
[1671008251.823687][3359:3359] CHIP:DL: HandlePlatformSpecificBLEEvent 16393
[1671008251.823739][3359:3359] CHIP:DL: CHIPoBLE advertising config complete
[1671008251.834162][3359:3362] CHIP:DL: RegisterAdvertisement complete
[1671008251.834494][3359:3359] CHIP:DL: HandlePlatformSpecificBLEEvent 16394
[1671008257.766338][3359:3362] CHIP:DL: Device 4B:50:01:11:8C:5D (Path: /org/bluez/hci0/dev_4B_50_01_11_8C_5D) Connected
[1671008258.866166][3359:3362] CHIP:DL: Bluez disconnected
[1671008258.866251][3359:3362] CHIP:DL: Bluez notify CHIPoBluez connection disconnected

```

robinmo, 2022-12-14:
I found it show: 
```
[Ble] BLE:Error writing Characteristics in Chip service on the device: [The specified UUID is not allowed for this operation.]
```
but if I use iPhone SE iOS16.0, it runs well, and pair success with buletooth.

bzbarsky-apple, 2022-12-14:
@robinmo Please see documentation at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md and in particular you need to enable developer mode on your phone and follow the steps https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#profile-installation for this to work.

robinmo, 2022-12-15:
@bzbarsky-apple thanks. This error dismiss, after I installed the [Bluetooth Central Matter Client Developer mode profile](https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig).

However, how can I help the app user install this profile if I distribute a matter app? Is it needed for app distributed through Apple App store?

bzbarsky-apple, 2022-12-17:
> However, how can I help the app user install this profile if I distribute a matter app? 

If you're distributing a Matter app through the app store, it needs to be using the MatterSupport framework.  See https://developer.apple.com/documentation/mattersupport

closed 2022-12-17 by bzbarsky-apple

robinmo, 2023-01-03:
> > However, how can I help the app user install this profile if I distribute a matter app?
> 
> If you're distributing a Matter app through the app store, it needs to be using the MatterSupport framework. See https://developer.apple.com/documentation/mattersupport

Hi, @bzbarsky-apple [mattersupport](https://developer.apple.com/apple-home/matter) is unavailable until 16.1, so if I try to use matter in lower versions of iOS(such as 15.3), how can I fulfill it?

bzbarsky-apple, 2023-01-03:
@robinmo Matter.framework is not available there either, so if you are targeting those versions presumably you are shipping your own version of the whole Matter stack.  On versions before MatterSupport is available, you would use that; on versions where MatterSupport is available you use MatterSupport.

robinmo, 2023-01-04:
@bzbarsky-apple Thanks. So, go back to  the [Bluetooth Central Matter Client Developer mode profile](https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig), how can I help the app (download from App Store) user install it.

bzbarsky-apple, 2023-01-04:
@robinmo You can't.  But that profile is also not needed on iOS versions where MatterSupport does not exist.

robinmo, 2023-02-02:
> @robinmo You can't. But that profile is also not needed on iOS versions where MatterSupport does not exist.

Hi, @bzbarsky-apple I upload a app version with this repo whole matter stack, while without [Bluetooth Central Matter Client Developer mode profile](https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig) commission will fail, and after installing the ble profile it turns out success. So, the profile is needed for  iOS versions where MatterSupport does not exist? Then should I deal with installing the profile through my app?

bzbarsky-apple, 2023-02-02:
> So, the profile is needed for iOS versions where MatterSupport does not exist?

Again, no.  The profile is only needed in 16.1 or newer, where MatterSupport exists.  

So for things before 16.1 you can use your own Matter.framework (which you are presumably shipping, since that _also_ does not exist before 16.1) and commission over BLE.  Starting with 16.1 you need to use MatterSupport.  If you want to support both, that needs to be a runtime decision.

> Then should I deal with installing the profile through my app?

Normal users should not be installing this profile, period.  For one thing, the profile only works if you enable Developer Mode on your device, and normal users should not be doing that.

robinmo, 2023-02-02:
> The profile is only needed in 16

> Again, no. The profile is only needed in 16.1 or newer, where MatterSupport exists.
>
> So for things before 16.1 you can use your own Matter.framework (which you are presumably shipping, since that [...]

Got it, many thanks!

robinmo, 2023-03-15:
Hi，@bzbarsky-apple 
In the CHIP Tool iOS Sample Commissioner App readme doc says:
> Requires building and running the app to a physical device on iOS >= 15.4.
Does it mean that the sample app matter framework cannot support iOS < 15.4?
I have just changed the sample project configuration mini deployment version to iOS 11.0, 

[image]

built and ran, soon the app crashed 
[image]

bzbarsky-apple, 2023-03-15:
@robinmo I am pretty sure no one has tested Matter.framework on anything before iOS 13.

MatterSupport.framework does not exist before 15.x.

robinmo, 2023-03-16:
> @robinmo I am pretty sure no one has tested Matter.framework on anything before iOS 13.

@balducci-apple Many thanks.
Got it. 

[image]
However, I found that MatterSupport.h still contains nothing but Foundation in Xcode 14.3 beta.
should I use Matter.h instead?

bzbarsky-apple, 2023-03-16:
@robinmo My understanding is that MatterSupport is a Swift-only API.

robinmo, 2023-03-21:
@bzbarsky-apple 

> And just to be clear, you would use MatterSupport.framework to generate a setup code and then Matter.framework to commission using that setup code and interact with the device.

Generate a setup code? You mean pairing code or a QR code payload? 
I run the sample code as doc https://developer.apple.com/documentation/mattersupport?changes=latest_minor
```
import MatterSupport

let request = MatterAddDeviceRequest(
    topology: .init(ecosystemName: "Acme SmartHome", homes: [
        .init(displayName: "Default Acme Home"),
    ])
)

do {
    try await request.perform()
    print("Successfully set up a device!")
} catch {
    print("Failed to set up a device with error: \(error)")
}
```

it runs but stuck in the scene below, never get the callback.
[image]

Another issue about MatterSupport, when integrated with objective-c the bridge file(eg xx-Swift.h) turn out to show error
`Cannot find interface declaration for 'MatterAddDeviceExtensionRequestHandler', superclass of 'MatterAddDeviceHandler'`
[image]

My code:
```
//
//  MatterHelper.swift
//  xx
//
//  Created by Robin on 2023/3/21.
//

import Foundation
import MatterSupport

@available(iOS 16.1, *)
class MatterHelper: NSObject
{
    @objc class func commission()
    {
        let request = MatterAddDeviceRequest(
            topology: .init(ecosystemName: "RRRRRR", homes: [
                .init(displayName: "R11111"), .init(displayName: "R222222")
            ])
        )
        Task.init
        {
            do
            {
                try await request.perform()
                print("Successfully set up a device!")
            }
            catch
            {
                print("Failed to set up a device with error: \(error)")
            }
        }
        
    }
}

@available(iOS 16.1, *)
class MatterAddDeviceHandler: MatterAddDeviceExtensionRequestHandler
{
    override func validateDeviceCredential(_ deviceCredential: MatterAddDeviceExtensionRequestHandler.DeviceCredential) async throws
    {
        print("validateDeviceCredential \(deviceCredential)")
    }
    
    override func selectThreadNetwork(from threadScanResults: [MatterAddDeviceExtensionRequestHandler.ThreadScanResult]) async throws -> MatterAddDeviceExtensionRequestHandler.ThreadNetworkAssociation
    {
        print("selectThreadNetwork \(threadScanResults)")
        return ThreadNetworkAssociation.defaultSystemNetwork
    }
    
    override func commissionDevice(in home: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID) async throws
    {
        print("Home: \(String(describing: home)) " + "payload: \(onboardingPayload) "  + "commissionID: \(commissioningID)")
    }
}
```

Many thanks for your help.

bzbarsky-apple, 2023-03-21:
> Generate a setup code? You mean pairing code or a QR code payload?

I mean that you use MatterSupport.framework to generate a QR code payload, yes.

Past that, I'm afraid I can't help you much with MatterSupport, sorry....

aajain-com, 2023-03-21:
@robinmo  please file a Feedback Assistant request and attach the iOS logs to the ticket.

```
Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.
Apple Developer
Bug Reporting - Apple Developer
Now with Feedback Assistant available on iPhone, iPad, Mac, and the web, it's easier to submit effective bug reports and request enhancements to APIs and tools. (71 kB)
https://developer.apple.com/bug-reporting/#feedback-assistant
developer.apple.com
Profiles and Logs - Bug Reporting - Apple Developer
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.
developer.apple.com
Profiles and Logs - Bug Reporting - Apple Developer
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.
developer.apple.com
Profiles and Logs - Bug Reporting - Apple Developer
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.
```

robinmo, 2023-03-22:
> @robinmo I am pretty sure no one has tested Matter.framework on anything before iOS 13.
> 
> MatterSupport.framework does not exist before 15.x.

@bzbarsky-apple Hi，I found CHIP can support iOS from iOS11.0, the sample app darwin/CHIPTool use the **CoreNFC** framework, just need to change its `Link Binary With Libraries` **status to optional**

[image]

Then run and commission is OK.🎉🎉🎉

robinmo, 2023-03-23:
Hi, @aajain-com @bzbarsky-apple 

MatterSupport issue has been resolved by adding a matter extension.

However here is another problem:

we just want to make our mesh device hub  to be a controller and it do all the distribution of Node IDs and certificates(such as RAC ICAC). But with MatterSupport for now, I haven't found a way to configure these just as the CHIP do.
Do you have any suggestion?

bzbarsky-apple, 2023-03-23:
@robinmo What MatterSupport does is bring the device onto the IP network, open a commissioning window, and hand you a setup code you can use to commission the device into your fabric.

After that you use Matter.framework to do the commissioning, and you can do your own certificates/etc via that.

robinmo, 2023-03-24:
> @robinmo What MatterSupport does is bring the device onto the IP network, open a commissioning window, and hand you a setup code you can use to commission the device into your fabric.
> 
> After that you use Matter.framework to do the commissioning, and you can do your own certificates/etc via that.

So, could you provide OCW command related API in MatterSupport? If no? What's the duration of announcement(As the matter specification says, 5-15min)? And, if expired, how to trigger OCW and get the setup code?

bzbarsky-apple, 2023-03-24:
> So, could you provide OCW command related API in MatterSupport?

I'm not sure what you're asking for, exactly.

> Is it in MTRBaseDevice

The API you quote will open a commissioning window, yes, if you already have an MTRBaseDevice for an operational device.  But that does not sound like your situation, if you have not commissioned the device yet.

> What's the duration of announcement

The "duration" bit in the selector there.

> Is it the only way to do this

It is, yes, for devices that need to be commissioned over BLE.  Again: use MatterSupport, which will bring the device onto the IP network, open a commissioning window, and give you an onboarding payload that you can use to commission.  Then commission using that onboarding payload.

robinmo, 2023-03-24:
@bzbarsky-apple Got it. Many thanks.

robinmo, 2023-03-24:
@bzbarsky-apple 
I found MTRDevice initialization need params nodeID and controller. And to make a controller need some startup params.
At the first, I suppose to get these info from the MatterAddDeviceExtensionRequestHandler delegate, but found nothing about that.
MTRClusterAdministratorCommissioning has to open commission window(OCW) too, BUT the ` init?(device: MTRDevice, endpoint: UInt16, queue: DispatchQueue)` indicate demand of the MTRDevice instance.
How could I get these info? 
How could I construct a MTRDevice to invoke opening commission window?
Or there is another way？

robinmo, 2023-03-25:
> > And to make a controller need some startup params.
> 
> Yes... The controller represents the fabric you are trying to bring the device onto, so you need to configure it for that fabric.
> 
> > At the first, I suppose to get these info from the MatterAddDeviceExtensionRequestHandler delegate
> 
> How could that possibly work? Again, the controller is supposed to represent _your fabric_, and you're the only one who knows what the state of that fabric should be.
> 
> > How could I construct a MTRDevice to invoke opening commission window?
> 
> MatterSupport is supposed to open a commissioning window for you. You just need to create a controller representing your fabric, and then commission the device using that controller....

OMG, I supposed the controller is the one that has already commissioned the device.😅
Got it.