project-chip/connectedhomeip #24083: [BUG] Can't commission Matter device to OTBR with CHIP-Tool

venlx, 2022-12-14:
### Reproduction steps

```bash
I have a Raspberry Pi with an OTBR Docker and a nRF52840 Dongle as an RCP attached to it.
I start the Docker with a docker-compose file and establish a Thread Network with the OpenThread CLI.
The Matter device is a nRF52840DK flashed with a lock example using the nRF Connect 2.2.0 SDK and Toolchain.
I'm trying to commission the Matter device with the CHIP-Tool for Linux using the `pairing ble-thread` command. The CHIP-Tool is used in a Linux VM.
The device shortly connects to the Thread Network until the output `CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0x7fa3180491e0
CHIP:DL: HandlePlatformSpecificBLEEvent 16389` starts to loop. 
After a while a timeout occurs and the device disconnects from the Thread Network.
```

### Bug prevalence

Whenever i do this

### GitHub hash of the SDK that was being used

8a9ac9b

### Platform

nrf, raspi

### Anything else?

[chip-tool-output.txt](https://github.com/project-chip/connectedhomeip/files/10229765/chip-tool-output.txt)

ReneJosefsen, 2022-12-14:
Looking at the chip-tool log, it fails when it tries to find the node on the Thread network:

```
[1671034888.606729][3385:3387] CHIP:DIS: OperationalSessionSetup[1:00000000000004D2]: operational discovery failed: ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:109: CHIP Error 0x00000032: Timeout
[1671034888.606737][3385:3387] CHIP:CTL: Device connection failed. Error ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:109: CHIP Error 0x00000032: Timeout
[1671034888.606751][3385:3387] CHIP:CTL: Error on commissioning step 'FindOperational': '../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:109: CHIP Error 0x00000032: Timeout'
[1671034888.606756][3385:3387] CHIP:CTL: Failed to perform commissioning step 18
[1671034888.606761][3385:3387] CHIP:CTL: Going from commissioning step 'FindOperational' with lastErr = '../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:109: CHIP Error 0x00000032: Timeout' -> 'Cleanup'
```

There can be a couple of reasons why this fails and it is hard to dig much deeper without a device log as well.

venlx, 2022-12-15:
I added the device log

ReneJosefsen, 2022-12-15:
Looking at the device log, it seems ok. It gets its operational service registered in SRP, so I'm wondering if you have proper routing in your network 🤔

venlx, 2022-12-15:
How can I verify that?

ReneJosefsen, 2022-12-15:
You can try to use avahi-browse to see if you can see the services registered on the SRP server from the device that is running chip-tool.

venlx, 2022-12-16:
`srp server service` shows me `_matter._tcp and _matterc._udp` after trying to connect the devices. But `avahi-browse -rt _matter._tcp` and `avahi-browse -rt _matterc._udp` don't have any output.

venlx, 2022-12-21:
I tried to use chip-tool on the Raspberry Pi which gave me another error message
[pi-output.txt](https://github.com/project-chip/connectedhomeip/files/10275990/pi-output.txt)
[minicom-output.txt](https://github.com/project-chip/connectedhomeip/files/10276000/minicom-output.txt)

ReneJosefsen, 2022-12-21:
Looking at these logs it looks like it fails at setting up PASE:

`Failed to verify peer's MAC. This can happen when setup code is incorrect.` and `E: 11676 [SC]Failed during PASE session setup: 38`

venlx, 2022-12-21:
Is the setup code something different than the setup pin code from the matter device?

ReneJosefsen, 2022-12-21:
I could not see the command used for pairing, but from the devices log it should be 20202021.

venlx, 2022-12-21:
Yes that is the seetup code that I used.

venlx, 2022-12-21:
The command I'm using is: `chip-tool pairing ble-thread 1 hex:(the operational Dataset) 20202021 3840

venlx, 2022-12-21:
I tried to use chip-tool on the Raspberry Pi again but this time I used the chip-tool "chip-tool-linux_aarch64.zip" from https://github.com/nrfconnect/sdk-connectedhomeip/releases/tag/v2.2.0 and got again a different error message.
[pi-output.txt](https://github.com/project-chip/connectedhomeip/files/10277948/pi-output.txt)
[minicom-output.txt](https://github.com/project-chip/connectedhomeip/files/10277952/minicom-output.txt)

ReneJosefsen, 2022-12-21:
This time it looks like you are back to some kind of routing issue, since the DUT does register its service fine one SRP, but chip-tool is just not able to find it:

```
[1671627304.322591][34273:34278] CHIP:CTL: Error on commissioning step 'FindOperational': 
'../../src/inet/UDPEndPointImplSockets.cpp:411: OS Error 0x02000065: Network is unreachable'
[1671627304.322607][34273:34278] CHIP:CTL: Failed to perform commissioning step 18
```

Have you tried a different network setup?

venlx, 2022-12-21:
I'm using my companies network but I tried it also at my home network and it didn't work either.

Damian-Nordic, 2022-12-22:
Did you add static routes to Thread on-mesh prefix addresses (e.g. using sth like `sudo ip -6 route add fd11:22::/64 dev otbr0 via fd11:db8:1::2`) or configured the host on which you run chip-tool to accept router advertisements? "Network is unreachable" suggests that your host cannot figure out which network interface to use for sending the packet.

venlx, 2022-12-23:
I configured the host to accept router advertisements but it didn't help.
`sudo ip -6 route add fd11:22::/64 dev otbr0 via fd11:db8:1::2` only resulted in the error message `Cannot find device "otbr0"` so I tried to change it to `sudo ip -6 route add fd11:22::/64 dev docker0 via fd11:db8:1::2` but it also resulted in another error message `RTNETLINK answers: No route to host`

Damian-Nordic, 2022-12-23:
The `ip -6 route` command was just an example, if you can't/don't want to accept RAs, you need to set `via` to IPv6 address of the OTBR (be it a Raspberry Pi or address within a Docker network). You can take a look at our instructions for setting up the OTBR Docker with static routes: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_thread_tools.html#id8.

venlx, 2022-12-23:
Looking at https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_thread_tools.html#id8 I tried `sudo docker network create --ipv6 --subnet fd11:db8:1::/64 -o com.docker.network.bridge.name=otbr0 otbr` and `sudo ip -6 route add fd11:22::/64 dev otbr0 via fd11:db8:1::2` after restarting the docker but I still get the chip-tool timeout error

venlx, 2022-12-23:
I start the docker with a docker-compose file.

closed 2023-06-13 by venlx