project-chip/connectedhomeip #24092: [TC-IDM-3.2]Chip-tool support required to send null value as endpoint.

Rajashreekalmane, 2022-12-15:
### Feature description

...
Chip-tool support required to send the null value as endpoint  while performing the read and write functionality.
Related Test Case : 
TC-IDM-3.2
[image]

### Platform

raspi

### Anything else?

Test-plan Reference:
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-3-2-write-response-action-from-dut-to-th-dut_server

bzbarsky-apple, 2022-12-15:
What does "endpoint to null" mean?

If this is about wildcards, then this is already supported, no?  chip-tool can do a wildcard write.

@Rajashreekalmane

vivien-apple, 2022-12-16:
My understanding of this sense is that we want to set the value `null` to a given attribute on a given cluster and endpoint.
The wording just looks confusing.

If my understanding is correct you can do that with `chip-tool` already. e.g: 
```
$ chip-tool unittesting write nullable-boolean true 0x12344321 1
$ chip-tool unittesting read nullable-boolean 0x12344321 1
# Should read as "true"

$ chip-tool unittesting write nullable-boolean null 0x12344321 1
$ chip-tool unittesting read nullable-boolean 0x12344321 1
# Should read as "NULL"
```

bzbarsky-apple, 2022-12-17:
Ah, indeed, that is the point of this test: to set some attribute to the value null.  And yes, this is already supported.

closed 2022-12-17 by bzbarsky-apple

vivien-apple, 2022-12-19:
Reading it again and looking at the test cases that follows I suspect that this particular test case is about "replacing" one value at a particular index of an attribute of type "list".  And the new value should be set to "null".

Rajashreekalmane, 2022-12-19:
@vivien-apple and @bzbarsky-apple As per Test-Plan, require to send the wildcard WriteRequestmessage, The command path which is set to "Cluster=Particular, attribute=Particular, and Endpoint=Null"

bzbarsky-apple, 2022-12-19:
@Rajashreekalmane @vivien-apple No, step 20 is definitely talking about writing the value `null` to an attribute on "a given cluster and endpoint". 

Writing with a wildcard endpoint is step 2, not step 20.

I don't know what step 22 is supposed to be; there is no "delete an attribute value" operation (there is "delete an item from a list", but this is not what the test plan seems to be doing).

@cjandhyala could we please get this test plan clarified so people are not confused about it?