project-chip/connectedhomeip #24098: [Documentation] The letter 'c' is omitted in the word 'description'

LawrenceQuarshie, 2022-12-15:
### Documentation issues

### Omission of Letter 'c' in the word 'description'

The letter 'c' is missing in the word 'description' in the sentence "Please delete the instructions below and replace with PR description".

### Anything else?

[image]

https://github.com/project-chip/connectedhomeip/blob/master/.github/PULL_REQUEST_TEMPLATE.md

andy31415, 2023-02-08:
Consider creating a PR to fix this, should be an easy fix.

LawrenceQuarshie, 2023-02-09:
Well noted Andy.

That will be my first pull request on an open source project.

I will let you know when I am done.

Thank you for giving me this opportunity.

LawrenceQuarshie, 2023-02-10:
Hello @andy31415 

Done with the pull request.

It is opened yet to be closed.

Thank you.

LawrenceQuarshie, 2023-02-10:
Hello @andy31415,

I created a PR and now the issue is fixed and closed.

Thank you for the encouragement.

closed 2023-02-10 by LawrenceQuarshie