project-chip/connectedhomeip #24105: [nrfconnect] [nRF52840] how do I change button setting?

mmt-10, 2022-12-16:
### Reproduction steps

I would like to change DK_BTN1: P0.12 to P0.03.
However, DK_BTN1 = 1. it seems enum value.
How do I change it?

Host OS : Ubuntu 20.04
Board : nRF52840DK

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 7f2b0a5920d1a7acc202c2932ac4c626c54cd31e (HEAD -> v1.0-branch, origin/v1.0-branch)

### Platform

nrf

Damian-Nordic, 2022-12-19:
@mmt-10 Buttons are defined in the device tree. More precisely, in the device tree files of the given board. If you wish to override the default definitions, you need to create a device tree overlay: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/build/dts/howtos.html#set-devicetree-overlays. Not sure if the structure of `buttons` node is described somewhere, but if you look at the resulting DTS file in you build directory (`build/zephyr/zephyr.dts`) you should be able to figure out what needs to be changed.

In the future, consider using https://devzone.nordicsemi.com/ for non-Matter related help from Nordic.

mmt-10, 2022-12-20:
@Damian-Nordic 
Thank you for your advice.
I will try it.

closed 2022-12-20 by mmt-10