project-chip/connectedhomeip #24117: [BUG] No response to ScanNetworks command

kpark-apple, 2022-12-16:
### Reproduction steps

1. Set up PASE connection over BLE with a noncommissioned Nordic example app.
2. Send ScanNetworks command to Nordic.

### Bug prevalence

Heard about the problem just once

### GitHub hash of the SDK that was being used

2e2ae99dc270017dae758b3555a91f079b4b074d

### Platform

nrf

### Anything else?

From the nrf log attached below, ScanNetworks command was received at the following:

```
I: 74533 [EM]>>> [E:33912r M:138196100] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
D: 74544 [EM]Handling via exchange: 33912r, Delegate: 0x2000f53c
D: 74550 [DMG]Received command for Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0000
```

and then connection was terminated and no response was sent back:

```
I: 75717 [DL]BLE GAP connection terminated (reason 0x08)
```

[2022-12-15 at 02:57:01 PM.txt](https://github.com/project-chip/connectedhomeip/files/10248551/2022-12-15.at.02.57.01.PM.txt)

bzbarsky-apple, 2022-12-19:
@Damian-Nordic @kkasperczyk-no

kkasperczyk-no, 2022-12-20:
@kpark-apple could you describe how to reproduce the problem? Did you use chiptool? If yes, what commands did you invoke?

kkasperczyk-no, 2022-12-20:
Ok, I was able to test it with chiptool interactive mode and it worked for me. Here are the steps I made:

1. ./chiptool interactive start
2. ./chiptool pairing code-paseonly 1 34970112332
3. ./chiptool networkcommissioning scan-networks 1 0

CHIPTool log:
networkcommissioning scan-networks 1 0
```
[1671520873.534616][48617:48619] CHIP:TOO: Sending command to node 0x1
[1671520873.534815][48617:48619] CHIP:TOO: Sending cluster (0x00000031) command (0x00000000) on endpoint 0
[1671520873.534864][48617:48619] CHIP:DMG: ICR moving to [AddingComm]
[1671520873.534950][48617:48619] CHIP:DMG: ICR moving to [AddedComma]
[1671520873.535064][48617:48619] CHIP:EM: <<< [E:22954i M:143909953] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
[1671520873.535103][48617:48619] CHIP:IN: (S) Sending msg 143909953 on secure session with LSID: 46109
[1671520873.535175][48617:48619] CHIP:DMG: ICR moving to [CommandSen]
[1671520873.724184][48617:48619] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1671520874.670022][48617:48618] CHIP:DL: Indication received, conn = 0x7f1b00071230
[1671520874.670171][48617:48619] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1671520875.266624][48617:48619] CHIP:DL: Re-trying resolve
[1671520875.268966][48617:48619] CHIP:DL: Re-trying resolve
[1671520877.549673][48617:48619] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1671520878.901746][48617:48618] CHIP:DL: Indication received, conn = 0x7f1b00071230
[1671520878.901894][48617:48619] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1671520878.902053][48617:48619] CHIP:EM: >>> [E:22954i M:140549461] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
[1671520878.902086][48617:48619] CHIP:EM: Found matching exchange: 22954i, Delegate: 0x7f1af8042008
[1671520878.902109][48617:48619] CHIP:DMG: ICR moving to [ResponseRe]
[1671520878.902166][48617:48619] CHIP:DMG: InvokeResponseMessage =
[1671520878.902180][48617:48619] CHIP:DMG: {
[1671520878.902192][48617:48619] CHIP:DMG: 	suppressResponse = false, 
[1671520878.902205][48617:48619] CHIP:DMG: 	InvokeResponseIBs =
[1671520878.902248][48617:48619] CHIP:DMG: 	[
[1671520878.902267][48617:48619] CHIP:DMG: 		InvokeResponseIB =
[1671520878.902295][48617:48619] CHIP:DMG: 		{
[1671520878.902305][48617:48619] CHIP:DMG: 			CommandDataIB =
[1671520878.902317][48617:48619] CHIP:DMG: 			{
[1671520878.902327][48617:48619] CHIP:DMG: 				CommandPathIB =
[1671520878.902347][48617:48619] CHIP:DMG: 				{
[1671520878.902370][48617:48619] CHIP:DMG: 					EndpointId = 0x0,
[1671520878.902393][48617:48619] CHIP:DMG: 					ClusterId = 0x31,
[1671520878.902414][48617:48619] CHIP:DMG: 					CommandId = 0x1,
[1671520878.902435][48617:48619] CHIP:DMG: 				},
[1671520878.902460][48617:48619] CHIP:DMG: 				
[1671520878.902492][48617:48619] CHIP:DMG: 				CommandFields = 
[1671520878.902515][48617:48619] CHIP:DMG: 				{
[1671520878.902532][48617:48619] CHIP:DMG: 					0x0 = 0, 
[1671520878.902546][48617:48619] CHIP:DMG: 					0x3 = [
[1671520878.902580][48617:48619] CHIP:DMG: 							
[1671520878.902597][48617:48619] CHIP:DMG: 						{
[1671520878.902614][48617:48619] CHIP:DMG: 							0x0 = 4660, 
[1671520878.902633][48617:48619] CHIP:DMG: 							0x1 = 1229782938533634594, 
[1671520878.902655][48617:48619] CHIP:DMG: 							0x2 = "OpenThreadDemo" (14 chars), 
[1671520878.902673][48617:48619] CHIP:DMG: 							0x3 = 15, 
[1671520878.902691][48617:48619] CHIP:DMG: 							0x4 = 4, 
[1671520878.902716][48617:48619] CHIP:DMG: 							0x5 = [
[1671520878.902747][48617:48619] CHIP:DMG: 									0x5a, 0x01, 0x97, 0xdd, 0xb8, 0xf8, 0x5d, 0xd6, 
[1671520878.902776][48617:48619] CHIP:DMG: 							] (8 bytes)
[1671520878.902808][48617:48619] CHIP:DMG: 							0x6 = -44, 
[1671520878.902827][48617:48619] CHIP:DMG: 							0x7 = 200, 
[1671520878.902845][48617:48619] CHIP:DMG: 						},
[1671520878.902860][48617:48619] CHIP:DMG: 					],
[1671520878.902875][48617:48619] CHIP:DMG: 				},
[1671520878.902888][48617:48619] CHIP:DMG: 			},
[1671520878.902925][48617:48619] CHIP:DMG: 			
[1671520878.902938][48617:48619] CHIP:DMG: 		},
[1671520878.902967][48617:48619] CHIP:DMG: 		
[1671520878.902978][48617:48619] CHIP:DMG: 	],
[1671520878.903016][48617:48619] CHIP:DMG: 	
[1671520878.903030][48617:48619] CHIP:DMG: 	InteractionModelRevision = 1
[1671520878.903045][48617:48619] CHIP:DMG: },
[... 16 lines cut ...]
```

nRF device log:

```
D: 56497 [DMG]Received command for Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0000                                                    
D: 56508 [DMG]Decreasing reference count for CommandHandler, remaining 1                                                                 
D: 56516 [DL]OpenThread State Changed (Flags: 0x01009009)                                                                                
D: 56521 [DL]   Network Name: OpenThread                                                                                                 
D: 56524 [DL]   PAN Id: 0x0783                                                                                                           
D: 56527 [DL]   Extended PAN Id: 0xDEAD00BEEF00CAFE                                                                                      
D: 56532 [DL]   Channel: 11                                                                                                              
[... cut ...]
D: 61392 [DMG]Command handler moving to [AddingComm]                                                                                     
D: 61396 [DMG]Command handler moving to [AddedComma]                                                                                     
D: 61401 [DMG]Decreasing reference count for CommandHandler, remaining 0                                                                 
I: 61408 [EM]<<< [E:22954r M:140549461] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)              
I: 61419 [IN](S) Sending msg 140549461 on secure session with LSID: 37056
```

kpark-apple, 2022-12-20:
@kkasperczyk-no Note that we exercised this path many times and this issue was reported only once. I was hoping that the error reason code would provide an immediate hint.

closed 2023-08-11 by stale[bot]