project-chip/connectedhomeip #24143: [BUG]

FreyZhang007, 2022-12-20:
### Reproduction steps

1.  start `chip-all-clusters-app` and use `chip-tool` to pair
2. `./chip-tool onoff toggle 1 1` and kill chip-tool after it send CASE_Segma1.(Now `chip-all-clusters-app` is resending CASE_Segma2)
3. `./chip-tool onoff toggle 1 1` can't start a session before `chip-all-clusters-app` stop resending CASE_Segma2

### Bug prevalence

always

### GitHub hash of the SDK that was being used

SDK 1.0

### Platform

core

### Platform Version(s)

20.04.1-Ubuntu

### Anything else?

chip-tool
```txt
$ ./chip-tool-v1.0.0.1 onoff toggle 1 1 
[1671507864.915959][20017:20017] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1671507864.916235][20017:20017] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1671507864.916265][20017:20017] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1671507864.916282][20017:20017] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1671507864.916383][20017:20017] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-gBiXoP)
[1671507864.916569][20017:20017] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1671507864.916587][20017:20017] CHIP:DL: NVS set: chip-counters/reboot-count = 11 (0xB)
[1671507864.916809][20017:20017] CHIP:DL: Got Ethernet interface: enp0s3
[1671507864.916902][20017:20017] CHIP:DL: Found the primary Ethernet interface:enp0s3
[1671507864.917085][20017:20017] CHIP:DL: Failed to get WiFi interface
[1671507864.917093][20017:20017] CHIP:DL: Failed to reset WiFi statistic counts
[1671507864.917112][20017:20017] CHIP:IN: UDP::Init bind&listen port=0
[1671507864.917128][20017:20017] CHIP:IN: UDP::Init bound to port=42335
[1671507864.917131][20017:20017] CHIP:IN: UDP::Init bind&listen port=0
[1671507864.917141][20017:20017] CHIP:IN: UDP::Init bound to port=42712
[1671507864.917146][20017:20017] CHIP:IN: BLEBase::Init - setting/overriding transport
[1671507864.917148][20017:20017] CHIP:IN: TransportMgr initialized
[1671507864.917155][20017:20017] CHIP:FP: Initializing FabricTable from persistent storage
[1671507864.917186][20017:20017] CHIP:TS: Last Known Good Time: 2022-12-20T11:21:34
[1671507864.917309][20017:20017] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xFFA5FB03F2528B88, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1671507864.917778][20017:20017] CHIP:ZCL: Using ZAP configuration...
[1671507864.918118][20017:20017] CHIP:DL: MDNS failed to join multicast group on enp0s8 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:764: Inet Error 0x00000110: Address not found
[1671507864.918287][20017:20017] CHIP:DL: MDNS failed to join multicast group on enp0s9 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:764: Inet Error 0x00000110: Address not found
[1671507864.918594][20017:20017] CHIP:CTL: System State Initialized...
[1671507864.918613][20017:20017] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1671507864.918624][20017:20017] CHIP:CTL: Setting attestation nonce to random value
[1671507864.918630][20017:20017] CHIP:CTL: Setting CSR nonce to random value
[1671507864.918641][20017:20017] CHIP:IN: UDP::Init bind&listen port=5550
[1671507864.918650][20017:20017] CHIP:IN: UDP::Init bound to port=5550
[1671507864.918653][20017:20017] CHIP:IN: UDP::Init bind&listen port=5550
[1671507864.918680][20017:20017] CHIP:IN: UDP::Init bound to port=5550
[1671507864.918682][20017:20017] CHIP:IN: TransportMgr initialized
[1671507864.918861][20017:20019] CHIP:DL: CHIP task running
[1671507864.918984][20017:20019] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1671507864.919005][20017:20019] CHIP:CTL: Setting attestation nonce to random value
[1671507864.919021][20017:20019] CHIP:CTL: Setting CSR nonce to random value
[1671507864.919798][20017:20019] CHIP:CTL: Generating NOC
[1671507864.919995][20017:20019] CHIP:FP: Validating NOC chain
[1671507864.920411][20017:20019] CHIP:FP: NOC chain validation successful
[1671507864.920492][20017:20019] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1671507864.920534][20017:20019] CHIP:TS: Last Known Good Time: 2022-12-20T11:21:34
[1671507864.920540][20017:20019] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1671507864.920546][20017:20019] CHIP:TS: Retaining current Last Known Good Time
[... 587 lines cut ...]
CHIP:DIS: SRV record already actively processed.
CHIP:DIS: Warning: Attempt to mDNS broadcast failed on enp0s8:  ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:411: OS Error 0x02000065: Network is unreachable
CHIP:DIS: Warning: Attempt to mDNS broadcast failed on enp0s9:  ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:411: OS Error 0x02000065: Network is unreachable
CHIP:DIS: Warning: Attempt to mDNS broadcast failed on docker0:  ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:411: OS Error 0x02000065: Network is unreachable
CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 3 failures.
CHIP:DIS: SRV record already actively processed.
^CCHIP:DL: Caught signal 2
CHIP:DL: Select failed: ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/system/SystemLayerImplSelect.cpp:493: OS Error 0x02000004: Interrupted system call
CHIP:ZCL: Emitting ShutDown event
CHIP:EVL: LogEvent event number: 0x0000000000050003 priority: 2, endpoint id:  0x0 cluster id: 0x0000_0028 event id: 0x1 Sys timestamp: 0x0000000000825E73
CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
CHIP:IN: SecureSession[0x611000000f40]: Released - Type:2 LSID:35451
CHIP:DMG: IM WH moving to [Uninitialized]
CHIP:DMG: IM WH moving to [Uninitialized]
CHIP:DMG: IM WH moving to [Uninitialized]
CHIP:DMG: IM WH moving to [Uninitialized]
CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
CHIP:DMG: AccessControl: finishing
CHIP:DMG: Examples::AccessControlDelegate::Finish
CHIP:DL: writing settings to file (/tmp/chip_counters.ini-2Ga8Py)
CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
CHIP:DL: Inet Layer shutdown
CHIP:DL: System Layer shutdown
```

bzbarsky-apple, 2022-12-20:
@FreyZhang007 In the future, please attach logs as files instead of just pasting them in?

In any case, the SDK currently supports one incoming CASE handshake at a time (and constraint devices likely can't handle more than that in practice).  Given that, what behavior did you expect here?

FreyZhang007, 2022-12-21:
@bzbarsky-apple Thanks, I see. This is not a bug.
But, It waits 30 seconds(1671597307-1671597257) from `Sent Sigma2 msg` to `CASE Server enabling CASE session setups`. No new CASE can be established during this time. This time is a bit too long.
Can I reduce this waiting time?

```txt
[1671597257.514878][142643:142643] CHIP:SC: Sent Sigma2 msg
[1671597257.865415][142643:142643] CHIP:EM: Retransmitting MessageCounter:133262898 on exchange 51440r Send Cnt 1
[1671597257.865579][142643:142643] CHIP:IN: (U) Sending msg 133262898 to IP address 'UDP:[fdd5:582c:1c95:40e7:6949:dba:31f6:1fb8%enp0s3]:48712'
[1671597258.245189][142643:142643] CHIP:EM: Retransmitting MessageCounter:133262898 on exchange 51440r Send Cnt 2
...
[1671597307.536927][142643:142643] CHIP:IN: CASE Server enabling CASE session setups
```
[all-cluster.log](https://github.com/project-chip/connectedhomeip/files/10274596/all-cluster.log)

bzbarsky-apple, 2022-12-21:
The wait time is controlled by https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/protocols/secure_channel/CASESession.cpp#L127

You could reduce it, but of course that can cause problems if the device processing Sigma2 is a bit slow and the crypto takes a while...

FreyZhang007, 2022-12-22:
Thanks, It is useful for me.

closed 2022-12-22 by FreyZhang007