project-chip/connectedhomeip #24149: Need YAML support to validate the struct fileds from "CONTAINS" keyword

manjunath-grl, 2022-12-20:
### Feature description

When we tried to use the "contains" keyword to validate event struct,  YAML script build seems to fail are we missing anything with respect to the usage of this keyword ?

### Platform

other

### Anything else?

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Contains_Keyword_for_Struct

YAML screenshot:
[image]

Testplan screenshot:
[image]

[exe_log.txt](https://github.com/project-chip/connectedhomeip/files/10267867/exe_log.txt)

bzbarsky-apple, 2022-12-20:
"contains" as currently implemented really only works for lists of integer values.

bzbarsky-apple, 2023-10-07:
I can fix `contains` to work for structs, but it won't help here, because readEvent does not return a list....