project-chip/connectedhomeip #24158: [BUG] Failure in cert bins docker image creation

raju-apple, 2022-12-20:
### Reproduction steps

Running this command : 

docker buildx build --build-arg COMMITHASH=fb185956b987859633acc008bcba17db64d0e3df --platform linux/arm64 --tag connectedhomeip/chip-cert-bins:fb185956b987859633acc008bcba17db64d0e3df --push .

Seeing this fail here : 

```

#0 20.03 WARNING: you are using different node version than recommended.
#0 20.03 🔧 Using temporary state directory: /tmp/zap3qx01sx3/zap.h1xleM
#0 20.04 🤖 ZAP generation started: 
#0 20.04     🔍 input files: /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 20.04     🔍 output pattern: /root/connectedhomeip/scripts/examples/../../zzz_generated/placeholder/app1/zap-generated
#0 20.04     🔍 using templates: /root/connectedhomeip/src/app/zap-templates/app-templates.json
#0 20.04     🔍 using zcl data: /root/connectedhomeip/src/app/zap-templates/zcl/zcl.json
#0 20.04     🔍 zap version: ver. 2022.11.29, featureLevel 87, commit: 352ea84a27a6e0699c000b9a8cff72029b4b8b75 from 2022-11-29T14:37:33.000Z, mode: source, exe: node
#0 20.86 🕐 Setup time: 829ms 
#0 20.86 👉 using input file: /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 20.87 [Error: SQLITE_ERROR: cannot commit - no transaction is active] {
#0 20.87   errno: 1,
#0 20.87   code: 'SQLITE_ERROR'
#0 20.87 }
#0 20.87 ⇝ Zap generation error: Error: SQLITE_ERROR: cannot commit - no transaction is active
#0 22.46 👎 Program npx exited with error code: 1
#0 22.46 1
#0 22.47 Searching for zcl file from /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 22.47 Traceback (most recent call last):
#0 22.47   File "/root/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 295, in <module>
#0 22.47     main()
#0 22.47   File "/root/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 278, in main
#0 22.47     runGeneration(cmdLineArgs.zapFile, cmdLineArgs.zclFile, cmdLineArgs.templateFile, cmdLineArgs.outputDir)
#0 22.47   File "/root/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 175, in runGeneration
#0 22.47     subprocess.check_call(generate_cmd + ['-z', zcl_file, '-g', templates_file,
#0 22.47   File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
#0 22.47     raise CalledProcessError(retcode, cmd)
#0 22.47 subprocess.CalledProcessError: Command '['node', 'src-script/zap-start.js', 'generate', '-z', '/root/connectedhomeip/src/app/zap-templates/zcl/zcl.json', '-g', '/root/connectedhomeip/src/app/zap-templates/app-templates.json', '-i', '/root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap', '-o', '/root/connectedhomeip/scripts/examples/../../zzz_generated/placeholder/app1/zap-generated']' returned non-zero exit status 1.
------
Dockerfile:269
--------------------
 267 |         && npm ci
 268 |     ENV ZAP_DEVELOPMENT_PATH=/opt/zap-${ZAP_VERSION}
 269 | >>> RUN scripts/examples/gn_build_test_example.sh app1
 270 |     
 271 |     RUN source scripts/activate.sh && scripts/build_python.sh -m platform -d true -i no
--------------------
ERROR: failed to solve: process "/bin/bash -c scripts/examples/gn_build_test_example.sh app1" did not complete successfully: exit code: 1
```

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

fb185956b987859633acc008bcba17db64d0e3df

### Platform

raspi

### Platform Version(s)

linux/arm64

### Anything else?

Full logs attached :  [Docker_error_THv2.6beta2.txt](https://github.com/project-chip/connectedhomeip/files/10273411/Docker_error_THv2.6beta2.txt)

raju-apple, 2023-01-03:
@andy31415 can you take a look when you get a chance please ?

raju-apple, 2023-01-03:
Seeing the same error on TOT ( a2a3ac154a09ab7ea49488544ff817c2f86b440c )

```
 => ERROR [chip-build-cert-bins 4/5] RUN scripts/examples/gn_build_test_example.sh app1                                                                                                                                                  23.7s 
------                                                                                                                                                                                                                                         
 > [chip-build-cert-bins 4/5] RUN scripts/examples/gn_build_test_example.sh app1:                                                                                                                                                              
#0 1.238                                                                                                                                                                                                                                       
#0 1.238   WELCOME TO...                                                                                                                                                                                                                       
#0 1.238                                                                                                                                                                                                                                       
#0 1.240          █                                                                                                                                                                                                                            
#0 1.240          █
#0 1.240      ▄   █   ▄                                █     █
#0 1.240      ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
#0 1.240    ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
#0 1.240      ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
#0 1.240   ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
#0 1.240  ▀▀    █   █    ▀▀
#0 1.240 
#0 1.240   ACTIVATOR! This sets your shell environment variables.
#0 1.240 
#0 1.242 Activating environment (setting environment variables):
#0 1.242 
#0 1.242   Setting environment variables for CIPD package manager...done
#0 1.257   Setting environment variables for Python environment.....done
#0 1.261   Setting environment variables for pw packages............skipped
#0 1.261   Setting environment variables for Host tools.............done
#0 1.264 
#0 1.264 Checking the environment:
#0 1.264 
#0 1.766 20230103 21:27:08 INF Environment passes all checks!
#0 1.822 
#0 1.822 Environment looks good, you are ready to go!
#0 1.822 
#0 1.929 🚀 Executing: git log -1 --format={"hash": "%H","timestamp": %ct}
#0 1.937 🔍 Git commit: 352ea84a27a6e0699c000b9a8cff72029b4b8b75 from Tue Nov 29 2022 14:37:33 GMT+0000 (Coordinated Universal Time)
#0 1.957 🔍 Current src hash: BIDD5bIBBoOMtKS2mxFtAt6dYoQ=
#0 1.960 🔍 Current src-shared hash: TekYsEMUhqRD66IAkrwVUv1961g=
#0 1.961 👎 Error reading old hash file: /opt/zap-v2022.11.29-nightly/spa/hash.json
#0 1.961 🐝 Front-end code changed, so we need to rebuild SPA.
#0 1.961 🚀 Executing: npx quasar build
#0 2.133 
#0 2.133  Build mode........ spa
#0 2.133  Pkg quasar........ v1.19.1
#0 2.133  Pkg @quasar/app... v2.4.1
#0 2.133  Debugging......... no
#0 2.133  Publishing........ no
#0 2.133 
#0 2.134 ⇝ Browserslist: caniuse-lite is outdated. Please run:
#0 2.134   npx browserslist@latest --update-db
#0 2.134   Why you should do it regularly: https://github.com/browserslist/browserslist#browsers-data-updating
#0 2.136  Configured browser support (at least 47.62% of global marketshare):
#0 2.136  · Chrome for Android >= 102
#0 2.136  · Firefox for Android >= 101
#0 2.136  · QQ for Android >= 10.4
#0 2.136  · UC for Android >= 12.12
#0 2.136  · Android >= 101
#0 2.136  · Chrome >= 102
#0 2.136  · Edge >= 102
#0 2.136  · Firefox >= 101
#0 2.136  · IE >= 11
#0 2.136  · iOS >= 15.5
#0 2.136  · Kaios >= 2.5
#0 2.136  · Opera Mini >= all
#0 2.136  · Opera Mobile >= 64
#0 2.136  · Opera >= 86
#0 2.136  · Safari >= 15.5
#0 2.136  · Samsung >= 16.0
#0 2.136 
#0 2.371  App · Running "@quasar/testing" Quasar App Extension...
#0 2.373  App · Running "@quasar/testing-unit-jest" Quasar App Extension...
[... 75 lines cut ...]
#0 23.50 👎 Program npx exited with error code: 1
#0 23.50 1
#0 23.51 Traceback (most recent call last):
#0 23.51   File "/root/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 297, in <module>
#0 23.51 Searching for zcl file from /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 23.51     main()
#0 23.51   File "/root/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 280, in main
#0 23.51     runGeneration(cmdLineArgs.zapFile, cmdLineArgs.zclFile, cmdLineArgs.templateFile, cmdLineArgs.outputDir)
#0 23.51   File "/root/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 177, in runGeneration
#0 23.51     subprocess.check_call(generate_cmd + ['-z', zcl_file, '-g', templates_file,
#0 23.51   File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
#0 23.51     raise CalledProcessError(retcode, cmd)
[... cut ...] ted']' returned non-zero exit status 1.
------
Dockerfile:269
--------------------
 267 |         && npm ci
 268 |     ENV ZAP_DEVELOPMENT_PATH=/opt/zap-${ZAP_VERSION}
 269 | >>> RUN scripts/examples/gn_build_test_example.sh app1
 270 |     
 271 |     RUN source scripts/activate.sh && scripts/build_python.sh -m platform -d true -i no
--------------------
ERROR: failed to solve: process "/bin/bash -c scripts/examples/gn_build_test_example.sh app1" did not complete successfully: exit code: 1

```

ReneJosefsen, 2023-01-05:
It might be a stupid question, but have you tried deleting the .zap folder?

PR #24307 (2023-01-06): Update zap image version for cert bins docker file
Fixes #24158 

Old ZAP versions seem to give 'sqlite error: no transaction active' when running against new templates/javascript (unclear which because there is no stack trace or more information). Updating to newer zap should fix the error.