project-chip/connectedhomeip #24167: [BUG] [nrfconnect] [nRF52840]

mmt-10, 2022-12-21:
### Reproduction steps

1. flash `7.3.0SoftDevice S140` at first.
```bash
pyocd erase -t nrf5280 --chip
pyocd flash -t nrf52840 s140_nrf52_7.3.0_softdevice.hex
```
[https://www.nordicsemi.com/Products/Development-software/s140/download](https://www.nordicsemi.com/Products/Development-software/s140/download)

3. build lighting-app
```bash
west build -b nrf52840dongle_nrf52840 -- -DCONF_FILE=prj_no_dfu.conf
```

4. flash merge.hex
```bash
west flash -r pyocd
```

5. then, error occurs below code.
`err` is not `CHIP_NO_ERROR`. 

[https://github.com/project-chip/connectedhomeip/blob/621c69cd95c4cc1615f5e4a79340579e6751a13e/examples/lighting-app/nrfconnect/main/AppTask.cpp#L122](https://github.com/project-chip/connectedhomeip/blob/621c69cd95c4cc1615f5e4a79340579e6751a13e/examples/lighting-app/nrfconnect/main/AppTask.cpp#L122)

how do I resolve it?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 7f2b0a5920d1a7acc202c2932ac4c626c54cd31e (HEAD -> v1.0-branch, origin/v1.0-branch)

### Platform

nrf

### Anything else?

### Environment

OS : ubuntu 20.04

~/ncs/nrf$ git log -1
commit 053756565898659278f957e5f58000cc0491a88d (HEAD, tag: v2.0.2)

~/ncs/zephyr$ git log -1
commit be360e7db2db7e1f0e9fcac686a08e62e350bf55 (HEAD, tag: v3.0.99-ncs1-1, manifest-rev)

arm toolchain
gcc-arm-none-eabi-10.3-2021.10

mmt-10, 2022-12-22:
Hi,

I check more detail.

```c
printk("before\n");
err = ConnectivityMgr().SetThreadDeviceType(ConnectivityManager::kThreadDeviceType_Router); 
printk("after\n");
```
I can see `before`, but I cannot see `after`.

It may be stacked in `ConnectivityMgr().SetThreadDeviceType()`

Damian-Nordic, 2022-12-22:
Why do you flash soft device? Soft device controller had to be flashed separately back in time when Nordic used a bare metal SDK. Now that we switched to nRF Connect SDK, you don't need to do it. The second thing is that you selected nRF52840 Dongle - is that correct or you use nRF52840 **DK**?

mmt-10, 2022-12-26:
@Damian-Nordic 
Thank you for your reply.

> Why do you flash soft device?
nRF52840 DK / Dongle has not obtained the certificate of the radio law of our country.
So, I select Seeed XIAO nRF52840.
[https://www.seeedstudio.com/Seeed-XIAO-BLE-nRF52840-p-5201.html](https://www.seeedstudio.com/Seeed-XIAO-BLE-nRF52840-p-5201.html)

Seeed XIAO nRF52840 has their original bootloader.
I search bootloader, then I find softdevice.

I tried below command.

```bash
pyocd erase -t nrf52840 --chip 
west flash -r pyocd
```

then, below message showed.
And, nrf52840 seems it cannot boot normally.
Does merged.hex include bootloader?

> I Loading merged.hex at 0x00001000 [load_cmd]
> W Chip erase used when flash address 0x1000 is not the same as flash start 0x0 [builder]

> The second thing is that you selected nRF52840 Dongle - is that correct or you use nRF52840 DK?

I tried nRF52840DK option with modified .overlay and prj_no_dfu.conf.
it could run.

mmt-10, 2023-01-16:
I modified pin assignment and etc on nrf52840dk_nrf52840.overlay for XIAO_BLE.
I solved.

closed 2023-01-16 by mmt-10