project-chip/connectedhomeip #24170: [Spec] Thermostat cluster datamodel (XML) use anonymous bitmap and enum types in attributes

jsloth, 2022-12-21:
The data-model (XML) for [Thermostat](https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml) use anonymous BITMAP and ENUM types in attributes.

The various types that should be used are defined in https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/silabs/types.xml 

This issue is similar to https://github.com/project-chip/connectedhomeip/issues/24075 and https://github.com/project-chip/connectedhomeip/pull/23656

## Bitmaps: 

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L30

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L38

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L77

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L106

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L111

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L154

## Enums:

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L87

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L92

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L112

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L134

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L144

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L149

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L159

https://github.com/project-chip/connectedhomeip/blob/e1017305e3b0f51048ab7362c52dc6b851789f38/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L165

bzbarsky-apple, 2022-12-21:
Ideally this would get fixed in the spec, and then we can align the XML with the spec.

jsloth, 2022-12-21:
I think the spec is actually okay in this regard under each attribute or command using enum or bitmap its meaning is described. But the xml model doesn't include this. If you take a look e.g. at the Doorlock cluster xml, it is done correctly (in my humble opinion 😉).

bzbarsky-apple, 2022-12-22:
The spec is not OK, because the end goal here is to be able to generate the XML from the spec and vice versa without human intervention (which has led to a ton of issues with the two not matching).

So we want names for things in the spec if they have names in the XML.