project-chip/connectedhomeip #24177: [Scenes] Update Scene related XML

lpbeliveau-silabs, 2022-12-21:
Once Zap is able to handle quality field, update all scene related XML cluster to make use of it.

bzbarsky-apple, 2022-12-21:
What aspect of quality are you trying to encode here?

lpbeliveau-silabs, 2023-02-09:
The "S" that determines if an attribute can be added to ExtensionFieldSets.

bzbarsky-apple, 2023-02-09:
@lpbeliveau-silabs Ah, ok.  Is there a ZAP issue tracking the ZAP side support for this?

lpbeliveau-silabs, 2023-02-09:
@bzbarsky-apple This issue was opened a while ago: https://github.com/project-chip/zap/issues/174 I can check with Zap what is the progress on their side

bzbarsky-apple, 2023-02-09:
Ah.  The problem is that that issue is over-broad.  Access control bits are already supported, for example.

We need an issue specifically for the "S" quality.

lpbeliveau-silabs, 2023-03-27:
New [issue](https://github.com/project-chip/zap/issues/980) created specifically for the "S" quality.