project-chip/connectedhomeip #24180: How is a commissioner/controller supposed to (re)identify bridged device endpoints?

plan44, 2022-12-21:
When a commissioner/controller software (such as Homekit) commissions a bridge for the first time, it obviously needs to walk the partslists of the bridge node for bridged device endpoints, and examine their device types and capabilities (Matter Core Specs, 9.12.3).

While these bridged device endpointIDs can be considered stable for normal operation after commissioning (the standard mandates bridges to use ever  incrementing IDs for new bridged devices to get attached to the bridge, wrapping around only after reaching endpointId 0xFFFF, Core Specs, 9.2.1), IMHO they cannot be considered stable *forever*.

For example, when a bridge needs to be completely reset and re-commissioned, endpointIDs are likely to map to different bridged devices than before.

Commissioner/controller software might want to re-identify devices even across such re-installations for end user experience reasons (not loosing controller side device setup when a bridge needs to be reset).

What would be proper attributes to do so? I think that using endpointIDs, treating them as a permanently stable and unique ID for a bridged device even after the bridge was removed and re-comissioned is *not* a valid strategy. I'd assume attributes from `BridgedDeviceBasic` such as `SerialNumber` and `UniqueId` could be used, if present, as these should represent the bridged device (and not the bridge or its internal mapping mechanisms) and should be unique.

**Background for asking the above:** Apparently Homekit or Home.app caches device type (and name) even across complete deletion and re-setup of a "Home" based on nothing else but the endpointID, not looking at `SerialNumber` (which is a GUID in my test case, so would uniquely identify devices).

While the name can be corrected during the setup process, device *type* cannot, and so assuming it just from the fact that a certain type existed on a certain endpointId in the past, can lead to broken setups (not just theory, I have an endpoint that once hosted a plugin socket device which remains treated as such, although the bridge *and* the entire home were deleted and re-setup from scratch in between, and the device is now actually a light).

The particular behaviour might be just a Homekit bug (@nivi-apple is it?), but I'm generally interested to learn how bridged device's identities are *meant* to be established and used over time, and within what limits. The specs are not very conclusive on this.

bndkk, 2023-01-09:
This would interest me too, as I currently also have a problem with the home app caching device types.

bndkk, 2023-01-12:
@bzbarsky-apple @nivi-apple we would appreciate some insight here.

nivi-apple, 2023-01-12:
Looking into this. Will post an update shortly.

nivi-apple, 2023-01-12:
@plan44 @bndkk I checked with our HomeKit folks. This is by design even for non bridged accessories we do not want to support this feature due to some privacy concerns. So I wouldn't say this is a bug but it's intentional. Although I see the use case can enhance the user friendly experience to a degree, its not supported currently and we are not planning to support at the moment.

nivi-apple, 2023-01-12:
@plan44 and @bndkk For this issue we would recommend you to please file a feedback assistant ticket to proceed with this issue. In the future for all HomeKit/Home App issues please feel free to directly create a feedback assistant ticket so you can get the help faster. Here's the instructions to file the same - 

Feedback assistant ticket:

File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
Choose iOS/iPadOS to file an issue
Choose the area as Home App and fill in the details of the type of issue and the description
Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
iOS:
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
tvOS:
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.
Apple DeveloperApple Developer
Bug Reporting - Apple Developer
Now with Feedback Assistant available on iPhone, iPad, Mac, and the web, it's easier to submit effective bug reports and request enhancements to APIs and tools. (71 kB)
https://developer.apple.com/bug-reporting/#feedback-assistant
developer.apple.comdeveloper.apple.com
Profiles and Logs - Bug Reporting - Apple Developer
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.
developer.apple.comdeveloper.apple.com
Profiles and Logs - Bug Reporting - Apple Developer
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.
developer.apple.comdeveloper.apple.com
Profiles and Logs - Bug Reporting - Apple Developer
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.

Thanks,
Nivi.

plan44, 2023-01-13:
Hi @nivi-apple, thanks for the information.

The fact that there might be an issue with Homekit here is one thing, my main question however was matter related: **[...] I'm generally interested to learn how bridged device's identities are *meant* to be established and used over time, and within what limits. The specs are not very conclusive on this.**

Regarding Feedback Assistant: I'm well aware of it (I'm an Apple developer since decades, long back in the old Radar days). I have filed many issues there, but while I'm confident these were all processed somehow and might have contributed to solutions eventually, for a submitter FA is pretty much a black hole :-(

In this particular case it would be hard to even write a bug report before it is clear how bridged device's identities are *meant* to be established and persisted in matter. It might be a Apple bug, but it might also be a detail missing in the implementation of our bridges that prevents proper identification at this time. At least this part of the discussion should not be buried in FA 😉 but belongs here.

woody-apple, 2023-06-23:
@bzbarsky-apple any thoughts to the Matter side of things?

@plan44 Please note the FBA number, and we can follow up, however this is not an Apple support forum, so we can't help with any Apple specific questions here.

bzbarsky-apple, 2023-06-23:
> @bzbarsky-apple any thoughts to the Matter side of things?

This whole issue is about things that are outside Matter proper (i.e. preserving device identity across factory reset).

closed 2023-06-23 by bzbarsky-apple

plan44, 2023-06-25:
> > @bzbarsky-apple any thoughts to the Matter side of things?
> 
> This whole issue is about things that are outside Matter proper (i.e. preserving device identity across factory reset).

The mechanisms to do so are clearly outside Matter, agreed.

However I questioned the validity of an assumption one of these mechanisms (Apple's) seem to have taken (at that time - no longer judging from more recent tests):

> @plan44 wrote
> I think that using endpointIDs, treating them as a permanently stable and unique ID for a bridged device even after the bridge was removed and re-comissioned is not a valid strategy.

Correct?

**This is a matter-side question and it is important to clarify this**, for implementors of dynamic bridges as well as commissioning software.

(It just does not arise with most non-bridge matter devices, as those have a fixed endpoint layout established from the ZAP config at build time and thus do preserve endpoint ids across factory resets)

plan44, 2023-06-25:
@woody-apple

> @plan44 Please note the FBA number, and we can follow up, however this is not an Apple support forum, 
so we can't help with any Apple specific questions here.

I did not post a FBA issue, because the question was indeed a matter side one, with HomeKit behaviour only as an example. As it seems (judging from my more recent testing), the Apple implementation has changed and does not assume endpoint IDs stable over factory resets any more, so no need for an FBA issue.