project-chip/connectedhomeip #24184: [BUG] Changes do not reflect to Apple Home

mustafayuce33, 2022-12-22:
### Reproduction steps

I've used [this PR](https://github.com/project-chip/connectedhomeip/pull/23840) for handling commands. Everything worked fine but when I write the attribute I do not see any changes on Apple home. For instance when a switch is turned off on our side I send this change to Apple Home by writing attribute value like here [connectedhomeip/examples/dynamic-bridge-app/linux/main.cpp](https://github.com/project-chip/connectedhomeip/blob/52d72b12fb91d58ad591d9edcb99f6c123e09c74/examples/dynamic-bridge-app/linux/main.cpp#L458)
But I was expecting that change is reflected to Apple Home but no I don't see any change for the device value on Apple Home
Do I need to do anything except writing attribute like triggering Apple home which will tell it there are updates for device values/attributes. This is also a question for bridge-app. 

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/commit/74296b895ece6b6f8802dfb7fe34cde2d30f5604

### Platform

raspi

bzbarsky-apple, 2022-12-22:
This sounds like an issue with the dynamic bridge app, not with Apple Home....

mustafayuce33, 2022-12-23:
So which function do we use to inform Apple home to get the attribute changes? Is there any other example that I can see?

mustafayuce33, 2022-12-23:
> This sounds like an issue with the dynamic bridge app, not with Apple Home....

So which function do we use to inform Apple home to get the attribute changes? Is there any other example that I can see?

mustafayuce33, 2022-12-23:
I added `ScheduleReportingCallback` from `bridge-app` and that resolved the issue

```
namespace {
    void CallReportingCallback(intptr_t closure)
    {
        auto path = reinterpret_cast<app::ConcreteAttributePath *>(closure);
        MatterReportingAttributeChangeCallback(*path);
        Platform::Delete(path);
    }

    void ScheduleReportingCallback(Device * dev, ClusterId cluster, AttributeId attribute)
    {
        ChipLogProgress(DeviceLayer, "Report attribute change %d - %d - %d", dev->GetEndpointId(), cluster, attribute);

        auto * path = Platform::New<app::ConcreteAttributePath>(dev->GetEndpointId(), cluster, attribute);
        DeviceLayer::PlatformMgr().ScheduleWork(CallReportingCallback, reinterpret_cast<intptr_t>(path));
    }
}
```

closed 2022-12-23 by mustafayuce33