project-chip/connectedhomeip #24192: [BUG] Various compilation errors when implementing custom bridge device

raven-worx, 2022-12-22:
### Reproduction steps

Hi,

i am currently implementing a Bridge device (Linux based). I used the dynamic-bridge-app as a base example and struggling with the following build issues:

1. `scripts/idl/generators/bridge/BridgeClustersCpp.jinja` contains hard coded include path to example: `#include "third_party/connectedhomeip/examples/dynamic-bridge-app/linux/include/GeneratedClusters.h"`
    i could fix it by removing this line and ensuring my GeneratedClusters.h is included before.
2. The file (generated by "**chip_codegen("chip-bridge-codegen")**" target) `bridge/ModeSelect.h` fails with the following error:
    `bridge/ModeSelect.h:34:17: error: ‘SemanticTag’ was not declared in this scope; did you mean ‘chip::app::Clusters::ModeSelect::Structs::SemanticTag’?
	SemanticTag is defined right after the failing struct. Moving it before it remedies this error`
	The SemanticTag struct is defined after the usage which causes this error. Moving the definition up remedies at least this error.
3. `data-model/DataModel.h` (from the dynamic-bridge-app example) cant handle attribute definitions like `Attribute<std::string>` and fails at compilation time ( error at missing `emplace_back()` function). So its not possible to add BasicInformation cluster for example.
4. Also in generated file for ModeSelect cluster: `bridge/ModeSelect.h:40:19: error: no match for ‘operator=’ (operand types are ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’} and ‘chip::CharSpan’ {aka ‘chip::Span<const char>’})
   40 |         label = t.label;`

### Bug prevalence

always

### GitHub hash of the SDK that was being used

74296b895ece6b6f8802dfb7fe34cde2d30f5604

### Platform

other

### Platform Version(s)

1.0.0.1

bzbarsky-apple, 2022-12-22:
@andy31415

raven-worx, 2023-01-17:
Hello, are there any commitments yet?
We would need the support for using the BasicInformation and ModeSelect clusters specially.

raven-worx, 2023-01-27:
ad1) is fixed with this commit: https://github.com/project-chip/connectedhomeip/commit/8afa552309bbc86c105338118a16d5368d03c0b7

ad 3) i added a workaround in my project by adding template specializations for std::string (see attached code snippet). Might be a bit ugly but works so far for me. Looking forward for the official solution.
[string_attribute_snippet.txt](https://github.com/project-chip/connectedhomeip/files/10517339/string_attribute_snippet.txt)

bzbarsky-apple, 2023-08-07:
dynamic-bridge is gone.

closed 2023-08-07 by bzbarsky-apple