project-chip/connectedhomeip #24202: [BUG] "pw_env_setup_CIPD_PIGWEED" - "Undefined identifier"

vindicatorr, 2022-12-23:
### Reproduction steps

To get by the python10 issues (specifically wheel issue for gevent for gdbgui (I may not have needed to make the version changes for other stuff like pyparsing, but I just wiped out the "maximums" just in case)):
```
diff --git a/scripts/requirements.esp32.txt b/scripts/requirements.esp32.txt
index 4bf5479..436b2d9 100644
--- a/scripts/requirements.esp32.txt
+++ b/scripts/requirements.esp32.txt
@@ -1,12 +1,12 @@
 click>=7.0 ; platform_machine != 'aarch64' and sys_platform == 'linux'
 future>=0.15.2 ; platform_machine != 'aarch64' and sys_platform == 'linux'
-pyparsing>=2.0.3,<2.4.0 ; platform_machine != 'aarch64' and sys_platform == 'linux'
+pyparsing>=2.0.3 ; platform_machine != 'aarch64' and sys_platform == 'linux'
 idf-component-manager>=0.2.99-beta ; platform_machine != 'aarch64' and sys_platform == 'linux'
-gdbgui==0.13.2.0 ; platform_machine != 'aarch64' and sys_platform == 'linux'
-pygdbmi<=0.9.0.2 ; platform_machine != 'aarch64' and sys_platform == 'linux'
-reedsolo>=1.5.3,<=1.5.4 ; platform_machine != 'aarch64' and sys_platform == 'linux'
+gdbgui==0.15.1.0 ; platform_machine != 'aarch64' and sys_platform == 'linux'
+pygdbmi>=0.9.0.2 ; platform_machine != 'aarch64' and sys_platform == 'linux'
+reedsolo>=1.5.3 ; platform_machine != 'aarch64' and sys_platform == 'linux'
 bitstring>=3.1.6 ; platform_machine != 'aarch64' and sys_platform == 'linux'
 ecdsa>=0.16.0 ; platform_machine != 'aarch64' and sys_platform == 'linux'
-kconfiglib==13.7.1 ; platform_machine != 'aarch64' and sys_platform == 'linux'
-construct==2.10.54 ; platform_machine != 'aarch64' and sys_platform == 'linux'
-python-socketio<5 ; platform_machine != 'aarch64' and sys_platform == 'linux'
+kconfiglib>=13.7.1 ; platform_machine != 'aarch64' and sys_platform == 'linux'
+construct>=2.10.54 ; platform_machine != 'aarch64' and sys_platform == 'linux'
+python-socketio==5.4.0 ; platform_machine != 'aarch64' and sys_platform == 'linux'
```
To only get the pigweed submodule, and activate without cipd (`connectedhomeip_env` is outside of `connectedhomeip` in case I need to clean/reset):
```
$ git submodule sync --recursive
$ git submodule update --depth 1 ./third_party/pigweed
$ PW_CONFIG_FILE="scripts/environment_no_cipd.json" \
PW_ENVIRONMENT_ROOT="/<pathTo>/connectedhomeip_env" \
. ./scripts/activate.sh
```
Result:
```
...
  Setting up CIPD package manager...skipped (0.1s)
  Setting up Python environment.....[\]
ERROR at //third_party/pigweed/repo/pw_toolchain/clang_tools.gni:24:19: Undefined identifier
      rebase_path(pw_env_setup_CIPD_PIGWEED + "/bin/", root_build_dir)
                  ^------------------------
See //third_party/pigweed/repo/pw_toolchain/host_clang/toolchains.gni:17:1: whence it was imported.
import("$dir_pw_toolchain/clang_tools.gni")
^-----------------------------------------
See //third_party/pigweed/repo/pw_unit_test/test.gni:20:1: whence it was imported.
import("$dir_pw_toolchain/host_clang/toolchains.gni")
^---------------------------------------------------
See //third_party/pigweed/repo/pw_env_setup/BUILD.gn:22:1: whence it was imported.
import("$dir_pw_unit_test/test.gni")
^----------------------------------
See //BUILD.gn:131:7: which caused the file to be included.
      "$dir_pw_env_setup:pip_install_pigweed_package",
      ^----------------------------------------------
['gn', 'gen', '/<pathTo>/connectedhomeip_env/gn_out', '--args=dir_pigweed="/<pathTo>/connectedhomeip/third_party/pigweed/repo"']

Traceback (most recent call last):
  File "/<pathTo>/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 335, in install_packages
    subprocess.check_call(
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gn', 'gen', '/<pathTo>/connectedhomeip_env/gn_out', '--args=dir_pigweed="/<pathTo>/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/<pathTo>/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 857, in <module>
    sys.exit(main())
  File "/<pathTo>/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 849, in main
    return EnvSetup(**vars(parse())).setup()
  File "/<pathTo>/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 501, in setup
    result = step(spin)
  File "/<pathTo>/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 661, in virtualenv
    if not virtualenv_setup.install(
  File "/<pathTo>/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 370, in install
    install_packages(gn_target)
  File "/<pathTo>/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 343, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['gn', 'gen', '/<pathTo>/connectedhomeip_env/gn_out', '--args=dir_pigweed="/<pathTo>/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.
```

### Bug prevalence

Now And Always (I'm just starting out here)

### GitHub hash of the SDK that was being used

Where's the hash? It's the master I cloned yesterday.

### Platform

other

### Platform Version(s)

Linux? (N/A)

### Anything else?

I see `/<pathTo>/connectedhomeip/build_overrides/pigweed_environment.gni` isn't supposed to be touched, so I didn't, but it's essentially empty (other than that notice).

My main beef with projects like this (including Zephyr), is the insane amount/size of modules that are just automatically downloaded. I had gotten to 2.2GB of data on disk from this project alone before I stopped and cleared out the submodules and started seeing what was actually needed (pigweed for sure).

I only have desire to download the absolute minimum, then build out from there, picking/choosing what I actually need.

I've been doing that with Zephyr, where I manually `git clone` modules I'll use like the STM32 repo containing the HAL, ESP-IDF, and NRFConnect, but just what's needed there that I clone from elsewhere (like nrfx, and sdk-nrfxlib), and then just link to expected directories.
It's a mess, but at least it isn't a waste.

bzbarsky-apple, 2023-01-03:
@vindicatorr Please see the `scripts/checkout_submodules.py` for checking out only the submodules relevant to a given platform.

vindicatorr, 2023-01-04:
I took a quick look at that and `.gitmodules`.
I edited the script to just print the git `cmd` it would run and found a diff of what I had to clone to build chip-tool for `linux` vs what would be cloned...
Not cloned:
```
nlfaultinjection //though I think I had to clone that for the android chip-tool and for esp32 apps
mbedtls
QRCode
openthread
ot-br-posix
cirque
nanopb
pybind11
editline
boringssl
```
Now that's not to say that I won't be needing any of those as I progress further, and there are things that look like I'll want like `boringssl`/`mbedtls` (probably `openthread` if I am going to comm with Thread devices), but I'll see when the time comes.
Oh wait! Regarding `mbedtls`, that reminded me that was being used for other projects like Zephyr, and I had cloned that elsewhere.
So again, like I'd do with those other projects, I just manually clone `mbedtls` elsewhere and then link to where it's expected to be, or even better, if the project allows the use for an environment variable like `ZEPHYR_MODULES="<...>"`.
Perhaps this would be a "feature request"?

I'm fine with just getting what I need by going by the errors I come across.
I actually did just that when setting up my `tasks.json` in vscode for an ST project. Eliminating every file to be built, then finding the library/linker/script file for which the error is pertinent, and add that to the build task (along with a note for myself to justify it).
It's a way for me to learn.