project-chip/connectedhomeip #24203: [BUG] Bridge example over Wi-Fi (CHIPoBLE disabled) seems to be broken

thecashit, 2022-12-24:
### Reproduction steps

Hello and thank you for this project.

When trying to get the bridge device example to work over a local network (only over WiFi), I ran into the following problems:

1) If you disable the CHIPoBLE option in the menuconfig, the project is not built.

<details>
  <summary>Compiller output</summary>

  ```
`
matter/examples/bridge-app/esp32/third_party/connectedhomeip/examples/platform/esp32/common/CommonDeviceCallbacks.cpp
In file included from matter/src/lib/dnssd/Resolver.h:31,
                 from matter/src/lib/dnssd/TxtFields.h:21,
                 from matter/src/lib/dnssd/Advertiser.h:26,
                 from matter/src/app/server/CommissioningModeProvider.h:20,
                 from matter/src/app/server/Dnssd.h:20,
                 from matter/examples/bridge-app/esp32/third_party/connectedhomeip/examples/platform/esp32/common/CommonDeviceCallbacks.cpp:35:
matter/src/lib/dnssd/Constants.h:56:60: error: macro "max" requires 2 arguments, but only 1 given
 constexpr size_t kSubTypeMaxLength      = std::max(SUBTYPES);
                                                            ^
In file included from matter/src/lib/dnssd/Resolver.h:31,
                 from matter/src/lib/dnssd/TxtFields.h:21,
                 from matter/src/lib/dnssd/Advertiser.h:26,
                 from matter/src/app/server/CommissioningModeProvider.h:20,
                 from matter/src/app/server/Dnssd.h:20,
                 from matter/examples/bridge-app/esp32/third_party/connectedhomeip/examples/platform/esp32/common/CommonDeviceCallbacks.cpp:35:
matter/src/lib/dnssd/Constants.h:76:60: error: macro "max" requires 2 arguments, but only 1 given
 constexpr size_t kSubTypeMaxLength      = std::max(SUBTYPES);
                                                            ^
In file included from matter/src/app/util/af.h:43,
                 from matter/src/app/util/util.h:53,
                 from matter/examples/bridge-app/esp32/third_party/connectedhomeip/examples/platform/esp32/common/CommonDeviceCallbacks.cpp:36:
matter/src/lib/support/SafeInt.h:69:58: error: macro "max" requires 2 arguments, but only 1 given
         if (static_cast<intmax_t>(numeric_limits<V>::max()) <= static_cast<intmax_t>(numeric_limits<T>::max()) &&
                                                          ^
matter/src/lib/support/SafeInt.h:69:109: error: macro "max" requires 2 arguments, but only 1 given
         if (static_cast<intmax_t>(numeric_limits<V>::max()) <= static_cast<intmax_t>(numeric_limits<T>::max()) &&
                                                                                                             ^
matter/src/lib/support/SafeInt.h:70:58: error: macro "min" requires 2 arguments, but only 1 given
             static_cast<intmax_t>(numeric_limits<V>::min()) >= static_cast<intmax_t>(numeric_limits<T>::min()))
                                                          ^
matter/src/lib/support/SafeInt.h:70:109: error: macro "min" requires 2 arguments, but only 1 given
             static_cast<intmax_t>(numeric_limits<V>::min()) >= static_cast<intmax_t>(numeric_limits<T>::min()))
                                                                                                             ^
matter/src/lib/support/SafeInt.h:77:61: error: macro "min" requires 2 arguments, but only 1 given
         return static_cast<intmax_t>(numeric_limits<T>::min()) <= static_cast<intmax_t>(arg) &&
                                                             ^
matter/src/lib/support/SafeInt.h:78:88: error: macro "max" requires 2 arguments, but only 1 given
             static_cast<intmax_t>(arg) <= static_cast<intmax_t>(numeric_limits<T>::max());
                                                                                        ^
matter/src/lib/support/SafeInt.h:83:59: error: macro "max" requires 2 arguments, but only 1 given
         if (static_cast<uintmax_t>(numeric_limits<V>::max()) <= static_cast<uintmax_t>(numeric_limits<T>::max()))
                                                           ^
matter/src/lib/support/SafeInt.h:83:111: error: macro "max" requires 2 arguments, but only 1 given
         if (static_cast<uintmax_t>(numeric_limits<V>::max()) <= static_cast<uintmax_t>(numeric_limits<T>::max()))
                                                                                                               ^
matter/src/lib/support/SafeInt.h:90:93: error: macro "max" requires 2 arguments, but only 1 given
         return static_cast<uintmax_t>(arg) <= static_cast<uintmax_t>(numeric_limits<T>::max());
                                                                                             ^
matter/src/lib/support/SafeInt.h:95:46: error: macro "max" requires 2 arguments, but only 1 given
         static_assert(numeric_limits<T>::max() >= 0, "What weird type is this?");
                                              ^
matter/src/lib/support/SafeInt.h:96:59: error: macro "max" requires 2 arguments, but only 1 given
         if (static_cast<uintmax_t>(numeric_limits<V>::max()) <= static_cast<uintmax_t>(numeric_limits<T>::max()))
                                                           ^
matter/src/lib/support/SafeInt.h:96:111: error: macro "max" requires 2 arguments, but only 1 given
         if (static_cast<uintmax_t>(numeric_limits<V>::max()) <= static_cast<uintmax_t>(numeric_limits<T>::max()))
[... 108 lines cut ...]

2) With disabled CHIPoBLE i can specify the SSID and password of the access point in menuconfig, the device connects and 4 lamps appear in the Home.app app after entering 2020-2021 pin code, but all of them are displayed with the status "unavailable" and when you tap on them, the information does not go into the firmware (there are no changes in the log and so further). That is, devices are created, but they cannot be controlled.
I tried on several networks with or without Home Hub but the resultant is the same.

iOS: 16.2
Home Hub: Homepod mini with 16.2

Both of these points look like a bug, but, unfortunately, I don't know the source code well enough to figure it out and make a PR. 
Could you pls suggest how to fix this?

### Bug prevalence

Every time i try to control bridged device

### GitHub hash of the SDK that was being used

V1.0.0.1

### Platform

esp32

### Platform Version(s)

4.4.2

bzbarsky-apple, 2023-01-03:
@dhrishi @shubhamdp

dhrishi, 2023-01-06:
Thanks for the tag @bzbarsky-apple 

@CrowdedFuzzball We will check this

PSONALl, 2023-01-06:
Hi @CrowdedFuzzball The above error occurs because the nimble related code gets compiled even when CHIPoBLE option is disabled 
Added a fix here: https://github.com/project-chip/connectedhomeip/pull/24303

closed 2023-01-07 by andy31415

CrowdedFuzzball, 2023-01-08:
Great, it works now!
Thank you!

PR #24303 (2023-01-07): [ESP32] Fix compilation errors with bridge app
- Fixes #24203 
- Currently we have all the nimble related code under CHIPoBLE config option 
- Same thing should be followed for examples/platform/esp32/common/CommonDeviceCallbacks.cpp 

### Testing:
 - Compilation works after disabling CHIPoBLE config option and without disabling CHIPoBLE config option too
 - Tested commissioning and cluster control using chip-tool