project-chip/connectedhomeip #24210: [BUG] Identify device names on Apple home in dynamic-bridge

mustafayuce33, 2022-12-27:
### Reproduction steps

1. I use ```dynamic-bridge``` for publishing our devices to Apple Home. 
2. Every device name starts with ```Matter Accessory <number>```. 
3. Can I identify the name of the device on Apple Home rather than ```Matter Accessory <number>```. 

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/commit/b9d32ecefd004adb7cc4c70a5f4a567dca199da0

### Platform

raspi

mustafayuce33, 2023-01-03:
Can anyone reply back to this issue?

bzbarsky-apple, 2023-01-03:
For issues with Apple Home, please file a Feedback Assistant ticket.

- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.
Apple DeveloperApple Developer
[Bug Reporting - Apple Developer](https://developer.apple.com/bug-reporting/#feedback-assistant)
Now with Feedback Assistant available on iPhone, iPad, Mac, and the web, it’s easier to submit effective bug reports and request enhancements to APIs and tools. (71 kB)
https://developer.apple.com/bug-reporting/#feedback-assistant

mustafayuce33, 2023-01-03:
@bzbarsky-apple I am curious about I should send it to Apple side. Because it is the same question for Google Home. I want to know how can I set the device names on matter ```dynamic-bridge-app```?

bzbarsky-apple, 2023-01-03:
@achaulk-goog for dynamic-bridge issues.

mustafayuce33, 2023-01-05:
@achaulk-goog Did you implement naming devices on Apple home, Google Home, etc in ```dynamic-bridge-app```

mustafayuce33, 2023-01-06:
@bzbarsky-apple @achaulk-goog Any update?

mustafayuce33, 2023-01-17:
It took 2 weeks and no answer. I did it on my own with digging in to the code

closed 2023-01-17 by mustafayuce33

raven-worx, 2023-01-17:
> It took 2 weeks and no answer. I did it on my own with digging in to the code

What was your outcome/solution?

lboue, 2023-04-15:
@mustafayuce33
Can you tell us where to look?