project-chip/connectedhomeip #24220: [Feature] Add a way to stop the stack properly

caipiblack, 2022-12-30:
### Feature description

We are working on a matter controller device.

We need to stop the Matter stack properly to test our code with Valgrind to make sure that there is no memory leak. 

Currently it's difficult because the Matter stack is not able to be stopped properly.

It should be good to implement something that permit to cleanup all the memory used by the stack when we stop the stack.

Also, what is the good way to stop the stack ?

```
    // chip::DeviceLayer::PlatformMgr().HandleServerShuttingDown();
    // DeviceControllerFactory::GetInstance().ReleaseSystemState();
    // DeviceControllerFactory::GetInstance().Shutdown();
    // ReturnLogErrorOnFailure(chip::DeviceLayer::PlatformMgr().StopEventLoopTask());

    // chip::Platform::MemoryShutdown();
    chip::DeviceLayer::PlatformMgr().StopEventLoopTask();
    // chip::DeviceLayer::PlatformMgr().Shutdown();

    // Do this after StopEventLoopTask() so we don't need to Lock/Unlock the stack
    for (auto it = mCommissioners.begin(); it != mCommissioners.end(); it++) {
        ShutdownCommissioner(it->first);
    }
```

### Platform

other

caipiblack, 2023-01-02:
Our software is based on chip-tool, so all of my remarks can be tested with chip-tool and valgrind.

Currently I found that most of the uncleared memory is related to DBUS, maybe we initialise some pointers that are never cleared.

Disabling `chip_enable_wifi=false` helps, but I still have some leaks:

```
==16042== LEAK SUMMARY:
==16042==    definitely lost: 0 bytes in 0 blocks
==16042==    indirectly lost: 0 bytes in 0 blocks
==16042==      possibly lost: 1,352 bytes in 18 blocks
==16042==    still reachable: 35,508 bytes in 214 blocks
==16042==                       of which reachable via heuristic:
==16042==                         newarray           : 1,536 bytes in 16 blocks
==16042==         suppressed: 0 bytes in 0 blocks
==16042== Reachable blocks (those to which a pointer was found) are not shown.
==16042== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==16042== 
==16042== For lists of detected and suppressed errors, rerun with: -s
==16042== ERROR SUMMARY: 18 errors from 18 contexts (suppressed: 0 from 0)
```

bzbarsky-apple, 2023-01-03:
Stopping the stack properly on the controller side consists of:

1. Stop the event loop task.
2. Shut down all outstanding CHIPDeviceController instances.
3. Shut down the controller factory.

This will shut down the platform manager, etc, etc.

It sounds like you are running into a specific bug in the Linux bits where some things don't get shut down properly.  If so, that should be fixed in the Linux platform impl.

closed 2023-01-03 by bzbarsky-apple