project-chip/connectedhomeip #24221: [Workflow] use optional submodules and needlessly building everything

jahess, 2022-12-30:
### Workflow issues

Yes, I am a newbie.  But, wow the default matter clone is huge!  The build everything default takes equally long.

For a release, yes, everything needs to be built and tested.

But, this just wastes time for a developer focused on just one of the hardware vendors.

What about making the relevant submodules optional?  Yes, this would need some build system support and indication from the developer for which vendor hardware was needed.

But, wow, a 13GB all inclusive clone!  If this was reduced with optional submodules, it would seriously improve workflow times.  It would also reduce the active code set to be compiled for a developer which in turn might reduce the number of development platform issues they experience.

### Platform

all

Damian-Nordic, 2023-01-03:
You don't have to checkout all the submodules. You can use `scripts/checkout_submodules.py` to checkout only relevant ones.

jahess, 2023-01-03:
For a newbie getting started...

    https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md

Instructs:

    git submodule update --init

And then later...

    source scripts/activate.sh

Both result in cloning all submodules.

Looking at the .gitmodules, there is no appearance of "update = none" that would result from the use of optional submodules.  While not the best intro, a starting explanation would be:

    https://stackoverflow.com/questions/49110958/how-to-add-optional-submodules-in-git

I did see use of "platforms =  something" which I have never seen before.  I don't know what this does -- it doesn't appear in:

    https://git-scm.com/book/en/v2/Git-Tools-Submodules

jahess, 2023-01-04:
I'm only suggesting that minimal defaults for getting started would ease more developers into Matter.

I also kindly suggest that use of the "update" attribute would have at least a few more developers already familiar with it.  Yes, other infrastructure scripts like checkout_submodules.py would need updates.  Past that, I don't pretend to know what all else would need to be updated.

jahess, 2023-01-04:
Using optional submodules (the update attribute) means

    git submodule update --init

Would, by default, not clone those optional submodules.  Then modified checkout_submodules.py could be used to specify which additional submodules to work with.

closed 2023-08-11 by stale[bot]