project-chip/connectedhomeip #24223: [Feature] Windows support

dotMorten, 2022-12-31:
### Feature description

Will there ever be a Windows build of the client libraries, so we can build applications that can control/monitor matter devices from the most widely used operating system out there?
It doesn't seem like there's any focus on supporting Windows outputs here, or even building on Windows (using the subsystem doesn't count).

### Platform

windows

dotMorten, 2023-07-11:
Still an issue. I still want to control my devices from the leading desktop platform

zzz2016, 2023-08-10:
same question

TheWalkingDead1024, 2024-01-31:
same question

tomxue, 2025-01-16:
Same question. Is there any update?

richiemsft, 2026-09-15:
Give this a try: 
I forked the repo https://github.com/richiemsft/matterforwindows/tree/windows-port and started working on native windows support.
Details on how to build: https://github.com/richiemsft/matterforwindows/blob/windows-port/docs/guides/windows.md

richiemsft, 2026-09-16:
I now have a working native Windows implementation available for review and would like guidance on the best way to upstream it.

## Implementation

- Fork/branch: https://github.com/richiemsft/matterforwindows/tree/windows-port
- Build and usage guide: https://github.com/richiemsft/matterforwindows/blob/windows-port/docs/guides/windows.md
- Latest green Native Windows CI: https://github.com/richiemsft/matterforwindows/actions/runs/35032065163

The port builds directly on Windows 11 with MSVC, GN, Ninja, and PowerShell, without WSL or a POSIX compatibility layer. It currently includes:

- Native x64 and ARM64 builds
- WinSock System/Inet integration and native Windows DNS-SD
- Persistent storage, configuration, connectivity, and diagnostic platform implementations
- Bluetooth LE central/peripheral support for controller and commissionee roles
- `chip-tool.exe`, including commissioning, operational commands, interactive mode, WebSocket/YAML integration, and WinHTTP-based HTTPS requests
- `all-devices-app.exe` and an all-clusters application
- Native Python controller wheels
- Windows packaging, smoke tests, GoogleTests, Python integration tests, and CI

The linked CI run passed all four jobs: native x64, Python controller x64, native ARM64, and Python controller ARM64. Development also included real-device commissioning/discovery/interaction testing and an adversarial multi-axis review; all identified high- and medium-priority findings were addressed, and no exploitable security vulnerabilities were identified.

## Current limitations

This is presented as development-preview support, not a production support commitment. Notable limitations include:

- Windows 10, x86, MinGW, clang-cl, and MSBuild-only builds are not covered.
- BLE and external Thread interoperability need broader hardware coverage across x64 and ARM64.
- A complete physical-device certification matrix has not been run on every architecture and transport.
- CI ZIPs are unsigned validation artifacts.
- The preview does not yet promise ABI stability or long-term servicing.

## Proposed upstream approach

The integration branch is intentionally comprehensive (currently 87 commits and 320 changed files), but I do not propose submitting it as one pull request. Based on the repository's contribution guidance, I would like to split it into a dependency-ordered series along these lines:

1. MSVC build/toolchain foundation and narrowly required portable build fixes
2. Windows System Layer and Inet/WinSock support with focused tests
3. Windows platform services: storage, configuration, connectivity, diagnostics, and DNS-SD
4. Windows BLE transport and commissioning support
5. Controller, chip-tool, and Windows application integration
6. Native Python controller and Windows test-harness support
7. Packaging, CI, and documentation

Would SDK maintainers be open to this contribution, and could you advise on:

- preferred PR boundaries and ordering;
- any required design document or architecture review;
- expected ownership/maintenance commitments for the Windows platform;
- the appropriate maintainer and company reviewer groups to involve; and
- whether any shared/core changes should be submitted as separate prerequisite PRs?

I am happy to adjust the sequence before opening the first PR so that the work is reviewable and aligns with the SDK's maintenance expectations.

richiemsft, 2026-09-16:
@andy31415 Could you please review the implementation summary and proposed staged upstreaming approach above, and advise on architecture review, PR boundaries, ownership expectations, and reviewer routing? I do not have triage permission to apply labels; please also add the existing `windows` and `platform` labels to this issue.

dotMorten, 2026-09-16:
FYI I have a work in progress library that uses that fork to build a little a windows app for controlling devices. Still some work to do so more of a proof of concept: https://github.com/dotMorten/matter.winrt

andy31415, 2026-09-16:
My main worry is that the matter SDK is generally supported by CSA members so external contributions on a larger platform (assuming windows has significant differences from embedded/posix) may incur quite a bit of maintainance requirements which is hard when this is purely from non-csa members, as well as extra CI overhead (we probably want some compatibility checks, at a minimum compilation ability)

For our existing platforms supported, we have a list of direct responsible persons that can be reached through CSA channels as well for quick replies (i.e. CSA slack, that requires an elevated membership with the alliance). We need to determine how mandatory such requirements are and what could we offered in terms of the proposed windows support plan.

If we can make this long-term maintainable, we can look at having this part of the main SDK. Otherwise maybe staying on a fork is best, so you have control and have less friction on merging things (main repository requires a multi-company approval process for example)

andy31415, 2026-09-16:
@richiemsft could you provide some details to figure out:
  - scope of change - feels "large" because windwos does tend to behave differently, however if isolated enough maybe it is feasable. Build system may be the most tricy one.
  - how would windows platform maintainance be handled (upgrade between versions, platform specific failures that need resolution, bug routing for platform specific bugs, consultance when behavior differs among platforms, if we are unsure which platform introduces a regression if any ... i.e. we cannot say "this is the cause" but we need some platform specific expertise)

richiemsft, 2026-09-16:
Thanks, this concern makes sense. At this point, the primary gap is no longer whether native Windows support can be implemented and validated; it is whether that support can be maintained responsibly over the lifetime of the SDK.

The fork demonstrates that native x64 and ARM64 builds, controller functionality, applications, Python bindings, automated tests, and CI are technically feasible. However, merging the implementation would also create ongoing responsibilities around Windows/SDK version upgrades, CI regressions, platform-specific triage, security servicing, compatibility decisions, and reviews of future changes.

I do not want to overstate what an individual external contributor can guarantee. I can continue developing the fork, documenting the architecture, keeping the implementation isolated where possible, and helping investigate issues. That is different from providing the durable ownership, availability through CSA channels, organizational continuity, and backup maintainers expected of an officially supported SDK platform.

For full upstream Windows support, I agree that we likely need explicit organizational commitment from Microsoft—or another CSA member with the necessary Windows platform expertise—including named maintainers and a sustainable support model.

Until that ownership is established, keeping the fork as a development-preview implementation and proving ground may be the responsible approach. It can continue producing evidence, identifying upstream integration boundaries, and reducing the technical risk without prematurely transferring an unsupported maintenance obligation to the main SDK.

A useful next step may be to define the minimum ownership and serviceability requirements for accepting Windows upstream, then identify the appropriate Microsoft and CSA stakeholders who could evaluate that commitment. Once sponsorship exists, we can use the proposed staged PR series to make the technical review manageable.