project-chip/connectedhomeip #24245: [1.0]

mustafayuce33, 2023-01-03:
### Reproduction steps

I use ```dynamic-bridge-app``` and want to get a notification whenever Apple Home, Google Home or Alexa is connected to matter server. 

1. Commissioning is completed
2. Lets say I restart ```dynamic-bridge-app```
3. I want to know when Apple Home, Google Home or Alexa is connected to matter again

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/commit/b9d32ecefd004adb7cc4c70a5f4a567dca199da0

### Platform

raspi

### Type

Platform Issue

bzbarsky-apple, 2023-01-04:
What exactly do you mean by "connected"?  Subscription is established?  Something else?

mustafayuce33, 2023-01-04:
I want to know if my dynamic-bridge-app is commissioned to Apple Home, Google home, etc. Lets say my bridge is commissioned to commissioner (Apple Home, Google Home, etc). Then I restarted my dynamic-bridge-app again. Now the question is how do I know if my app is commissioned or not. 

After I delete my bridge from Apple Home, Google Home, Can matter library say that my app is not commissioned anymore?

bzbarsky-apple, 2023-01-04:
When you start it again, it's still commissioned, right?  As in, the fabric table still has the relevant entries.

> After I delete my bridge from Apple Home, Google Home, Can matter library say that my app is not commissioned anymore?

You can register with the fabric table to be notified when fabrics are removed...

mustafayuce33, 2023-01-05:
@bzbarsky-apple 
> When you start it again, it's still commissioned, right? As in, the fabric table still has the relevant entries.
Yes it is still commissioned. But how can I retrieve that information from Matter Library? Because you know client can delete bridge from Apple Home, Google Home, etc. So I should know that if they are still connected or not.

> You can register with the fabric table to be notified when fabrics are removed...
First of all what is fabric table and how can I register with that table to be notified? I guess that is the thing what I am looking for.

mustafayuce33, 2023-01-06:
@bzbarsky-apple  Thanks for clarifying the issue. One more question is that when I commissioned a network (Apple Home or Google Home) there seems 2 records in fabric table in which one label name is empty. If I delete bridge on network (Apple Home or Google Home), one of the records with index ID 1 is being deleted but the records with Index ID 2 still remains in fabric table. That record remains in fabric table until I delete all files (chip_config.ini, chip_kvs, etc.). Is this a bug? 

```
[1673001296.929660][5557:5557] CHIP:DL: Matter Fabric Table: Fabric index 0x1. Compressed Fabric Id 0x334619CD9C8BAD3E, FabricId 0000000000000001, NodeId 000000000D137F10, VendorId 0x1349 Label Test Matter, Initialized 1
[1673001296.929692][5557:5557] CHIP:DL: Matter Fabric Table: Fabric index 0x2. Compressed Fabric Id 0x33432B454C31534B, FabricId 0000000000000002, NodeId 000000002F73F24D, VendorId 0x1384 Label , Initialized 1
```

mustafayuce33, 2023-01-06:
@bzbarsky-apple And one more thing when fabric label is set (https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/FabricTable.cpp#L2049), OnFabricUpdated is not called.

bzbarsky-apple, 2023-01-09:
> when I commissioned a network (Apple Home or Google Home) there seems 2 records in fabric table which one of them label name is empty.

When commissioning with Apple Home, I would expect it to be commissioned into the System Commissioner fabric and the Home-specific fabric.  Removing it from the one Home will remove from the latter, but not the former.

> And one more thing is that when fabric label is set

OnFabricUpdated is documented as being called when the fabric's operational credentials change.  A fabric label is not credentials.

closed 2023-08-11 by stale[bot]