project-chip/connectedhomeip #24254: [ICD] Add broadcasting logic over DNS-SD for controllers

mkardous-silabs, 2023-01-04:
### Feature description

When an ICD wakes up from a long sleep, it's possible that it will no longer have any active sessions with resgistered controllers. Since controllers can change IP addresses over time, the ICD will have to resolve the controllers IP address.

As it stands, it is not possible to resolve a controllers IP address since controllers don't broadcast their existance over the network. Controllers need to maintain an active dns-sd entry for ICDs to be able to resolve their current IP address.

### Platform

all

bzbarsky-apple, 2023-01-04:
> it is not possible to resolve a controllers IP address since controllers don't broadcast their existance over the network.

Sure they do, if "enable server interactions" is enabled in the controller factory setup.

bzbarsky-apple, 2023-01-04:
> where is the controller factory setup done?

It depends on the specific controller.  See `FactoryInitParams` and its `enableServerInteractions` member.

bzbarsky-apple, 2023-01-04:
In particular, in-tree controllers set things up at:
```
examples/chip-tool/commands/common/CHIPCommand.cpp:    chip::Controller::FactoryInitParams factoryInitParams;
examples/platform/linux/CommissionerMain.cpp:    Controller::FactoryInitParams factoryParams;
src/controller/java/AndroidDeviceControllerWrapper.cpp:    chip::Controller::FactoryInitParams initParams;
src/controller/python/ChipDeviceController-ScriptBinding.cpp:    FactoryInitParams factoryParams;
src/controller/python/chip/internal/CommissionerImpl.cpp:        chip::Controller::FactoryInitParams factoryParams;
src/darwin/Framework/CHIP/MTRDeviceControllerFactory.mm:        chip::Controller::FactoryInitParams params;
```

mkardous-silabs, 2023-02-15:
The Chip-tool in interactive mode does this and thus can be resolved by an end device. Will close this issue since this is already done for the chip-tool.

Other contronllers will have to do this on their platforms.

closed 2023-02-15 by mkardous-silabs