project-chip/connectedhomeip #24265: Timestamps in Diagnostic Logs cluster do not match spec

bzbarsky-apple, 2023-01-04:
Per spec, in RetrieveLogsResponse the UTCTimeStamp should be `epoch-us` (64-bit microseconds) and the TimeSinceBoot should be `systime-us` (64-bit microseconds).

In the SDK right now UTCTimeStamp is an `epoch-s` in the XML, but the actual value the cluster stores in there is a _millisecond_ monotonic timestamp (which is neither seconds, nor an epoch value).

In the SDK right now TimeSinceBoot is an INT32U in the XML.  It's never actually sent on the wire by the current cluster implementation.

msandstedt, 2023-01-04:
Changing the two fields to 64-bit microseconds seems easy enough, and I think a change can be justified since few or no devices will currently be sending anything sensible based on the issue description.

And clearly we can't have UTCTimeStamp as required if knowledge of time isn't guaranteed, so can't this be made an optional field?

bzbarsky-apple, 2023-01-04:
> so can't this be made an optional field?

Both timestamp fields are already optional.

closed 2023-02-14 by bzbarsky-apple