project-chip/connectedhomeip #24281: [1.0] Thread Network Diagnostics cluster: impossible to obtain RLoc16 for the current node

slavikus, 2023-01-05:
### Reproduction steps

The Thread Network Diagnostics cluster seems to lack an attribute that would return an RLoc16 for the node being queried. There are RLoc16s in the routes and neighbors table, but no way of obtaining the device's own RLoc16.

It could have been derived from the General Diagnostics Cluster -> Network Interfaces -> IPv6Addresses, but these also do not expose the routing locator IP (`::ff:fe00:`).

Is there any way to obtain the current Thread node RLoc16 that I have overlooked?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

561d23d0db215a99705ff0696e73853c8edf11b2

### Platform

core

### Type

Common Cluster Logic

turon, 2023-01-05:
It looks like it may be an oversight.  I would ask why you need the ephemeral RLOC of a node, but I would assume the answer is “to make meaning out of the RouteTable and NeighborTable entries”.

The IPv6 address parsing method could potentially work. My reading of `General Diagnostics Cluster -> Network Interfaces -> IPv6Addresses` is that the full 128-bit IPv6 addresses are listed, so RLOCs could potentially be gleaned as follows:

- Thread_Mesh_Local_Prefix::RLOC_IID
- Where RLOC_ID has the form per `5.2.2.1 Routing Locator (RLOC)`: `0000:00FF:FE00:xxxx where ‘xxxx’ is ... the RLOC`

slavikus, 2023-01-06:
Thank you @turon for a quick response!

Indeed, I also thought that I could derive the RLoc from the IPv6 addresses, but as I have originally mentioned, the needed IP containing that is also not exposed.

I also been thinking the routes table may have a "self" entry with a cost of 0 (or something), but it seems to not be the case as well.

So far the only weak method I found is if the surveyed node is a router having children connected, it's possible to deduct its RLoc16 from the children's.

And yeah, I need the RLoc to create the topology map of the network (shown in the Thread Network view of the Eve app for the iOS). :) 

I agree it changes over time, but having rloc references in the routes and neighbors lists and not having it for the node itself is a little odd.

So I assume we will need an additional attribute in the cluster for the future versions of Matter.

BTW, the description for the PanId attribute reads like it was really about an RLoc16 (as the panId identifies the network and not a single node):

> 11.13.7.4. PanId Attribute 
> The PanId attribute SHALL indicate the 16-bit identifier of the Node on the Thread network. A value of null SHALL indicate that the Thread interface is not currently configured or operational.

turon, 2023-01-06:
I created https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6153 to track the spec change.

The PANID description bug is an interesting observation as well.

woody-apple, 2023-06-24:
Closing this, given the spec issue is tracking this now.

closed 2023-06-24 by woody-apple