project-chip/connectedhomeip #24297: [Platform] Implement a DeviceAttestationVerifier in Java to allow user to overwrite the DefaultDACVerifier

yufengwangca, 2023-01-06:
### Reproduction steps

Per @tcarmelveilleux
 
See `AndroidDeviceControllerWrapper::AllocateNew` in src/controller/java/AndroidDeviceControllerWrapper.cpp:
```
    // Initialize device attestation verifier
    // TODO: Replace testingRootStore with a AttestationTrustStore that has the necessary official PAA roots available
    const chip::Credentials::AttestationTrustStore * testingRootStore = chip::Credentials::GetTestAttestationTrustStore();
    SetDeviceAttestationVerifier(GetDefaultDACVerifier(testingRootStore));
```

The Device Attestation Verifier is what could be bridged to Java.

Overall, the flow could be something like:

[image]

```
@startuml

participant "Client" as client 
participant "AndroidChipDeviceController" as wrapper
' from AndroidDeviceControllerWrapper
participant "C++ ChipDeviceController" as DevCtrl
' owned by ChipDeviceController
participant "DeviceCommissioner" as Commissioner
' owned by ChipDeviceController
participant "AutoCommissioner" as AutoCommissioner
' Make your custom one
participant "CustomDeviceAttestationVerifier" as Verifier
participant "DCL Client" as DCL

client -> wrapper: <<new ChipDeviceController>>
client <- wrapper: devCtrl
' The following is new
client -> wrapper: SetDeviceAttestationVerifier\n(CustomDeviceAttestationVerifier)
activate wrapper
client -> wrapper: CommissionDevice(...)
wrapper -> DevCtrl: Commission(...)
DevCtrl -> Commissioner: Commission(...)
Commissioner -> AutoCommissioner: StartCommissioning(...)

... some operations pass ...

AutoCommissioner -> Verifier: VerifyAttestationInformation(..., Callback)
Verifier -> DCL: FetchPAA(PAI.authority, PaaFetchedCallback)
Verifier <-- DCL: PaaFetchedCallback(PAA Cert)
Verifier -> Verifier: Bridge to DefaultDeviceAttestationVerifier impl
Note left of Verifier: An ephemeral truststore can be provided\nto DefaultDeviceAttestationVerifier\nthat now has PAA
Note left of Verifier: DefaultDeviceAttestationVerifier does not\nneed to be used, could be custom Java version
AutoCommissioner <-- Verifier: Callback(result)

... more operations pass ...
Commissioner <-- AutoCommissioner: CommissioningStepFinished(kAttestationVerification)
deactivate wrapper

@enduml
```

The general concept:

1. It should be possible to implement a DeviceAttestationVerifier in Java fully
2. That DeviceAttestationVerifier would then NOT need a truststore directly, unless it called into the C++ DefaultDeviceAttestationVerifier with JNI bridging. This should be optional
3. If we abstract at DeviceAttestationVerifier call, then there is already async implementation from DeviceCommissioner to AutoCommissioner to DeviceAttestationVerifier. You can then take a while to obtain PAA from DCL if needed
4. We have to make sure not to break existing public API, only make it better. Right now, most Java clients just use the default Device Attestation Verifier and likely know what PAAs they want to trust (e.g. their own). If you are a fully general commissioner, you always need more customization anyway. That is why I was suggesting *also bridging* the attestation trust store.

Regarding the CD trust store, there is already a set of Well-known keys. The CdWellKnownKeysTrustStore used by default should be good enough for a while.

Overall, I agree that these Java APIs are cumbersome. However, unless someone is willing to help rewrite them to "flow-through" with a frameworked design that works *for everyone*, we have to keep going incrementally.

### Platform

android

### Type

Manually tested with SDK

yufengwangca, 2023-01-18:
@panliming-tuya in your refined workflow, is the AndroidDeviceAttestationVerifier new C++ class talk to PaaFetcherDelegate?Why this new flow is better than the one Tennessee proposed?   I think both flows do the similar job, allow the client to fetch the PAA from DCL on the fly during attestation process.

panliming-tuya, 2023-01-19:
> @panliming-tuya in your refined workflow, is the AndroidDeviceAttestationVerifier new C++ class talk to PaaFetcherDelegate?Why this new flow is better than the one Tennessee proposed? I think both [...]

@yufengwangca It has been modified and the interface design has been added

closed 2023-03-07 by andy31415

luobosu, 2023-03-10:
how about provide a path which store the PAA files  from java-layer to native-layer .   
then we can new a FileAttestationTrustStore.cpp with the paa path.
then set the  FileAttestationTrustStore into the DefaultDevieAttestationVerifier