project-chip/connectedhomeip #24300: [Documentation] Matter 1.0 Application Cluster Specifications "Reportable" for On/Off cluster

Apollon77, 2023-01-06:
### Documentation issues

1. In the Matter 1.0 Application Cluster Specifications the "Reportable" quality information is missing for the OnOff attrible for the OnOff Cluster. The chip XMLs contain this as "reportable" and it also seems that some manufactorers expect it to be reportable and pairing breaks without it

### Platform

other

bzbarsky-apple, 2023-01-06:
In Matter all attributes are "reportable" except the ones that have the C quality.

closed 2023-01-06 by bzbarsky-apple

bzbarsky-apple, 2023-01-06:
Or to be more precise, the "reportable" quality (P) in the cluster specs only affects Zigbee behavior.  In Matter, the behavior is as follows:

1. The priming part of subscription setup sends values for all attributes which match the given attribute paths.
2. After that, updates are sent for all attributes except the ones that have the "C" ("Changes Omitted") quality.

The Reportable quality definition in Matter is:

> The Subscribe interaction supports all attribute data. This quality is supported by other interactions that only require attribute data with this quality to support interval or change reporting.

and "other interactions" in this case is the Zigbee attribute reporting interaction.

Apollon77, 2023-01-06:
Thank you for the clarification!

Apollon77, 2023-01-06:
But @bzbarsky-apple Then we still have a mismatch in specs vs chip XML files regarding "onOff" attribute of the onoff cluster! WHich is right? Spec or XML? The other should be fixed

bzbarsky-apple, 2023-01-09:
The spec is wrong here, and I expect it will be fixed once it's merged with the Zigbee-targeted data model specs.