project-chip/connectedhomeip #24301: [Documentation] Matter Core Specs openCommissioningWindow - discriminator length incorrect (or iOS error)

Apollon77, 2023-01-06:
### Documentation issues

1. in the Matter Core Specs 1.0 in  section 11.18.8.1. the maximum lengthof the discriminator field is defined with 2047.  The discriminator is an unsigned 12-bit so the max is 4095 and iOS sends discriminators above 2047.

So it more seems to be that the specs is wrong, but in fact it could also be that iOS is behaving wrong

### Platform

other

bzbarsky-apple, 2023-01-06:
This was a mistake in the spec, which was fixed in https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5714 (PR title: "Discriminator range is incorrect in 11.18.8.1 (OpenCommissioningWindow command)").

bzbarsky-apple, 2023-01-06:
And @Apollon77, thank you for catching this!  If this had not been fixed yet, it would obviously have been critical to fix ASAP.

closed 2023-01-06 by bzbarsky-apple

Apollon77, 2023-01-06:
When an updated Spec is about to be released?

bzbarsky-apple, 2023-01-09:
@Apollon77 I believe the goal is sometime this spring.