project-chip/connectedhomeip #24308: [BUG] Group message fails with Invalid condition found in packet header

Jakebluhm, 2023-01-06:
### Reproduction steps

I am trying to send a command to the OnOff cluster to a group of end devices. Below are my environment and reproduction steps:

## **Environment:**
Matter Controller: chip-tool running on Linux
End devices: nRF52840 Dev Kits running modified Template from nRF Connect SDK v2.1.0

## **Reproduction steps:**
#### Commission 2 end devices
```sudo ./chip-tool pairing ble-thread 1 hex:0e080000... 20202021 3840```

```sudo ./chip-tool pairing ble-thread 2 hex:0e080000... 20202021 3841```

#### Key set write to endpoint 0 of both end devices
```sudo ./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": "0x01a1", "groupKeySecurityPolicy": 0, "epochKey0":"a0a1a2a3a4a5a6a7a8a9aaabacadaeaf", "epochStartTime0": 1110000, "epochKey1":"b0b1b2b3b4b5b6b7b8b9babbbcbdbebf", "epochStartTime1": 1110001, "epochKey2":"c0c1c2c3c4c5c6c7c8c9cacbcccdcecf", "epochStartTime2": 1110002 }' 1 0```
 
```sudo ./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": "0x01a1", "groupKeySecurityPolicy": 0, "epochKey0":"a0a1a2a3a4a5a6a7a8a9aaabacadaeaf", "epochStartTime0": 1110000, "epochKey1":"b0b1b2b3b4b5b6b7b8b9babbbcbdbebf", "epochStartTime1": 1110001, "epochKey2":"c0c1c2c3c4c5c6c7c8c9cacbcccdcecf", "epochStartTime2": 1110002 }' 2 0```

#### Map key to group
 ```sudo ./chip-tool groupkeymanagement write group-key-map '[{"fabricIndex": 0, "groupId": "0x0101", "groupKeySetID": "0x01a1"}]' 1 0```

 ```sudo ./chip-tool groupkeymanagement write group-key-map '[{"fabricIndex": 0, "groupId": "0x0101", "groupKeySetID": "0x01a1"}]' 2 0```

#### Add endpoints to group
```sudo ./chip-tool groups add-group  0x0101 "ARGB" 1 1```
```sudo ./chip-tool groups add-group  0x0101 "ARGB" 2 1```

#### Send on command to group
```sudo ./chip-tool onoff on 0xffffffffffff0101 1```

This command returns an error on the end device, also attached output from chip-tool:

end device output:
E: 7401806 [IN]Invalid condition found in packet header

This points me to [https://github.com/project-chip/connectedhomeip/blob/c279578c5bc37f117335aa96cec6c5552f070cc0/src/transport/SessionManager.cpp](url) line 763

```c    
// Check if Message Header is valid first
    if (!(packetHeader.IsValidMCSPMsg() || packetHeader.IsValidGroupMsg()))
    {
        ChipLogError(Inet, "Invalid condition found in packet header");
        return;
    }
```

chip-tool 
[output.txt](https://github.com/project-chip/connectedhomeip/files/10361577/output.txt)

                                                                                                                                                                                                                                       

Any suggestions are appreciated

### Bug prevalence

Every attempt

### GitHub hash of the SDK that was being used

16bcd30c33241b4b1c8993cfbc92e65f4de9708d

### Platform

nrf

### Platform Version(s)

v2.1.0

bzbarsky-apple, 2023-01-06:
> 16bcd30c33241b4b1c8993cfbc92e65f4de9708d

Which branch is that hash from?  I just tried the above steps on tip and they seem to work fine....

Jakebluhm, 2023-01-13:
@bzbarsky-apple I am using Nordic Semiconductor connectedhomeip repo, which makes me question if this is the right place to post this error. 
https://github.com/nrfconnect/sdk-connectedhomeip
v2.1-branch

bzbarsky-apple, 2023-01-14:
@Jakebluhm I would start with reporting the error to them, then...

Jakebluhm, 2023-01-16:
Thanks, @bzbarsky-apple

For anyone with similar issues of not being able to send multicast commands, I've created an issue for this error here -> https://devzone.nordicsemi.com/f/nordic-q-a/95698/matter---how-can-i-control-the-group-created-in-nrf-connect-sdk-matter-light-switch-and-light-bulb-samples-from-the-chip-tool-rather-than-the-light-switch

LuDuda, 2023-01-30:
Hello @Jakebluhm I think our tech support replied.

The nRF Connect SDK 2.1.0 includes experimental and old version of Matter SDK (pre Matter 1.0) - so I'm not sure if group communication worked at this moment of time. I suggested to use nRF Connect SDK 2.2.0 which includes production support for Matter.

Btw. I have tested that with below commands group communication works well.

My setup:
 SDK: nRF Connect SDK 2.2.0
 HW: nRF52840
 OTBR: RaspPi and nrfconnect/otbr:1813352
 Sample: samples/matter/light_bulb build without any extra parameters
 Chip-tool: From nRF Connect SDK 2.2.0 - https://github.com/nrfconnect/sdk-connectedhomeip/releases/tag/v2.2.0

```
# Commission device to the Thread network
sudo ./chip-tool-debug pairing ble-thread 1 hex:0e080000000000010000000300000f35060004001fffe0020811111111222222220708fd7b8210c9e19eb1051000112233445566778899aabbccddeeff030e4f70656e54687265616444656d6f010212340410445f2b5ca6f2a93a55ce570a70efeecb0c0402a0f7f8 20202021 3840
# Check that unicast onoff works well (observe LED2)
sudo ./chip-tool-debug onoff on 1 1
sudo ./chip-tool-debug onoff off 1 1
# Configure test group
sudo ./chip-tool-debug groupkeymanagement key-set-write '{"groupKeySetID": "0x01a1", "groupKeySecurityPolicy": 0, "epochKey0":"a0a1a2a3a4a5a6a7a8a9aaabacadaeaf", "epochStartTime0": 1110000, "epochKey1":"b0b1b2b3b4b5b6b7b8b9babbbcbdbebf", "epochStartTime1": 1110001, "epochKey2":"c0c1c2c3c4c5c6c7c8c9cacbcccdcecf", "epochStartTime2": 1110002 }' 1 0
sudo ./chip-tool-debug groupkeymanagement write group-key-map '[{"fabricIndex": 0, "groupId": "0x0101", "groupKeySetID": "0x01a1"}]' 1 0
sudo ./chip-tool-debug groups add-group 0x0101 "ARGB" 1 1
# Configure ACL (257 = 0x0101)
sudo ./chip-tool-debug accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": null, "targets": null },{"fabricIndex": 1, "privilege": 3, "authMode": 3, "subjects": [257], "targets": null }]' 1 0
# Test that Group communication works as expected
sudo ./chip-tool-debug onoff on 0xffffffffffff0101 1
sudo ./chip-tool-debug onoff off 0xffffffffffff0101 1
```

closed 2023-10-15 by stale[bot]