project-chip/connectedhomeip #24319: [Documentation] Status ILLEGAL_COMMAND (used in Application Cluster Specs) is nowhere detailed

Apollon77, 2023-01-07:
### Documentation issues

1. In the Application cluster specification 1.0 there is ILLEGAL_COMMAND used in several places (e.g. Scenes Cluster), but I can not find that code in any other document to get the status Code number

I could only assume that ILLEGAL_COMMAND is another "old name" for

InvalidCommand = 0x85

(which already have old name INVALID_FIELD)

Or what is correct here?

### Platform

other

bzbarsky-apple, 2023-01-09:
As far as I can tell ILLEGAL_COMMAND is only used in "Scenes".  It used to say INVALID_FIELD, and looks like someone messed up while renaming to INVALID_COMMAND.  I filed https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6162

closed 2023-01-09 by bzbarsky-apple

Apollon77, 2023-01-09:
So the real future proof status to use is InvalidCommand. Thanky for confirmation