project-chip/connectedhomeip #24329: [BUG] Unable to perform factory reset after removing fabric during ota

fuxiaoming-lumi, 2023-01-09:
### Reproduction steps

1. Use chip-tool to commission DUT(Matter over Thread) and chip-ota-provider-app to the same fabric.
2. Configure the OTA Provider with the access control list (ACL) that grants Operate privileges to all nodes in the fabric. 
3. Use chip-tool to send the Announce OTA Provider command to the DUT and trigger OTA progress.
4. Use chip-tool to remove-fabric when the ota progress is about 10%.
5. In this case, the factory reset can‘t be performed successfully ( API - chip::Server::GetInstance().ScheduleFactoryReset() ). 
6. Reboot the DUT and the factory reset can be performed successfully.

### Bug prevalence

100% reproduce

### GitHub hash of the SDK that was being used

83c3b6b15164034c1f7c56938f5249289955ee1e

### Platform

efr32, k32w

### Anything else?

With NXP K32W Platform:
[chip-ota-provider-app-with-nxp-k32w.txt](https://github.com/project-chip/connectedhomeip/files/10371484/chip-ota-provider-app-with-nxp-k32w.txt)
[chip-tool-nxp-k32w.txt](https://github.com/project-chip/connectedhomeip/files/10371485/chip-tool-nxp-k32w.txt)
[dut-nxp-k32w.txt](https://github.com/project-chip/connectedhomeip/files/10371487/dut-nxp-k32w.txt)

With Silicon Labs MG24 Platform:
[chip-ota-provider-app-silicon-mg24.txt](https://github.com/project-chip/connectedhomeip/files/10371489/chip-ota-provider-app-silicon-mg24.txt)
[chip-tool-silicon-mg24.txt](https://github.com/project-chip/connectedhomeip/files/10371492/chip-tool-silicon-mg24.txt)
[dut_silicon-mg24.txt](https://github.com/project-chip/connectedhomeip/files/10371493/dut_silicon-mg24.txt)

It is necessary to verify whether other platform has the same problem.

bzbarsky-apple, 2023-01-09:
> In this case, the factory reset can‘t be performed successfully ( API - chip::Server::GetInstance().ScheduleFactoryReset() ).

What happens when you call the API?  Is the lambda just not scheduled, or does it for some reason not do the right thing @fuxiaoming-lumi ?

marius-alex-tache, 2023-01-27:
I can reproduce this on K32W0 contact sensor. What I'm seeing is that the app event queue becomes full because the events are never consumed. I'm not sure why this happens (edit: please see the note below). I've put a breakpoint inside `AppTaskMain` main loop, but it never triggers.

Actually, it seems that there is a failed assert, but somehow the button interrupts are still handled.
[image]

Is there a race condition somewhere?

bzbarsky-apple, 2023-01-28:
Does the OTA stuff keep running even though the fabric has been removed or something?

marius-alex-tache, 2023-01-30:
I saw that some packets sent by OTAP were still received after removing the fabric (at least I saw some UDP message received logs), but I'm not certain if I remember correctly. Will have to test more to confirm.

marius-alex-tache, 2023-01-30:
```[117866[P][ZCL]OpCreds: **RemoveFabric successful**
[117880[D][DMG]Command handler moving to [ Preparing]
[117885[D][DMG]Command handler moving to [AddingComm]
[117897[D][DMG]Command handler moving to [AddedComma]
----> [117903[D][IN]Expiring all sessions for fabric 0x1!!
[117912[D][IN]SecureSession[0x400da18]: MarkForEviction Type:2 LSID:21793
[117922[P][SC]SecureSession[0x400da18]: Moving from state 'kActive' --> 'kPendingEviction'
[117931[D][IN]SecureSession[0x400da18]: Released - Type:2 LSID:21793
[117941[D][IN]SecureSession[0x400dad0]: MarkForEviction Type:2 LSID:21796
[117950[P][SC]SecureSession[0x400dad0]: Moving from state 'kActive' --> 'kPendingEviction'
[117960[D][IN]SecureSession[0x400db88]: MarkForEviction Type:2 LSID:21795
[117970[P][SC]SecureSession[0x400db88]: Moving from state 'kActive' --> 'kPendingEviction'
[117981[D][IN]SecureSession[0x400db88]: Released - Type:2 LSID:21795
[117988[D][IN]SecureSession[0x400dc40]: MarkForEviction Type:2 LSID:21797
[117999[P][SC]SecureSession[0x400dc40]: Moving from state 'kActive' --> 'kPendingEviction'
[118010[E][EM]Failed to send Solitary ack for MessageCounter:41142393 on exchange 35046i:2
----> [118019[D][IN]SecureSession[0x400dc40]: Released - Type:2 LSID:21797
[118030[D][DMG]Decreasing reference count for CommandHandler, remaining 0
[118037[P][EM]<<< [E:58927r M:118536181 (Ack:94750963)] (S) Msg TX to 1:000000000001B669 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
[118053[P][IN](S) Sending msg 118536181 on secure session with LSID: 21796
[118061[D][DMG]Command handler moving to [CommandSen]
[118070[D][DMG]Command handler moving to [AwaitingDe]
[118076[E][DL]Long dispatch time: 848 ms, for event type 3
[118084[P][EM]>>> [E:58926r M:199548407 (Ack:166991627)] (U) Msg RX from 0:8B8E894580E092EE [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[118103[D][EM]Found matching exchange: 58926r, Delegate: 0
[118110[D][EM]Rxd Ack; Removing MessageCounter:166991627 from Retrans Table on exchange 58926r
----> [118120[E][IN]Data received on an unknown session (LSID=21797). Dropping it!
[118127[E][EM]Failed to send Solitary ack for MessageCounter:41142393 on exchange 35046i:2
[118136[E][EM]Failed to send Solitary ack for MessageCounter:41142393 on exchange 35046i:2
[118144[E][EM]Failed to send Solitary ack for MessageCounter:41142393 on exchange 35046i:2
[118159[E][EM]Failed to send Solitary ack for MessageCounter:41142393 on exchange 35046i:2
[118174[E][EM]Failed to send Solitary ack for MessageCounter:41142393 on exchange 35046i:2
[118188[E][EM]Failed to send Solitary ack for MessageCounter:41142393 on exchange 35046i:2
[118209[D][IN]UDP Message Received packet nb : 42 SrcAddr : fd11:33:0:0:573e:df41:e3b0:afa6[5540] DestAddr : fd11:33:0:0:cf57:8fd8:ccac:a903[5540] Payload Length 1062
[118229[D][IN]UDP Message Received packet nb : 43 SrcAddr : fd11:33:0:0:573e:df41:e3b0:afa6[44291] DestAddr : fd11:33:0:0:cf57:8fd8:ccac:a903[5540] Payload Length 34
Solitary ack for MessageCounter:41142393 on exchange 35046i:2
[118250[E][IN]Data received on an unknown session (LSID=21797). Dropping it!
[118258[P][EM]>>> [E:58927r M:94750964 (Ack:118536181)] (S) Msg RX from 1:000000000001B669 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[118271[D][EM]Found matching exchange: 58927r, Delegate: 0
[118277[D][EM]Rxd Ack; Removing MessageCounter:118536181 from Retrans Table on exchange 58927r
[118286[D][IN]SecureSession[0x400dad0]: Released - Type:2 LSID:21796
[118292[E][EM]Failed to send Solitary ack for MessageCounter:41142393 on exchange 35046i:2
```
Seems like a session is used after it's released (hence the assert in the stack trace above). Marked some entries with `---->`.

bzbarsky-apple, 2023-01-30:
@marius-alex-tache Is a fabric removing another fabric, or is a fabric self-removing itself in this case?

marius-alex-tache, 2023-01-31:
The use case is done using only one controller, so it's self-removing.

bzbarsky-apple, 2023-01-31:
@marius-alex-tache Can you please attach, not paste, a full log, not one starting with the "[34112[D][IN]Expiring all sessions for fabric 0x1!!" bit?  I need to understand what happened before that bit, because for self-removing we should not really be getting there at all.

marius-alex-tache, 2023-02-01:
Hi, sorry about the logs. I've attached one: [contact-sensor-remove-fabric.log](https://github.com/project-chip/connectedhomeip/files/10558436/contact-sensor-remove-fabric.log)

marius-alex-tache, 2023-02-01:
I've added a log at the beginning of the lambda function. Please see this log: [contact-sensor-remove-fabric-print-exchange-context.log](https://github.com/project-chip/connectedhomeip/files/10559429/contact-sensor-remove-fabric-print-exchange-context.log). It seems the exchange context is `6275i`, which is the context related to BDX messages.

bzbarsky-apple, 2023-02-01:
Thanks, I think I know what's going on now.  In that log, we tear down the session here:
```
[49100[P][SC]SecureSession[0x400fcb0]: Moving from state 'kActive' --> 'kPendingEviction'
[49113[E][BDX]exchange timed out
[49117[D][BDX]aborting due to timeout
...
[49202[D][IN]SecureSession[0x400fcb0]: Released - Type:2 LSID:19002
```

Now the exchange should have been removed from the MRP bits as part of that, but it's not because we end up doing the following:

1. `ExchangeContext::OnSessionReleased` is called.  We are not in the `IsSendExpected` state and _are_ in the `IsResponseExpected` state, so we notify our delegate that we have timed out.
2. This calls `BDXMessenger::OnResponseTimeout` (that's out delegate).
3. That calls `BDXDownloader::OnDownloadTimeout`.
4. That calls `SetState(State::kIdle, OTAChangeReasonEnum::kTimeOut);`
5. `BDXDownloader::SetState` calls `mStateDelegate->OnDownloadStateChanged()`.
6. That lands us in `DefaultOTARequestor::OnDownloadStateChanged`, which sees we are entering the `kIdle` state and calls `mBdxMessenger.Reset();`
7. `BDXMessenger::Reset` does `mExchangeCtx->Close();`
8. This calls `ExchangeContext::DoClose(false)`, so that we do not remove the exchange from the MRP tracking.
9. Now the stack unwinds and we call `Abort` in `ExchangeContext::OnSessionReleased`, but we are already closed so the `DoClose(true)` call that makes ignored.
10. Now we have a reference to a session-less exchange sitting around and that blows up.

The key part: Item 7 above is an API contract violation.  You are NOT allowed to call `Close` from inside `OnResponseTimeout`.  If we did not run into this other issue, we would run into the problem that we are going to underflow the exchange refcount when the stack unwinds and we do the `Abort`...

@Damian-Nordic @holbrookt @carol-apple what is the best way to fix this code to follow the expected contract for exchanges?  Some thoughts:
* It looks like we can't just rely on `OTAChangeReasonEnum::kTimeOut` because that's used for things other than exchange timeouts.
* We could do something where we null out `mExchangeCtx` before calling `OnDownloadTimeout` in `BDXMessenger::OnResponseTimeout`, if we are sure nothing under `OnDownloadTimeout` will try to use the exchange.
* We could keep some state in `BDXMessenger` where we flag that we are inside `OnResponseTimeout` and ignore `Reset` calls during that time, but that makes it use a bit more memory.

Other options?

Damian-Nordic, 2023-02-02:
Nulling out `mExchangeCtx` in `OnDownloadTimeout` sounds like the best option to me. `mExchangeCtx` is a private member of `BDXMessenger`, so the messenger can drop the exchange whenever it no longer needs it and knows that it will be destroyed properly.

bzbarsky-apple, 2023-02-02:
@marius-alex-tache Could you verify that the change in https://github.com/project-chip/connectedhomeip/pull/24818 fixes the issue for you?

marius-alex-tache, 2023-02-02:
I confirm the fix works. Thank you! @fuxiaoming-lumi, could you also try to test on your side with this fix?

jrhees-cae, 2023-02-02:
I can also confirm this bug and that the fix corrects the issue (on a Nordic/Thread platform).

closed 2023-02-02 by bzbarsky-apple

fuxiaoming-lumi, 2023-02-03:
I also confirm this bug is fixed with our product. Thanks all your support!