project-chip/connectedhomeip #24330: [BUG] GHA remove silicon (BRD4161A) device, device can't recommisstion again

resghst, 2023-01-09:
### Reproduction steps

connectedhomeip version: b953c0b8e5587d7c21a53b62e7c4dfae0ece08ba
Google nest hub2 version: 1.56.324896 
Example: lighting-app
Board: BRD4161A

I use GHA with Google nest hub2 to commission a silicon light, and I can control the light.
Every-things are working, but when I  remove the device from GHA and try to reconnect it.

I think the Board's state will change to commissionable and start advertising, but the Board's state is already provisioned. I found the LED 0 state is  _Solid On_ (The device is fully provisioned and has full Thread network and service connectivity.)

The same issue is occurring on the Raspberry pi platform. Maybe this issue is all of matter device issue because during the remove action (RemoveFabric Endpoint=0 Cluster=0x0000_003E Command=0x0000_000A) will not send any onplatfrom event to BLE implement layer. Without the platform event, I think providers can't process this.

When GHA sents remove action, the BRD4161A will receive the following logs:

[efr32.log](https://github.com/project-chip/connectedhomeip/files/10371881/efr32.log)

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

b953c0b8e5587d7c21a53b62e7c4dfae0ece08ba

### Platform

efr32, raspi, other

bzbarsky-apple, 2023-01-09:
Removing the device from the last fabric does not automatically open a commissioning window.  Was a commissioning window opened on the device @resghst ?

resghst, 2023-01-13:
Hi @bzbarsky-apple, thanks for the replay.
I try to open a commissioning window (```$ chip-tool pairing open-commissioning-window```), and it works.
I have another question,
When I use open-commissioning-window, the device will use the IP network (Thread) for another connection. Thus, The device can't return to BLE advertising mode for scan QRCode commissioning.  With the last RemoveFabric trigger, the device will become a state, in which no client can control the device. 

My question is "When I RemoveFabric or unpair the device (last fabric), Which command can help the device open BLE advertising".
If I want to set the device's state to BLE adv and wait for commissioning, I guess FactoryReset is the only way

bzbarsky-apple, 2023-01-14:
> the device will use the IP network (Thread) for another connection.

That sounds like a (common) bug, if the network configuration was added by Matter.  Removal of the last fabric should then remove that network configuration.

> for scan QRCode commissioning

Scanning the original QR code should still work if a basic commissioning window was opened (which a device may not support, of course) and the commissioner is on the same IP network (which it might not be).

> With the last RemoveFabric trigger, the device will become a state, in which no client can control the device.

Until it's returned to a "commissioning window is open" mode, yes.

Just like if you unbox a new device, plug it in, then wait 15 minutes.  That's the state you will be in.

>  If I want to set the device's state to BLE adv and wait for commissioning, I guess FactoryReset is the only way

Or just power-cycle it after removing the last fabric, for devices that normally advertise on power-on when fresh out of the box?

resghst, 2023-01-16:
My question has been solved. Thank you.

closed 2023-01-16 by resghst

chuckrui, 2023-04-13:
Hello, I would like to ask a question about how to remove the device from the thread network and put it into the broadcast state. Do I use 
"$./chip-tool pairing unpair" ,
 but I find that the parameter is <node id >, and I find what appears to be 4 commissionables via the 
"$./chip-tool discover commissionables" command?
 But all 4 devices are the same, is it because every time I add them to thread network during testing, I usually restore factory Settings to make them Bluetooth LE discoverable, I now want to delete them from thread network.
`chuckrui@ubuntu:~/matter/chip-tool/chip-tool-V2.3.0$ ./chip-tool-debug discover commissionables
[1681356403.068400][3786:3786] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1681356403.068808][3786:3786] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1681356403.068928][3786:3786] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1681356403.069010][3786:3786] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1681356403.069296][3786:3786] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-0eMz0r)
[1681356403.069774][3786:3786] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1681356403.069821][3786:3786] CHIP:DL: NVS set: chip-counters/reboot-count = 18 (0x12)
[1681356403.070336][3786:3786] CHIP:DL: Got Ethernet interface: eth0
[1681356403.070626][3786:3786] CHIP:DL: Found the primary Ethernet interface:eth0
[1681356403.070935][3786:3786] CHIP:DL: Got WiFi interface: wlan0
[1681356403.072623][3786:3786] CHIP:DL: Found the primary WiFi interface:wlan0
[1681356403.072680][3786:3786] CHIP:IN: UDP::Init bind&listen port=0
[1681356403.072762][3786:3786] CHIP:IN: UDP::Init bound to port=48179
[1681356403.072782][3786:3786] CHIP:IN: UDP::Init bind&listen port=0
[1681356403.072846][3786:3786] CHIP:IN: UDP::Init bound to port=40903
[1681356403.072864][3786:3786] CHIP:IN: BLEBase::Init - setting/overriding transport
[1681356403.072880][3786:3786] CHIP:IN: TransportMgr initialized
[1681356403.072910][3786:3786] CHIP:FP: Initializing FabricTable from persistent storage
[1681356403.073040][3786:3786] CHIP:TS: Last Known Good Time: 2023-03-02T09:31:36
[1681356403.073451][3786:3786] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xAF43809D4244791B, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1681356403.075826][3786:3786] CHIP:ZCL: Using ZAP configuration...
[1681356403.079427][3786:3786] CHIP:DL: Avahi client registered
[1681356403.080697][3786:3786] CHIP:CTL: System State Initialized...
[1681356403.080803][3786:3786] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1681356403.080856][3786:3786] CHIP:CTL: Setting attestation nonce to random value
[1681356403.080888][3786:3786] CHIP:CTL: Setting CSR nonce to random value
[1681356403.081121][3786:3788] CHIP:DL: CHIP task running
[1681356403.081304][3786:3788] CHIP:DL: HandlePlatformSpecificBLEEvent 32784
[1681356403.081574][3786:3788] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1681356403.081618][3786:3788] CHIP:CTL: Setting attestation nonce to random value
[1681356403.081653][3786:3788] CHIP:CTL: Setting CSR nonce to random value
[1681356403.095224][3786:3788] CHIP:CTL: Generating NOC
[1681356403.107936][3786:3788] CHIP:FP: Validating NOC chain
[1681356403.151706][3786:3788] CHIP:FP: NOC chain validation successful
[1681356403.151751][3786:3788] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1681356403.151776][3786:3788] CHIP:TS: Last Known Good Time: 2023-03-02T09:31:36
[1681356403.151793][3786:3788] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1681356403.151808][3786:3788] CHIP:TS: Retaining current Last Known Good Time
[1681356403.158433][3786:3788] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1681356403.160235][3786:3788] CHIP:TS: Committing Last Known Good Time to storage: 2023-03-02T09:31:36
[1681356403.161897][3786:3788] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: AF43809D4244791B)
[1681356403.246041][3786:3788] CHIP:DL: Long dispatch time: 165 ms, for event type 2
[1681356403.246653][3786:3788] CHIP:DL: Avahi browse: cache new
[1681356403.247028][3786:3788] CHIP:DL: Avahi browse: cache new
[1681356403.247350][3786:3788] CHIP:DL: Avahi browse: cache exhausted
[1681356404.248633][3786:3788] CHIP:DL: Avahi browse: all for now
[1681356404.253852][3786:3788] CHIP:DL: Avahi resolve found
[1681356404.254006][3786:3788] CHIP:DIS: Discovered node:
[1681356404.254061][3786:3788] CHIP:DIS: 	Hostname: BEE833D10DB5F538
[1681356404.254122][3786:3788] CHIP:DIS: 	IP Address #1: fd11:22::a017:9558:1205:acab
[1681356404.254173][3786:3788] CHIP:DIS: 	Port: 5540
[1681356404.254220][3786:3788] CHIP:DIS: 	Mrp Interval idle: 1000 ms
[1681356404.254268][3786:3788] CHIP:DIS: 	Mrp Interval active: 1000 ms
[1681356404.254318][3786:3788] CHIP:DIS: 	TCP Supported: 0
[1681356404.254419][3786:3788] CHIP:DIS: 	Vendor ID: 65521
[1681356404.254514][3786:3788] CHIP:DIS: 	Product ID: 32773
[1681356404.254562][3786:3788] CHIP:DIS: 	Long Discriminator: 3840
[1681356404.254609][3786:3788] CHIP:DIS: 	Pairing Hint: 36
[1681356404.254679][3786:3788] CHIP:DIS: 	Instance Name: 46AD2B56A167CF2B
[1681356404.254749][3786:3788] CHIP:DIS: 	Commissioning Mode: 0
[1681356404.254806][3786:3788] CHIP:DIS: Discovered node:
[... cut ...]
[1681356433.087007][3786:3786] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1681356433.087187][3786:3786] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1681356433.087247][3786:3786] CHIP:DL: Inet Layer shutdown
[1681356433.087293][3786:3786] CHIP:DL: BLE shutdown
[1681356433.087340][3786:3786] CHIP:DL: System Layer shutdown
[1681356433.087700][3786:3786] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:512: CHIP Error 0x00000032: Timeout
`

strangermans, 2023-04-17:
> 试

hello,  Could you please tell me how to solve this problem？

chuckrui, 2023-04-17:
Hello @strangermans ，I have solved the problem，
The version of the OpenThread Boundary Router docker image you downloaded must be the same as the version in the Start OPenThread Boundary Router command.
`sudo docker pull nrfconnect/otbr:a892bf7		    // for ncs2.3.0`
`sudo docker run -it --rm --privileged --name otbr --network otbr -p 8080:80 \
--sysctl "net.ipv6.conf.all.disable_ipv6=0 net.ipv4.conf.all.forwarding=1 net.ipv6.conf.all.forwarding=1" \
--volume /dev/ttyACM0:/dev/radio nrfconnect/otbr:a892bf7 --radio-url spinel+hdlc+uart:///dev/radio?uart-baudrate=1000000`