project-chip/connectedhomeip #24342: [Spec] Occupancy Sensing cluster datamodel (XML) use anonymoous bitmap and enum types in attributes

jsloth, 2023-01-10:
The data-model (XML) for [Occupancy Sensing](https://github.com/project-chip/connectedhomeip/blob/cb430dbe1f81de0bdbd81422b6c3c9a8d1ea97cb/src/app/zap-templates/zcl/data-model/silabs/ha.xml#L390) use anonymous BITMAP and ENUM types in attributes:

https://github.com/project-chip/connectedhomeip/blob/cb430dbe1f81de0bdbd81422b6c3c9a8d1ea97cb/src/app/zap-templates/zcl/data-model/silabs/ha.xml#L398-L400

Should use following types instead:
https://github.com/project-chip/connectedhomeip/blob/cb430dbe1f81de0bdbd81422b6c3c9a8d1ea97cb/src/app/zap-templates/zcl/data-model/silabs/types.xml#L66
https://github.com/project-chip/connectedhomeip/blob/cb430dbe1f81de0bdbd81422b6c3c9a8d1ea97cb/src/app/zap-templates/zcl/data-model/silabs/types.xml#L253
https://github.com/project-chip/connectedhomeip/blob/cb430dbe1f81de0bdbd81422b6c3c9a8d1ea97cb/src/app/zap-templates/zcl/data-model/silabs/types.xml#L259

This issue is similar to https://github.com/project-chip/connectedhomeip/issues/24075 and https://github.com/project-chip/connectedhomeip/pull/23656

bzbarsky-apple, 2023-01-11:
This needs to be fixed in the spec, then the fix can be followed in the SDK.  That way the names will get feedback before being shipped as API to consumers.

bzbarsky-apple, 2023-01-11:
@jsloth Please raise spec issues as needed.

PR #24363 (2023-02-16): Fixed wrong datamodel on Occupancy Sensing Cluster
Fixed #24342 Occupancy Sensing Cluster using anonymous bitmap and enum types, now it uses the correctly defined types (according to spec)
closed 2023-02-16 by jmartinez-silabs