project-chip/connectedhomeip #24343: [BUG]  Device act as dead

CS-Suthar, 2023-01-10:
### Reproduction steps

I'm trying to build the image from [here](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nxp/k32w/k32w0)
I'm just following the steps mentioned in ReadMe.
After building the image it's successfully done after the below command.
**_$NXP_K32W0_SDK_ROOT/tools/imagetool/sign_images.sh out/debug/_**
the result is ....
          _No compatibility list
          Boot Block Offset: 00090f20
          Writing checksum fbfb750c to file out/debug//chip-k32w0x-light-example
          Writing CRC32 of header 0dce2942 to file out/debug//chip-k32w0x-light-example
          Binary size is 00090f40 (593728)_
and when I look in out/debug dir it's showing me both **chip-k32w0x-light-example.bin** and **chip-k32w0x-light-example.hex**

the issue is when I flashed my device and run it's giving me nothing on the terminal even after doing this. 
 _In case that Openthread CLI is needed, chip_with_ot_cli build argument must be set to 1._ 

and now if I flashed my old up-and-running image that is acting the same nothing. 

### Bug prevalence

always

### GitHub hash of the SDK that was being used

K32W0 SDK 2.6.7   &   K32W0 SDK 2.6.9

### Platform

k32w

CS-Suthar, 2023-01-11:
Thanks @bzbarsky-apple 
the issue I had is resolved ( by this [Link](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nxp/k32w/k32w0#ota ) ).
...
I have something new here now I'm using lighting-App on NXP K32W.
**_chip-tool pairing ble-thread <Node_ID> <hex: dataset> 20202021 3840_**
my device is not commissioning on BLE, it gives me a BLEConnection Error as per the below image.

[image]

only 4-5ft is the distance between both devices

bzbarsky-apple, 2023-08-07:
@CS-Suthar Looks like your bluez install might not be ok or something?  I really don't know much about the BLE setup on Linux....