project-chip/connectedhomeip #24367: [BUG] Build Failure --> error: dangling pointer to an unnamed temporary may be used [-Werror=dangling-pointer=]

sid514-m, 2023-01-11:
### Reproduction steps

Steps to Reproduce:
1) Follow the Building steps as mentioned in the link,
https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nrfconnect#building

2) When the Final build command is fired the below error is generated and the build is not completed. 
[Dangling_Pointer_error.log](https://github.com/project-chip/connectedhomeip/files/10393100/Dangling_Pointer_error.log)

Building files for the nrf kit and came across this issue. 

### Bug prevalence

Frequent

### GitHub hash of the SDK that was being used

v1.0.0

### Platform

nrf

### Platform Version(s)

MatterStack v1.0.0

bzbarsky-apple, 2023-01-11:
This got fixed in #23827 which unfortunately fixed it as a drive-by and hence this did not get backported to 1.0...

sid514-m, 2023-01-12:
Thankyou for the Clarity, Boris! Great Help!

bzbarsky-apple, 2023-01-12:
@SiddMishra20 Thank you for finding this!

PR #24373 (2023-01-12): Stop holding a reference into a temporary.
This was drive-by fixed on tip by
https://github.com/project-chip/connectedhomeip/pull/23827 but we should fix on 1.0 too.

Fixes #24367