project-chip/connectedhomeip #24409: [ACL]  Error while encoding attribute-values[0].data as a hex string for D_BAD_NONE: "" which is an octstr of length 0

Survensa, 2023-01-13:
### Reproduction steps

In test case "TC-ACL-2.3 Extension attribute" as per test plan step 16  when TH writes DUT Endpoint 0 AccessControl cluster Extension attribute with AccessControlExtensionStruct containing none element response status should be  0x87 (CONSTRAINT_ERROR) but on V1.0.01 & V1.0.0.2 commit TH is not able to encode NONE attribute-value 

Struct used : D_BAD_NONE: "" which is an octstr of length 0

Command used : ./chip-tool accesscontrol write extension '[{"data":""}]' 1 0

Expected Outcome : 

[1657609694.046042][8236:8241] CHIP:DMG:
[1657609694.046078][8236:8241] CHIP:DMG:                        StatusIB =
[1657609694.046115][8236:8241] CHIP:DMG:                        {
[1657609694.046154][8236:8241] CHIP:DMG:                                status = 0x87 (CONSTRAINT_ERROR),
[1657609694.046192][8236:8241] CHIP:DMG:                        },

Actual Outcome :

[1673601099.394649][21869:21869] CHIP:TOO: Error while encoding attribute-values[0].data as a hex string.
Usage:
  ./chip-tool accesscontrol write attribute-name attribute-values destination-id endpoint-id-ignored-for-group-commands [--paa-trust-store-path] [--cd-trust-store-path] [--commissioner-name] [--commissioner-nodeid] [--use-max-sized-certs] [--only-allow-trusted-cd-keys] [--trace_file] [--trace_log] [--trace_decode] [--ble-adapter] [--timedInteractionTimeoutMs] [--data-version] [--suppressResponse] [--repeat-count] [--repeat-delay-ms] [--timeout]

### Bug prevalence

Issue occurs only in V1.0.0.1 and V1.0.0.2 but not in latest master. 

### GitHub hash of the SDK that was being used

V1.0.0.1 - ca086f2f397fc4128f4964afb17433377e01faf4  V1.0.0.2 - 4088a77f557e8571a39338fad51a1d8eb0131d79	  Master - 2a55f232cf705779d19f05497fa529cd661eef45 Author: Shubham Patil <shubham.patil@espressif.com> Date:   Fri Jan 13 11:12:18 2023 +0530 			 			

### Platform

raspi

bzbarsky-apple, 2023-01-14:
> Issue occurs only in V1.0.0.1 and V1.0.0.2 but not in latest master.

Presumably fixed by https://github.com/project-chip/connectedhomeip/pull/24097

lisaaaas, 2023-03-09:
I have met the same problem. May I ask what type it will be changed to?

bzbarsky-apple, 2023-03-09:
@lisaaaas I'm not sure what you are asking.  What makes you think the type of anything will be changing?

lisaaaas, 2023-03-16:
> @lisaaaas I'm not sure what you are asking. What makes you think the type of anything will be changing?

lisaaaas, 2023-03-16:
> @lisaaaas I'm not sure what you are asking. What makes you think the type of anything will be changing?

What I mean is that the execution of instruction  ./chip-tool accesscontrol write extension '[{"data":""}]' 1 0  is wrong. Is there any relevant modification of this command?

bzbarsky-apple, 2023-03-16:
> What I mean is that the execution of instruction ./chip-tool accesscontrol write extension '[{"data":""}]' 1 0 is wrong.

Wrong in what sense?  The expected behavior of that instruction is:

1. A write is sent to the server with an empty octet string for the data.
2. The server responds with CONSTRAINT_ERROR (because empty string is not a valid extension value).

This issue was about step 1 not happening correctly.

What behavior are you seeing @lisaaaas and what behavior are you expecting to see instead and why?