project-chip/connectedhomeip #24418: [Platform] M5stack runs out of memory pool

kpark-apple, 2023-01-13:
### Reproduction steps

Attempt to commission M5stack.
From M5 log (attached), Packet Buffer: pool EMPTY and Failed to send status report message: b was seen. B is the code for no memory error AFAIK.
It failed occasionally and not always (per tester, one out of five times).

[M5_Pairing_failed.txt](https://github.com/project-chip/connectedhomeip/files/10415829/M5_Pairing_failed.txt)

### Platform

esp32

### Type

Manually tested with SDK

### (Optional) If manually tested please explain why this is only manually tested

It's tested with Apple devices as commissioner.

dhrishi, 2023-01-18:
@kean-apple Can you please share the Matter SDK version and the ESP-IDF version this was tested with? Also, any other specifics to consider about the network or the environment?  I will try this out once these details are added.

esliu-apple, 2023-01-19:
@dhrishi, ESP-IDF version: v4.4. Thanks.

dhrishi, 2023-01-19:
@esliu-apple Is it `v4.4.3`? We support `v4.4.3` for Matter

dhrishi, 2023-01-20:
@esliu-apple Ohh.. so you should be moving to `v4.4.3` tag. The one above is pretty old, March 2022. Will you please give it a try and see if the issue can be recreated.

dhrishi, 2024-10-25:
Closing this as the memory related issues with M5 have been solved

closed 2024-10-25 by dhrishi