project-chip/connectedhomeip #24434: [ESP32] Thermostat cluster usage: how to correctly recognize the change in temperature and mode

CrowdedFuzzball, 2023-01-15:
Hello!

Not really sure if it's okay to ask this question here, but searched through a lot of information and couldn't find an answer.
The examples also don't include an ESP32 thermostat example, so I hope I get the info here.

What I'm trying to do is add a thermostat as a bridged device.

<details>
  <summary>Here is how I add lighting (based on the example)</summary>
  
  ### Some Code
  ```
const EmberAfDeviceType gBridgedOnOffDeviceTypes[]  = { { DEVICE_TYPE_LO_ON_OFF_LIGHT, DEVICE_VERSION_DEFAULT },{ DEVICE_TYPE_BRIDGED_NODE, DEVICE_VERSION_DEFAULT } };

// ---------------------------------------------------------------------------
//
// LIGHT ENDPOINT: contains the following clusters:
//   - On/Off
//   - Descriptor
//   - Bridged Device Basic

// Declare On/Off cluster attributes
DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(onOffAttrs)
    DECLARE_DYNAMIC_ATTRIBUTE(ZCL_ON_OFF_ATTRIBUTE_ID, BOOLEAN, 1, 0), /* on/off */
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

// Declare Cluster List for Bridged Light endpoint
// TODO: It's not clear whether it would be better to get the command lists from
// the ZAP config on our last fixed endpoint instead.
constexpr chip::CommandId onOffIncomingCommands[] = {
    chip::app::Clusters::OnOff::Commands::Off::Id,
    chip::app::Clusters::OnOff::Commands::On::Id,
    chip::app::Clusters::OnOff::Commands::Toggle::Id,
    chip::app::Clusters::OnOff::Commands::OffWithEffect::Id,
    chip::app::Clusters::OnOff::Commands::OnWithRecallGlobalScene::Id,
    chip::app::Clusters::OnOff::Commands::OnWithTimedOff::Id,
    chip::kInvalidCommandId,
};

DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(bridgedLightClusters)
    DECLARE_DYNAMIC_CLUSTER(chip::app::Clusters::OnOff::Id, onOffAttrs, onOffIncomingCommands, nullptr),
    DECLARE_DYNAMIC_CLUSTER(chip::app::Clusters::Descriptor::Id, descriptorAttrs, nullptr, nullptr),
    DECLARE_DYNAMIC_CLUSTER(chip::app::Clusters::BridgedDeviceBasic::Id, bridgedDeviceBasicAttrs, nullptr, nullptr) DECLARE_DYNAMIC_CLUSTER_LIST_END;

// Declare Bridged Light endpoint
DECLARE_DYNAMIC_ENDPOINT(bridgedLightEndpoint, bridgedLightClusters);
  ```
</details>

As a result, I receive a request to read attributes in emberAfExternalAttributeReadCallback and see attributes written to emberAfExternalAttributeWriteCallback

It's all good - I can work with that.

Other types of devices - sensors, for example, which I create also work well.

Next, I'm trying to create a thermostat.

<details>
  <summary>Thermostat Definition</summary>
  
  ### Some Code
  ```
const EmberAfDeviceType gBridgedThermostatDeviceTypes[] = { { DEVICE_TYPE_THERMOSTAT, DEVICE_VERSION_DEFAULT },
{ DEVICE_TYPE_BRIDGED_NODE, DEVICE_VERSION_DEFAULT } };

// ---------------------------------------------------------------------------
//
// THERMOSTAT ENDPOINT: contains the following clusters:
//   - Thermostat
//   - Descriptor
//   - Bridged Device Basic

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(thermostatAttrs)
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_LOCAL_TEMPERATURE_ATTRIBUTE_ID, INT16S, 2, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_ABS_MIN_HEAT_SETPOINT_LIMIT_ATTRIBUTE_ID, INT16S, 2, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_ABS_MAX_HEAT_SETPOINT_LIMIT_ATTRIBUTE_ID, INT16S, 2, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_ABS_MIN_COOL_SETPOINT_LIMIT_ATTRIBUTE_ID, INT16S, 2, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_ABS_MAX_COOL_SETPOINT_LIMIT_ATTRIBUTE_ID, INT16S, 2, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_OCCUPIED_COOLING_SETPOINT_ATTRIBUTE_ID, INT16S, 2, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_OCCUPIED_HEATING_SETPOINT_ATTRIBUTE_ID, INT16S, 2, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_MIN_SETPOINT_DEAD_BAND_ATTRIBUTE_ID, INT8S, 1, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_CONTROL_SEQUENCE_OF_OPERATION_ATTRIBUTE_ID, INT8U, 1, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_SYSTEM_MODE_ATTRIBUTE_ID, INT8U, 1, 0),
	DECLARE_DYNAMIC_ATTRIBUTE(ZCL_FEATURE_MAP_SERVER_ATTRIBUTE_ID, BITMAP32, 4, 0),
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

constexpr chip::CommandId ThermostatIncomingCommands[] = {
	chip::app::Clusters::Thermostat::Commands::SetpointRaiseLower::Id,
    chip::kInvalidCommandId,
};

DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(bridgedThermostatClusters)
	DECLARE_DYNAMIC_CLUSTER(Thermostat::Id, thermostatAttrs, ThermostatIncomingCommands, nullptr),
    DECLARE_DYNAMIC_CLUSTER(Descriptor::Id, descriptorAttrs, nullptr, nullptr),
    DECLARE_DYNAMIC_CLUSTER(BridgedDeviceBasic::Id, bridgedDeviceBasicAttrs, nullptr, nullptr), DECLARE_DYNAMIC_CLUSTER_LIST_END;

// Declare Bridged Thermostat endpoint
DECLARE_DYNAMIC_ENDPOINT(bridgedThermostatEndpoint, bridgedThermostatClusters);

  ```
</details>

Feature map defined like default one.
Cluster revision for thermostat is 5u like documentation said.

The thermostat itself is perfectly displayed in the Home.app interface, I can set the temperature and change modes (although they are not saved and immediately return to their original states)

But the main problem that I see is that there is a call to emberAfExternalAttributeReadCallback, but there is no call to emberAfExternalAttributeWriteCallback. That is, Home.app does not transfer the record of attribute changes to Matter, or I do not understand how to correctly handle the change in thermostat attributes that user made.

MatterPreAttributeChangeCallback and emberAfActionsClusterInstantActionCallback are also empty.

Could you please tell me how to correctly receive thermostat changes on the ESP32?

bzbarsky-apple, 2023-01-15:
> but there is no call to emberAfExternalAttributeWriteCallback

Why would there be a write callback?  The Thermostat cluster itself does not write these attributes, and writing them via the Matter protocol is disabled because the last argument to `DECLARE_DYNAMIC_ATTRIBUTE` is set to 0 for all of them.  For the ones that should be writable, that argument should be `ZAP_ATTRIBUTE_MASK(WRITABLE)`.

CrowdedFuzzball, 2023-01-15:
@bzbarsky-apple 
Amazing, thank you.

closed 2023-01-15 by CrowdedFuzzball