project-chip/connectedhomeip #24436: [1.0] Changing organization name to Matter

guyluz11, 2023-01-15:
### Reproduction steps

If I understand correctly there was a move from naming chip to Matter.

Right now the GitHub organization that this repo is located in is called `project-chip`, but if we moved to Matter it should change to `project-matter`. 

I am not sure if the organization name got changed or just the project for the smart home protocol so I just raising it up.

I have put it as `Issue: 🚀 1.0 Issue` because I think it should be resolved before version one of Matter.

### Bug prevalence

In each url to one of the repo in this ogranization

### GitHub hash of the SDK that was being used

non

### Platform

core

### Type

Test Improvement

guyluz11, 2023-01-16:
Example:
[Here](https://github.com/home-assistant-libs/chip-wheels/issues/3#issuecomment-1383670050)  maintainer of the package explained in response to my question that he called the repo chip-wheel because this organization name is still use the name chip and he will change it whenever it will get changed.

bzbarsky-apple, 2023-01-18:
@guyluz11 Version 1 of Matter was back in October 2022 or so.  Makes it hard to do something "before version one of Matter".

bzbarsky-apple, 2023-01-18:
That said, we should consider renaming our Python wheels for sure.  @tehampson

tehampson, 2023-01-31:
All new python modules that we have been working on recently start with `matter-`. As a part of a larger renaming effort we should be able to move over from chip to matter

hidaris, 2023-02-15:
@tehampson Sorry to be off-topic here, but is matter python currently being considered for release to pypi?

guyluz11, 2023-10-15:
Don't close